marcone/teslausb #502: my TeslaCam Sentry clip is JSON not playable

ashkasto, 2020-07-01:
Can someone help answer why on my windows 10 PC the folder with the sentry mode clips has these JSON files that are not able to be played by my computer.  Thank you

Nakatomi2010, 2020-07-01:
The JSON files are meant to be read by an application.  The JSON file contains the following:

{
	"timestamp":"YYYY-MM-DDTHH:MM:SS",
	"city":"City",
	"est_lat":"Estimated Latitute location",
	"est_lon":"Estimated Longitude location",
	"reason":"How recording was saved"
}

So they aren't playable files, they additional bits of information so a person can see where the video was taken, and how it was recorded.

marcone, 2020-07-01:
As noted above the json files contain a description of the sentry event.
The fact that the json file was transferred but you're not seeing any video files means that transfer is working correctly, but there were no video files to be transferred.
Ssh in to the Pi and check /mutable/archiveloop.log for errors. You might find all the video files were truncated, in which case restarting the MCU should fix it.

closed 2020-07-01 by marcone