marcone/teslausb #503: Headless setup won't progress...

DZDee, 2020-07-07:
I have tried to get my setup to self setup a few times now. I have flashed the SD card and edit the config file to add only my NAS CISF and my wifi details:
`# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=192.168.1.86
export sharename=Teslacam
export shareuser=Teslacam
export sharepassword=not to day
# export cifs_version="3.0"

# Uncomment the following line and select drive if you want to use an external drive instead of built-in MicroSD card.
# External USB drive is supported only for RPi4.
# export usb_drive=/dev/sda

# Uncomment to disable snapshots. This also disables samba and RecentClips-extension
#export SNAPSHOTS_ENABLED=false

# Recommendations on image sizes:
# Notes:
#   * You can use relative sizes to the available space (e.g., 10%) or explicit sizes (e.g., 16G)
#   * The /backingfiles partition holds both the camera and music disk images.
#   * Please note that the setup script will reserve a fixed 6GB of free space regardless of sd card size.
# When using snapshots:
#    Using images that fill the entire available space will make snapshots inoperable.
#    The system removes snapshots when the available space on the /backingfiles partition falls
#    below the size of the camera image.
#    * camsize should be a fairly small explicit value to leave room for the snapshots.
#      The recommended size is between 8GB and 16GB.
#    * musicsize should also be an explicit size  but should be large enough to hold your music
#      library (obviously).
# When not using snapshots:
#    It is usually easiest to specify an explicit size for your music drive based on your actual
#    library plus an expected growth factory. You can then set the camera size to a portion of the
#    remaining space.
#    * camsize should still be a fairly small explicit value. The size is determined by the number of
#      sentry mode and saved clips you expect to have before the RPi can offload them.
#      * A single SentryClips/SavedClips event is about 1.2GB (MCU2.5) to 1.5GB (FSD) of data.
#      * The hour of RecentClips is about 6.9GB (MCU2.5) to 8.8GB (FSD) of data.
#        The car will delete clips from the RecentClips that are older than an hour.
#    * musicsize can be an explicit or relative size but should be large enough to hold your music
#      library (obviously).
export camsize=16G
#export musicsize=4G

# If you want to automatically copy music from a CIFS share every time
# the Pi connects to wifi, set the following variable. The share is
# assumed to exist on the same server as the archive share. It can
# be the same share as the share used for backing up recordings, but
# the folder needs to be different.
# export musicsharename=your_music_share/music_folder

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID='Carlisto'
export WIFIPASS='not to day'

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true

# uncomment to enable a samba/cifs server that makes the recordings
# available as a network share
#export SAMBA_ENABLED=true
# uncomment to enable guest access to the samba server
#export SAMBA_GUEST=true

# uncomment to enable teslausb to act as a wifi access point with the given
# SSID, so you can access it while on the road.
#export AP_SSID='TESLAUSB WIFI'
# Change this! The setup script will not accept the default 'password'
# Also note that the wifi password must be at least 8 characters.
#export AP_PASS='password'
# IP address is optional. The AP will give its clients IP addresses in the
# x.x.x.100-150 range so make sure you don't put the AP's IP address in
# that range.
#export AP_IP='192.168.66.1'

# uncomment if you want to set the time zone to something other than the default 'BST'
# Can be an actual timezone, or "auto" to attempt automatic timezone detection
# export timezone="America/Los_Angeles"

# By default there is a 20 second delay between connecting to wifi and
# starting the archiving of recorded clips. Uncomment this to change
# the duration of that delay
# export archivedelay=20

# Uncomment if you want to override the default hostname of "teslausb"
# export TESLAUSB_HOSTNAME=teslausb-Model3

# Uncomment if setting up Pushover push notifications
# export pushover_enabled=false
# export pushover_user_key=user_key
# export pushover_app_key=app_key

# Uncomment if setting up Gotify push notifications
# export gotify_enabled=false
# export gotify_domain=https://gotify.domain.com
# export gotify_app_token=put_your_token_here
# export gotify_priority=5

# Uncomment if setting up IFTTT push notifications
# export ifttt_enabled=false
# export ifttt_event_name=put_your_event_name_here
# export ifttt_key=put_your_key_here

# Uncomment if setting up AWS SNS notifications
# export sns_enabled=true
# export aws_region=us-east-1
# export aws_access_key_id=put_your_accesskeyid_here
# export aws_secret_key=put_your_secretkey_here
# export aws_sns_topic_arn=put_your_sns_topicarn_here

# TeslaUSB can optionally use the Tesla API to keep your car awake, so it can
# power the Pi long enough for the archiving process to complete. To enable
# that, please provide your Tesla account email and password below.
# TeslaUSB will only send your credentials to the Tesla API itself.
#
# NOTE:  if you do this, it's highly recommended that you set up 'PIN to Drive'
#   as if the Pi or the car is stolen, the thief has access to your Tesla account 
#   and could change the password, add their own devices as a key, steal the car,
#   etc.  If either is stolen, CHANGE YOUR TESLA PASSWORD IMMEDIATELY.
#
# export tesla_email=joeshmo@gmail.com
# export tesla_password=teslapass
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
# export tesla_vin=5YJ3E1EA4JF000001

# Uncomment if you want to increase the size of the root
# filesystem so there's extra space for installing additional
# software. This only works if the backing store partition hasn't
# been created yet, i.e. during early setup, and requires an extra
# reboot. Sizes can be specified as for example "500M" or "2G"
# export INCREASE_ROOT_SIZE=500M

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev
`

and tried to edit the supplement. The pi flashes for about a minute then goes steady green, but I don't get any of the patterns that are in the documentation. It does generate this log:

`Wed 22 Jan 23:46:34 GMT 2020 : Detecting whether to update wpa_supplicant.conf`

Is my pi bad? or am I doing something wrong?

DZDee, 2020-07-07:
here is my config file.
```
# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=192.168.1.86
export sharename=Teslacam
export shareuser=Teslacam
export sharepassword=not to day
# export cifs_version="3.0"

# Uncomment the following line and select drive if you want to use an external drive instead of built-in MicroSD card.
# External USB drive is supported only for RPi4.
# export usb_drive=/dev/sda

# Uncomment to disable snapshots. This also disables samba and RecentClips-extension
#export SNAPSHOTS_ENABLED=false

# Recommendations on image sizes:
# Notes:
#   * You can use relative sizes to the available space (e.g., 10%) or explicit sizes (e.g., 16G)
#   * The /backingfiles partition holds both the camera and music disk images.
#   * Please note that the setup script will reserve a fixed 6GB of free space regardless of sd card size.
# When using snapshots:
#    Using images that fill the entire available space will make snapshots inoperable.
#    The system removes snapshots when the available space on the /backingfiles partition falls
#    below the size of the camera image.
#    * camsize should be a fairly small explicit value to leave room for the snapshots.
#      The recommended size is between 8GB and 16GB.
#    * musicsize should also be an explicit size  but should be large enough to hold your music
#      library (obviously).
# When not using snapshots:
#    It is usually easiest to specify an explicit size for your music drive based on your actual
#    library plus an expected growth factory. You can then set the camera size to a portion of the
#    remaining space.
#    * camsize should still be a fairly small explicit value. The size is determined by the number of
#      sentry mode and saved clips you expect to have before the RPi can offload them.
#      * A single SentryClips/SavedClips event is about 1.2GB (MCU2.5) to 1.5GB (FSD) of data.
#      * The hour of RecentClips is about 6.9GB (MCU2.5) to 8.8GB (FSD) of data.
#        The car will delete clips from the RecentClips that are older than an hour.
#    * musicsize can be an explicit or relative size but should be large enough to hold your music
#      library (obviously).
export camsize=16G
#export musicsize=4G

# If you want to automatically copy music from a CIFS share every time
# the Pi connects to wifi, set the following variable. The share is
# assumed to exist on the same server as the archive share. It can
# be the same share as the share used for backing up recordings, but
# the folder needs to be different.
# export musicsharename=your_music_share/music_folder

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID='Carlisto'
export WIFIPASS='not to day'

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true

# uncomment to enable a samba/cifs server that makes the recordings
# available as a network share
#export SAMBA_ENABLED=true
# uncomment to enable guest access to the samba server
#export SAMBA_GUEST=true

# uncomment to enable teslausb to act as a wifi access point with the given
# SSID, so you can access it while on the road.
#export AP_SSID='TESLAUSB WIFI'
# Change this! The setup script will not accept the default 'password'
# Also note that the wifi password must be at least 8 characters.
#export AP_PASS='password'
[... 45 lines cut ...]
# NOTE:  if you do this, it's highly recommended that you set up 'PIN to Drive'
#   as if the Pi or the car is stolen, the thief has access to your Tesla account 
#   and could change the password, add their own devices as a key, steal the car,
#   etc.  If either is stolen, CHANGE YOUR TESLA PASSWORD IMMEDIATELY.
#
# export tesla_email=joeshmo@gmail.com
# export tesla_password=teslapass
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
# export tesla_vin=5YJ3E1EA4JF000001

# Uncomment if you want to increase the size of the root
# filesystem so there's extra space for installing additional
# software. This only works if the backing store partition hasn't
# been created yet, i.e. during early setup, and requires an extra
# reboot. Sizes can be specified as for example "500M" or "2G"
# export INCREASE_ROOT_SIZE=500M

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev
```

marcone, 2020-07-07:
```
export sharepassword=not to day
```
if you have spaces in your password, you need to enclose it in quotes, e.g:
```
export sharepassword='not to day'
```

Have you looked at https://github.com/marcone/teslausb/wiki/Troubleshooting ?
Specifically, running `sudo /etc/rc.local` might show more useful info.

DZDee, 2020-07-07:
Yes I looked at the the FAQ. I couldn’t even ssh into the pi over usb so I could run any commands.  I think the pi might be the issue?

marcone, 2020-07-11:
Sounds like maybe it's not getting sufficient power. Try a different cable or a wall charger.

DZDee, 2020-07-14:
so I've tried two pi ZEROs, I must be missing something in my setup, keep getting this...

Wed 22 Jan 23:46:17 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:46:34 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:46:14 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:46:34 GMT 2020 : Detecting whether to update wpa_supplicant.conf

I did also try editing my supplicant with my wifi details. ¯\_(ツ)_/¯

ymiris, 2020-07-14:
DZDee, I just upgraded my pizero to the latest teslaUSB build and I'm getting something similar;

Wed 22 Jan 23:49:47 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:50 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:54 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:38 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:47:38 GMT 2020 : Grabbing main setup file.
Wed 22 Jan 23:47:38 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:41 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:44 GMT 2020 : get_script failed, retrying

Interesting as well, the .conf file disappears once this is tried, so when I look at the SD Card it only has the sample config, but I have verified the normal config is  there till I boot.  Is yours doing anything like that? Maybe mine is just user error, but the previous teslaUSB version worked fine on this pi.

marcone, 2020-07-16:
> so I've tried two pi ZEROs, I must be missing something in my setup, keep getting this...
>
> Wed 22 Jan 23:46:17 GMT 2020 : Detecting whether to update wpa_supplicant.conf
> Wed 22 Jan 23:46:34 GMT 2020 [...]

Did you manually restart the P several times therei, or did it keep rebooting on its own?

resevdog, 2020-07-29:
I had the same problem. Multiple times I flashed the SD, configured the conf files, etc. Always the same process.

But then I switched up my flashing method going back to what I used on my original installation and it worked. Instead of using Etcher, I used Image Writer. I'm not sure why that would make a difference, or if it even did, but for me, it the pi then gave me those lovely stage blinks and then the double blinks as everything sync'd from my NAS.

DZDee, 2020-07-29:
I’ll give this a try and report back to see if it’s a one off.

closed 2020-08-07 by marcone