marcone/teslausb #505: Teslausb Not Recognized in Car, but Recognized in Windows

ninjanwong, 2020-07-08:
I am able to verify that recording to a standard SD card works from the Tesla to a card reader so that USB interface is ok.  I can also verify that TeslaUSB functions on my Win10 machine.  I verified functionality by being able to read/write to the CAM and MUSIC partitions created.  Wireless is also verified working along with the CIFS file share as it pulls music files from my NAS.  When I plug TeslaUSB into the car, the LED blinks as if it's booting, but the camera icon never appears and I can't ssh into the pi.  Sounds like it's not booting, in the car, but is on a regular Win10 machine.  Any thoughts?

marcone, 2020-07-11:
Are you using the same USB cable when connecting to the car?
The fact that you can't ssh in to the Pi might indicate the Pi isn't getting sufficient power, so try a thicker cable, or even two cables at the same time.

ninjanwong, 2020-07-11:
I am using the same USB cable from PC to vehicle.  I tried 2 cables (PWR and USB) and still does not boot.  What is interesting is that I can only connect PWR to the Pi, let it boot, then plug in USB and it shows up.  I'm sure the TeslaUSB image is fairly light weight and turns off all unnecessary applications/features to minimize power, but is there a way to configure a delay between boot and when the drives are active?

ninjanwong, 2020-07-11:
Very similar to https://github.com/marcone/teslausb/issues/9

marcone, 2020-07-15:
> is there a way to configure a delay between boot and when the drives are active?

There is already a delay. It takes about 30-60 seconds before the drives will be activated.

marcone, 2020-07-15:
closing this since this is probably #9, as you said

closed 2020-07-15 by marcone