marcone/teslausb #509: Sentry/Autolock exclude home fails, both turn back on

sherrysj, 2020-07-15:
As above. Removing the rasp pi restores normal function of Sentry/Autolock exclude home option.

marcone, 2020-07-15:
Did you configure teslausb to use the Tesla API to keep the car awake while archiving?

sherrysj, 2020-07-15:
Yes, I set up the Teslausb to use to the Tesla API to keep the car awake while archiving, which it does, but it also might be uncovering a conflict in the way Tesla implements the exclude home preference for walk away lock and sentry mode. 

My expectation is that the car wouldn’t turn on these modes in my garage, but it does when using the Teslausb - configured with the option to keep awake while archiving, and does not when the Teslausb is unplugged.

Thanks for creating the Teslausb - it’s a great project,

-Steve

marcone, 2020-07-16:
When configured to use the Tesla API, teslausb will turn on Sentry mode during archiving to keep the car away. If that interferes with the "exclude home" feature of Sentry mode, then the only remedy is to not have teslausb use the Tesla API. You would have to find some other way to keep the Pi awake during archiving, e.g. by turning on climate, or using a battery.

closed 2020-07-16 by marcone