marcone/teslausb #512: USB drive not detected in car

Remuzen, 2020-07-16:
Hi

I have read on so many forums now but I can't get it working. My setup is a little special, but not that much. I have a wifi router in my car that I use to keep the rpi0w and the car connected all time. The rpi0w makes an OpenVPN tunnel to my home NAS as soon as it boots.

When I boot the rpi0w it is detected with my PC as a Linux mass storage device, but I cant use it for anything, guessing that is because of xfs. When I connect it to the car I cant get it detected by the car (model 3, HW3). 

dmesg after boot:
`[   38.636107] Mass Storage Function, version: 2009/09/11
[   38.636135] LUN: removable file: (no medium)
[   38.636375] LUN: file: /dev/mmcblk0p3
[   38.636389] Number of LUNs=1
[   38.641617] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[   38.641640] g_mass_storage gadget: userspace failed to provide iSerialNumber
[   38.641649] g_mass_storage gadget: g_mass_storage ready
[   38.641668] dwc2 20980000.usb: bound driver g_mass_storage
[   38.658729] dwc2 20980000.usb: new device is high-speed
[   38.670733] dwc2 20980000.usb: new device is high-speed
[   39.047644] dwc2 20980000.usb: new device is high-speed
[   39.293023] dwc2 20980000.usb: new address 20
[   39.307014] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[   40.355335] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.473961] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.474006] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.474139] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   40.474499] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.593897] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.593942] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.594071] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   40.596101] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.714060] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.714104] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.714228] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   40.714485] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.833884] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.833929] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.834058] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   40.838335] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.953891] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.953936] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.954053] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   40.956800] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   40.956833] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   40.956914] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   41.084667] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 1)
[   41.084697] dwc2 20980000.usb: dwc2_hsotg_start_req: ep1 is stalled
[   41.084771] dwc2 20980000.usb: dwc2_hsotg_ep_sethalt(ep 7741ea34 ep1in, 0)
[   42.594029] tun: Universal TUN/TAP device driver, 1.6`

Redo it:
`root@teslausb-Model3:~# modprobe -r g_mass_storage
root@teslausb-Model3:~# modprobe g_mass_storage file=/dev/mmcblk0p3 stall=0
`

dmesg:
`[  141.120812] Mass Storage Function, version: 2009/09/11
[  141.120837] LUN: removable file: (no medium)
[  141.121089] LUN: file: /dev/mmcblk0p3
[  141.121104] Number of LUNs=1
[  141.128388] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[  141.128413] g_mass_storage gadget: userspace failed to provide iSerialNumber
[  141.128421] g_mass_storage gadget: g_mass_storage ready
[  141.128440] dwc2 20980000.usb: bound driver g_mass_storage
[  141.273978] dwc2 20980000.usb: new device is high-speed
[  141.338030] dwc2 20980000.usb: new device is high-speed
[  141.391666] dwc2 20980000.usb: new address 21
[  141.405157] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
`
Can I make it FAT32 just to test it in my PC in a simple way? Any idea of why the car dont want to see the USB drive?

marcone, 2020-07-16:
> I have a wifi router in my car that I use to keep the rpi0w and the car connected all time

Teslausb doesn't work that way. It will only archive when it transitions from "not connected" to "connected"

> Can I make it FAT32 just to test it in my PC in a simple way? 

The drives that are exported to the car/PC are already FAT32.

For troubleshooting, please start at #35

closed 2020-07-16 by marcone

Remuzen, 2020-07-16:
> > I have a wifi router in my car that I use to keep the rpi0w and the car connected all time
> 
> Teslausb doesn't work that way. It will only archive when it transitions from "not connected" to "connected"
> 
This I do with a cronjob that takes wifi offline and then online again.

> > Can I make it FAT32 just to test it in my PC in a simple way?
> 
> The drives that are exported to the car/PC are already FAT32.
> 
> For troubleshooting, please start at #35

Will take a look