marcone/teslausb #514: What are "snapshots"? This doesn't seem to be documented anywhere

craigkovatch, 2020-08-02:
In the sample conf file:
`# Uncomment to disable snapshots. This also disables samba and RecentClips-extension`

But there's no documentation on this github or in the README etc. of what "snapshots" area and why I might want to disable them.

marcone, 2020-08-07:
snapshots are an implementation detail, and disabling them is not something I plan on supporting for much longer, so I'm not going to document them any further.

closed 2020-08-07 by marcone

craigkovatch, 2020-08-07:
@marcone thanks. Until then though, I would kindly suggest adding some additional comments in the config, since it does present itself as a "must-make" decision (whether that's actually true or not) for first-time setup.