marcone/teslausb #517: Consider a slightly larger default root partition size

rmyadsk, 2020-08-09:
The default root partition (/) size is not large enough so survive a Raspbian applications & kernel upgrade. Post-install, this can be trigged by:

<code>/root/bin/remountfs_rw; apt-get update; apt-get upgrade -y; apt autoremove</code>

With the default partition size, the system will run out of space before completing the kernel upgrade.

For a subsequent install I set INCREASE_ROOT_SIZE to 1GB and monitored disk usage during the upgrade process; root bottomed out around 750MB free before finishing at 1.1GB free post-upgrade. Following a reboot the system reports 1.2GB free on root.

marcone, 2020-08-11:
How much free space did you have before you started the upgrade?
I'm seeing about 495MB free on the root partition on a plain teslausb install, which is enough for update/upgrade, with free space dipping down to into the reserved space during upgrade.

marcone, 2020-08-11:
I set up a fresh Pi0 with teslausb, and saw 529 MB free after setup. Running `apt-get update && apt-get upgrade && apt-get autoremove` worked without issues.
Afterwards the root partition had 345 MB free, which became 390 MB after rebooting, and 604 MB after `apt-get clean`

closed 2020-08-11 by marcone

rmyadsk, 2020-08-16:
My mistake - I just re-ran the install and I can confirm the partition does not run out of space. It's possible for a short while the apt repo might have been loaded with some duplicate versions or something...or my SD write wasn't correct. Thanks for looking into it.