marcone/teslausb #518: SNS notifications do not specify hostname or other identifier

rmyadsk, 2020-08-09:
SNS notifications have the subject line "TeslaUSB:" and have the body:

<blockquote>
Archiving 2 event folder(s) with 21 file(s) starting at Sun  9 Aug 15:17:44 EDT 2020
</blockquote>

...and...

<blockquote>
Moved 21 dashcam file(s), failed to copy 0, deleted 0.
</blockquote>

The notification does not specify where it came from, so if the user has two vehicles each with a teslausb, if it not possible to know which one the notification concerns.

Perhaps the notification could key off TESLAUSB_HOSTNAME, or ideally the setup file would incorporate a user-configurable "friendly name", e.g. "Alice's Black Model S" or "Bob's Blue Model 3" which could be used through the application (notification, logging, folder paths, etc.).

marcone, 2020-08-13:
Good idea

kdvlr, 2020-08-31:
I was wondering why this felt like a Deja Vu. This is what I thought when I first created the “TeslaUSB:” prefix - https://github.com/marcone/teslausb/pull/107#discussion_r298806360

Just never got around to it :-(. Let me see if I can add it.

kdvlr, 2020-09-03:
What's your recommendation on the identifier instead of "TeslaUSB" . Would you prefer the hostname or a new variable for that ?

rmyadsk, 2020-09-03:
Hostname (if unique) would be a good default value, with the option for the user to edit the config file to specify a different one - though come to think of it, what is the default hostname of the embedded Linux image? I haven't wiped one of my cards in a while - is it derived from MAC address, for example?

I'm still leaning heavily towards employing a "friendly" aka display name for notifications and other functions. But, global uniqueness for hostname is also beneficial.

kdvlr, 2020-09-03:
Yeah. I am not going to do anything for the "other functions", but the identification for Notification is a valid point. Earlier there was no indicator at all, I added TeslaUSB because I get multiple notifications and I wanted a clear way to identify that. 

Feel free to submit a PR for the other fixes.

kdvlr, 2020-09-04:
Should be closed in PR https://github.com/marcone/teslausb/pull/529. After a bit of back and forth I decided on the hostname because
    - if people have configured different hostnames, they automatically benefit
    - for users who have not configured hostname, the impact is minimal (change from TeslaUSB to teslausb)
    - no real advantage to adding another variable and doesn't benefit existing users

marcone, 2020-09-05:
https://github.com/marcone/teslausb/commit/e3c932ef33b84aaa4a50779d744c3c9969915b9a was merged

closed 2020-09-05 by marcone