marcone/teslausb #520: Ignored vs. default values

rmyadsk, 2020-08-16:
While retesting #517 I noticed this near the bottom:

<code># export INCREASE_ROOT_SIZE=500M</code>

When I read this I thought for a while that by default root was being expanded by a half-GB. But, up at the top we have:

<code>Lines with "#" are comments/ignored.</code>

In my experience this is not the case for many .conf-based packages. For example, look at /etc/ssh/sshd_config on a Linux system - this is a good example of such a config and also features a good explanation:

<code># The strategy used for options in the default sshd_config shipped with OpenSSH is to specify options with their default value where possible, but leave them commented.  Uncommented options override the default value.</code>

Look at line 19 back in our .conf:

<code># export cifs_version="3.0"</code>

So is v3 loaded as a default, or is it left undefined and is SMB left to its own devices?

marcone, 2020-08-16:
The teslausb .conf file is a bash script that sets a bunch of environment variables (hence all the `export` commands, which most other config files don't have). I agree it's perhaps not very consistent in how it specifies enabling/disabling of features and default values, but I think it's decently documented (the specific setting you referred to has 5 lines of comments), and trying to make teslausb's configuration file more like other configuration files is not something I'm interested in spending time on.

closed 2020-08-16 by marcone

rmyadsk, 2020-08-16:
Fair enough. I have had luck changing values post-install so far after setting root r/w and export=NEW_VALUE and rebooting.

marcone, 2020-08-16:
That works for some of the settings, but not for all. Best to get it right the first time 😄