marcone/teslausb #525: TelsaUSB not syncing content of folders with rsync

ronnyandre, 2020-09-01:
I have set up rsync as the preferred method of syncing data from TeslaUSB to my home server. TeslaUSB has successfully connected and created folders on my home server, but there is no content. Even if there exists content in e.g. the SentryCam folder on the TeslaUSB.

I am not sure if I have configured it wrong, but it also seems that the varialble $CAM_MOUNT is not set properly, at least it doesn't print something when echoing it out.

Here are the diagnostics:

```
====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
archive method: rsync
lun0 connected, from file /backingfiles/music_disk.bin
lun1 connected, from file /backingfiles/cam_disk.bin
1 snapshots mounted
====== disk / images ======
Model: SD EB1QT (sd/mmc)
Disk /dev/mmcblk0: 32.0GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2261MB  1988MB  primary  ext4
 3      2261MB  31.9GB  29.6GB  primary  xfs
 4      31.9GB  32.0GB  105MB   primary  ext4
cam disk image has 397 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 17.2GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  17.2GB  17.2GB  primary  fat32        lba
music disk image has 1 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 5768MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  5768MB  5767MB  primary  fat32        lba
====== network ======
ap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.66.1  netmask 255.255.255.0  broadcast 192.168.66.255
        inet6 fe80::ba27:ebff:fe8d:2ccd  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8d:2c:cd  txqueuelen 1000  (Ethernet)
        RX packets 27  bytes 5680 (5.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 181  bytes 37773 (36.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 14056  bytes 873627 (853.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14056  bytes 873627 (853.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fe8d:2ccd  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8d:2c:cd  txqueuelen 1000  (Ethernet)
        RX packets 165988  bytes 12242668 (11.6 MiB)
        RX errors 0  dropped 107  overruns 0  frame 0
        TX packets 332342  bytes 484108707 (461.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

ronnyandre, 2020-09-01:
Further research showed that there are errors in the `archive-rsync-cmd.log` file:

```
2020/09/01 06:50:15 [747] building file list
2020/09/01 06:50:15 [747] .d..t...... 2020-08-28_15-42-46/
2020/09/01 06:53:10 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-34-46-right_repeater.mp4
2020/09/01 06:55:15 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-35-47-back.mp4
2020/09/01 06:56:30 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-35-47-front.mp4
2020/09/01 06:58:14 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-35-47-left_repeater.mp4
2020/09/01 07:00:29 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-35-47-right_repeater.mp4
2020/09/01 07:02:48 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-36-48-back.mp4
2020/09/01 07:04:10 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-36-48-front.mp4
2020/09/01 07:05:03 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-36-48-left_repeater.mp4
2020/09/01 07:05:53 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-36-48-right_repeater.mp4
2020/09/01 07:06:41 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-37-48-back.mp4
2020/09/01 07:08:04 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-37-48-front.mp4
2020/09/01 07:09:15 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-37-48-left_repeater.mp4
2020/09/01 07:11:29 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-37-48-right_repeater.mp4
2020/09/01 07:12:55 [747] <f+++++++++ 2020-08-28_15-42-46/2020-08-28_15-38-49-back.mp4
2020/09/01 07:14:33 [747] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:14:33 [747] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:16:01 [2255] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:16:01 [2255] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:17:18 [3451] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:17:18 [3451] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:19:48 [6167] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:19:48 [6167] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:21:38 [8111] building file list
2020/09/01 07:21:38 [8111] .d..t...... 2020-08-28_15-42-46/
2020/09/01 07:22:40 [8111] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:22:40 [8111] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:23:43 [10275] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:23:43 [10275] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:24:49 [11401] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:24:49 [11401] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:26:43 [13672] building file list
2020/09/01 07:27:45 [13672] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:27:45 [13672] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:29:21 [16507] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:29:21 [16507] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:30:31 [17927] building file list
2020/09/01 07:30:31 [17927] .d..t...... 2020-08-28_15-42-46/
2020/09/01 07:31:34 [17927] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:31:34 [17927] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:33:04 [20732] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:33:04 [20732] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:34:35 [22433] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:34:35 [22433] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:37:02 [25379] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:37:02 [25379] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:38:17 [26838] building file list
2020/09/01 07:38:17 [26838] .d..t...... 2020-08-28_15-42-46/
2020/09/01 07:39:19 [26838] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:39:19 [26838] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:41:02 [29816] building file list
2020/09/01 07:41:03 [29816] .d..t...... 2020-08-28_15-42-46/
2020/09/01 07:42:04 [29816] [sender] io timeout after 60 seconds -- exiting
2020/09/01 07:42:04 [29816] rsync error: timeout in data send/receive (code 30) at io.c(204) [sender=3.1.3]
2020/09/01 07:43:52 [516] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:43:52 [516] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
2020/09/01 07:44:59 [1911] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 07:44:59 [1911] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
[... cut ...]
2020/09/01 10:56:05 [12922] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2020/09/01 10:56:05 [12922] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]
````

Not sure why this fails, but I made the target folder completely writeable, to see if this helped.

Also, trying to invoke the `archive-clips.sh` scripts generates following error:
```
root@teslausb:~/bin# ./archive-clips.sh 
./archive-clips.sh: line 3: log: command not found
```

So multiple issues here it seems...

marcone, 2020-09-01:
> it also seems that the varialble $CAM_MOUNT is not set properly, at least it doesn't print something when echoing it out

"CAM_MOUNT" is a variable that's only set and used with the archiveloop script, so typing `echo $CAM_MOUNT` in the shell isn't going to show anything.

Your diagnostics are incomplete, there should be much more than that. I'm guessing you only copy/pasted whatever was visible in the shell window.

> Also, trying to invoke the archive-clips.sh scripts generates following error:

The archive-clips.sh script is not meant to be run as a standalone command, it is run by archiveloop, which defines the log function and other necessary environment functions and variables.

I suspect the actual problem you're having is poor wifi connectivity. You can see in the rsync log that it takes a really long time to archive each clip, between 50 seconds and 3 minutes per clip, averaging about a minute and a half for the clips shown. Apparently, the connectivity is sometimes so poor that rsync is unable to communicate with your server for an entire minute, at which point it gives up. I suggest unplugging the Pi from the car and connecting it to a power supply, portable charger or computer much closer to your wifi access point, to see if that speeds up wifi and allows rsync to complete without error.

closed 2020-09-01 by marcone

ronnyandre, 2020-09-02:
Strange. I did copy everything from the diagnostics file. But I will try the measured you mentioned.