marcone/teslausb #526: Backup Snapshot?

THX723, 2020-09-02:
@marcone Thanks for the tremendous contribution with this fork!

I was able to gather that Snapshots can only be accessed via Samba as a shared folder. Curious if there are any plans to also include it in the archive loop for server backup?

... for cases in which the RPi may have been stolen from a break-in or the RPi/SD spontaneously combusts in a puff of smoke (tongue-in-cheek). Would be invaluable to have a backup copy of the past recent (and track?) clips on the server, should there be a need weeks/months down the road to reference back for any events of interest.

marcone, 2020-09-03:
archiving of RecentClips is tracked by https://github.com/marcone/teslausb/issues/114

closed 2020-09-03 by marcone

miles267, 2020-12-20:
Appears the method changed in teslausb 20201211 image.  See #114