marcone/teslausb #528: Telegram Notification

DonC52, 2020-09-03:
I am new to Telegram and it is great to have an example of how to create a BOT. however, where do I get the chat id?

marcone, 2020-09-03:
No idea, but I bet @kdvlr does.

DonC52, 2020-09-03:
Did your message above ping @kdvlr that I need an assist?

kdvlr, 2020-09-03:
The easiest way would be to use a bot. 

1. Add [this](https://thereisabotforthat.com/bots/userinfobot) bot to your telegram client
2. Send any message (e.g. “Hi”) and it will respond with an id. 
3. Use that number as the chat_id

Let me know if these instructions work or not and I will update the docs.

DonC52, 2020-09-03:
So it is not the first part (before the : ) of the BOT created. It is the actual user ID of me. Correct?
If so, you may want to add the userinfobot information into the directions.

This seems like a cool way to get notifications for FREE.  It is actually to bad that SMS messages are not that easy. (Every provider seems to do it differently!)

kdvlr, 2020-09-04:
Yes. The bot token is the sender and the chat_id is the recipient. My PR https://github.com/marcone/teslausb/pull/529 addresses this. Have a look and see if it is clear now. 

I think of all the options Telegram is the best; free, secure, and simple to setup.

DonC52, 2020-09-04:
@kdvlr  I found the issue I was having. I am a VERY VERY new Telegram user. (Started TODAY)  I had to start a chat with the BOT (it did a /start) and now I get the messages. I had to find the bot with the search of the t.me name given when the bot was created. I take it that I don't have to keep restarting it as long as the chat is in my chat list correct?
 
Now that I have a little more knowledge of Telegram, it is MUCH easier to setup and is free!  I think if include some of these items I found it will make it MUCH easier for the new Telegram users.  Like me.  ;-)

Will you be using the teslausb_setup_variables.conf file to handle the variables rather than the new file?  (Yea - I have a fork as well.)   Great work!!!

kdvlr, 2020-09-05:
Thanks. Please do  edit the docs to make it easier.

marcone, 2020-09-05:
Docs were updated in https://github.com/marcone/teslausb/commit/e3c932ef33b84aaa4a50779d744c3c9969915b9a and seem pretty self-explanatory now.

closed 2020-09-05 by marcone

DonC52, 2020-09-06:
Do I have to re-install all the system or will a ./setup_teslausb be all that's needed?   I tried that and it seems that when run as archiveloop, it never sends a message to telegram. The send-push-message is not called it seems. I put some log entries at the bottom of the send-push-message file.