marcone/teslausb #530: Music File Deleted and Copied Each Archive Connection

DonC52, 2020-09-04:
It seems that every time I get connected to the archive for file transfer, one music file is always deleted and copied. I have 200 music files and they have not changed. Is this a bug or do I have something going on with one of my music files. (Does it delete files not on the archive?) I will look at the code when I can and add some debug information. Maybe I can find out whats happening.

If its a known bug, I just didn't see it.  I have attached the [diagnostics.txt](https://github.com/marcone/teslausb/files/5172845/diagnostics.txt) file for you to take a look at.

DonC52, 2020-09-04:
It always seems to be the same file. The code does not seem to look at the date/time on the files when deleting (is that true) but just the directory/filename of the file - so not sure what is going on. I will attempt to do some more debugging tomorrow. 

I also noticed a lot more fs errors in the recent files.

marcone, 2020-09-04:
could it be a special character in a file name, which gets translated when the file is copied over? That would make it see one file to be deleted and one file to be copied every time.

DonC52, 2020-09-04:
I don't see any special characters or anything strange about the file name. What I have done is to rename the file (just the file, not the path as other files in that directory are okay) and now it has gone through the force_sync.sh fine TWICE and now it FAILS and does the delete/copy again. The log looks like:
Fri  4 Sep 13:47:09 PDT 2020: Copying music from archive...
Fri  4 Sep 13:47:10 PDT 2020: Deleting Amy Grant/Age To Age/01InALittleWhile.m4a
Fri  4 Sep 13:47:14 PDT 2020: Copied file Amy Grant/Age To Age/01InALittleWhile.m4a
Fri  4 Sep 13:47:16 PDT 2020: Copied 1 music file(s), deleted 1, skipped 199 previously-copied files, encountered 0 copy errors and 0 delete errors.
Fri  4 Sep 13:47:16 PDT 2020: Sending Telegram message for moved files.
Fri  4 Sep 13:47:17 PDT 2020: Trimming free space in /mnt/music, which has 5 extents
Fri  4 Sep 13:47:17 PDT 2020: Trim complete, image now has 5 extents
Fri  4 Sep 13:47:17 PDT 2020: Unmounting /mnt/music...
Fri  4 Sep 13:47:17 PDT 2020: Unmounted /mnt/music.
...
Not sure why it works fine for a while and then fails.  I have renamed the file on the Archive back to the file with spaces and it fails right after. I commented out the touch that is supposed to set the datetime on the file and it worked ONCE without doing the copy again. (Not sure why that would matter!) 

Any ideas what to test out next?

marcone, 2020-09-04:
I'm stumped. The fact that it deletes the file means it thinks the file doesn't exist anymore on the server, but then after it deletes it, it goes and copies the exact same named file from the server, so clearly it *does* exist on the server.

DonC52, 2020-09-05:
It is a total history for sure. I ran the find command and that file is not the first or last in the output. There were others in the same directory on the archive server. I have renamed the file and still it fails a bunch of times after the initial copy. I commented out the touch and it seemed to work for a while but then deleted/copied again.  If i just let the Pi sit there for a good deal of time, it does not fail.  That's why I wondered if something is causing a cache issue with the virtual disk. 

I will keep trying to catch the issue and a solution.  I take it that this is not reported by anyone? (Maybe they don't know - not looking at the logs.)

emerout, 2021-01-02:
I have the same behavior.
Whatever the files are synced, a single file is detected as deleted.
Not a tricky filename : `/mnt/music/Bombino/02.flac`
There is only three files in this unique folder.

marcone, 2021-03-08:
Teslausb now uses rsync to do the transfer even when using a network share. Hopefully that fixes the problem for you. If not, I don't know what to do about it, so closing this issue.

closed 2021-03-08 by marcone