marcone/teslausb #532: New Telegram Update

DonC52, 2020-09-06:
Do I have to re-install all the system or will a ./setup_teslausb be all that's needed?   I tried that and it seems that when run as archiveloop, it never sends a message to telegram. The send-push-message is not called it seems. I put some log entries at the bottom of the send-push-message file.

_Originally posted by @DonC52 in https://github.com/marcone/teslausb/issues/528#issuecomment-687695717_

DonC52, 2020-09-06:
@kdvlr I am now doing a re-install on my Pi.

marcone, 2020-09-06:
I think that updating the .conf file is all that's needed.

kdvlr, 2020-09-06:
You shouldn’t need a re-install.. ./setup-teslausb upgrade should upgrade all the files.

DonC52, 2020-09-06:
@kdvlr I think I found it after a bunch of testing.   The TESLAUSB_HOSTNAME seems not to be defined and the script stops there. Not sure why that is.

DonC52, 2020-09-06:
I wonder if I do a source setup-teslausb  in that file it will work better without messing up stuff?

DonC52, 2020-09-06:
I am just looking at the copy-music.sh to play with. I do not give a host name in the config file for TESLAUSB_HOSTNAME (taking the teslausb default) so maybe that is the issue.  I am sure many people do that.

marcone, 2020-09-06:
@DonC52 copy-music is meant to be called from archiveloop. When called from archiveloop, TESLAUSB_HOSTNAME will be set.

DonC52, 2020-09-06:
@marcone Well - I added the following code to the top of copy-music.sh and it now the notification works.

if [ -z "${TESLAUSB_HOSTNAME+x}" ]
then
  export TESLAUSB_HOSTNAME=$HOSTNAME
fi

Could it be because I do not set it in the config file? (use the default)

DonC52, 2020-09-06:
@marcone  and  @kdvlr   I changed the config file in /root to directly specify a name and it works without the addition of the above code. I specified the same name as the default so I would not have to go through the setup stuff.  I am unsure why when it is exported in archiveloop directly it was not available to this and other scripts. (Failed in the archive-clips.sh as well.

Got ideas?

marcone, 2020-09-06:
Were you calling copy-music.sh and archive-clips.sh directly from the command line? Your earlier comment seemed to imply you were, and the command line shell is not going to see the environment variables that archiveloop exports.

DonC52, 2020-09-06:
NO - I am using the regular system with archiveloop running.  As I noted above, simply putting the default name in the config file and removing the comment so it reads:
...
# Uncomment if you want to override the default hostname of "teslausb"
export TESLAUSB_HOSTNAME=teslausb
...
allows it to work as expected. Like I said, I am unsure why the export of TESLAUSB_HOSTNAME (when it is not set) at the top of archiveloop is not reaching down to the specifically run scripts, I am unsure.  I am wondering if the copy-music.sh is actually in a new bash shell when it is called from archiveloop via its functions.

It may also be that because your do a "source setup-teslausb" and it does not set the TESLAUSB_HOSTNAME because it's not set.  doing a source xxxx is like using the ". xxxx" in the command line. It makes the variables perm.  

Could that be it?  (Would have to look real close at the setup-teslausb to see what it does. Looks like it just doing a read of the config file exporting the variables as it goes.) It looks like TESLAUSB_HOSTNAME is set with the default but not exported. Still looking....

marcone, 2020-09-06:
> I am using the regular system with archiveloop running

Sure, but how are you calling copy-music.sh and archive-clips.sh?

marcone, 2020-09-06:
I would suggest running `setup-teslausb upgrade` one more time to make sure you have all the latest scripts. There should be no need to set TESLAUSB_HOSTNAME in the configuration file, since archiveloop will set it if it is not set.

DonC52, 2020-09-06:
I am doing the debugging within the regular program running. I added some debugging to the source files and found that if it is not specified in the config file, TESLAUSB_HOSTNAME becomes undefined.  The reason is that inside of setup-teslausb, the setting for that variable is NOT exported.  I then modified setup-teslausb to include the export and all is well now.

You do not have to set the variable, unless you want to, in the config file and the TESLA_HOSTNAME is defined everywhere.

Do you want me to bundle it all up and send it your way?  It is only the export in setup-teslausb for that variable.

I will run the update again (did just yesterday) but not sure that will help. Setting it in the archiveloop will not set it in a new bash shell. That is the idea of doing a source setup-teslausb.

marcone, 2020-09-06:
ah ok, I see what you mean now. `setup-teslausb` also sets TESLAUSB_HOSTNAME to a default value, but doesn't export it. So then archiveloop, which sources `setup-teslausb`, sees the value already being set (to the default) so it won't change it, but since it's not exported, other scripts like copy-music.sh and archive-clips.sh won't see it.

Adding an export in setup-teslausb like you did is right, and the code to set and export it in archiveloop can be removed.

DonC52, 2020-09-06:
That's very correct!  Even though you exported it in archiveloop, it does not stay around when another shell is started to do something.

You going to change it or do you want me to (if I can remember how to get it to YOU) .

Been using git a good deal but forgot how to get it to you to implement.  I think it is a pull request?

marcone, 2020-09-06:
> Been using git a good deal but forgot how to get it to you to implement. I think it is a pull request?

Yes. Make the change in your own fork, and then use github to send a pull request for it.

DonC52, 2020-09-06:
BTW... what is the difference between ./setup-teslausb upgrade and ./setup-teslausb selfupdate and ./setup-teslausb

DonC52, 2020-09-06:
@marcone  I think that the pull request looks right.  Take a look...

marcone, 2020-09-06:
> That's very correct! Even though you exported it in archiveloop, it does not stay around when another shell is started to do something.

That's not entirely right. When it's exported in archiveloop, any child process of archiveloop (like copy-music.sh and archive-clips.sh) will see it too. The problem is that archiveloop does NOT export it, because sees the variable is already set (but not exported), and so won't export it with the default value.

marcone, 2020-09-06:
Fixed by https://github.com/marcone/teslausb/pull/533

closed 2020-09-06 by marcone