marcone/teslausb #535: rclone problem

maccis, 2020-09-10:
In this guide: https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md
it says to run:
curl https://rclone.org/install.sh | sudo bash

Problem is that the rclone binary is incompatible with raspberry pi zero w, I changed the binary to another one and then it works.
I dont now what the problem is but I think it would save many users the trouble I had.

Maybe its so simple that you can use this instead.

curl -L https://raw.github.com/pageauc/rclone4pi/master/rclone-install.sh | bash

marcone, 2020-09-10:
What's the actual problem you're seeing?
Both the rclone.org and the rclone4pi install script end up downloading the same binary as far as I can tell, which fails with "Illegal instruction" when run.

marcone, 2020-09-13:
This was an rclone problem, and using the "rclone4pi" script linked above would not have helped as it downloads the same file.
The rclone project has released a fixed binary now, so closing this issue.

closed 2020-09-13 by marcone