marcone/teslausb #536: Raspberry Pi Zero not consistently connecting to Wi-Fi

miles267, 2020-09-10:
Has anyone else noticed their Teslausb raspberry pi device does not consistently connect to their wifi AP upon returning home each trip?  I've seen this over the past few months, but haven't quite been able to pinpoint the root cause.  I've since enabled the teslausb feature to broadcast its own SSID just so that I can tell when it is within range and online.  In these case, I can see that it is in fact home, in range, but simply isn't connecting the the home AP (Unifi AP).  After a later trip, it may connect to wifi AP successfully at which time it will sync the video files as expected.

Wasn't sure whether Teslausb already has some sort of retry variable for how many times or how often it attempts to connect to the wifi SSID specified in the config file?

marcone, 2020-09-11:
There is no retry variable. The wifi connection is entirely managed by the standard linux wpa_supplicant, and teslausb is not involved in that in any way.

You could try connecting to the Pi when it is in this state (since it sounds like you enabled the AP function) and see if there are any driver/hardware errors in dmesg, whether the Pi even sees your access point, etc.

miles267, 2020-09-11:
Thanks @marcone.  I've since re-imaged my teslausb with the latest 20200709 image.  This time, I've not applied any apt-get update or apt-get upgrade commands to apply the latest raspberry pi OS updates.  This seems to have helped.  It leads me to believe there could be an OS update that negatively impacts raspberry pi wifi connectivity that's otherwise not contained in the teslausb base image.

marcone, 2020-09-13:
Closing this because wifi connection problems are outside the scope of the teslausb

closed 2020-09-13 by marcone