marcone/teslausb #538: Model X/ Pi Zero: 2 issues: #1 USB not ready even with 2 cables when Car Starts #2 Music not synching

Rathna-K, 2020-09-16:
My current setup uses a 64GB card with prebuilt image on a Raspberry Pi Zero. 
Connecting the RPi directly to my PC with single USB Cable to the data port of Pi opens the drives Cam and Music correctly. 
But doing the above in my X goes in to a boot loop I think as I don't see the eventual 2 blinks. When I connect another cable to RPi power USB and to an external power, it boots properly and I can see two blinks. But I still have to remove and reconnect the USB Data cable to X to see the drives. 
Anyone ran in to this problem before and how you solved it? I think the USB OTG should come with a delay after boot or maybe do another reboot if Tesla doesn’t recognize the drive? 

For Music, the sync doesn’t happen even though I have files in the shared folder. Any guide or procedure to make this work? 

Thanks,
RK

marcone, 2020-09-19:
There are two separate issues here. One is the power issue. This is essentially https://github.com/marcone/teslausb/issues/9. Some cars apparently don't provide sufficient power to the USB ports to keep the Pi powered. Some of the workarounds in https://github.com/marcone/teslausb/issues/9 may work for you.
The second issue is the music sync. Since you say the Pi works when connected to a PC, I suggest you debug the issue using that: connect to the PC, wait until the cam and music drives show up, then ssh in to the Pi and gather some diagnostics as described in https://github.com/marcone/teslausb/issues/35.

Rathna-K, 2020-09-19:
I dont think its a power issue as the second usb comes from an external power source and not the X’s second USB port. Any other suggestions? Will gather diagnostics for music sync issue.

marcone, 2020-09-19:
If you're using a separate power source you should probably make sure to use a data-only cable on the USB port. You can make one relatively easily by putting a piece of tape over the power contact as shown in this picture: https://github.com/marcone/teslausb/issues/9#issuecomment-589959287

closed 2021-03-08 by marcone