marcone/teslausb #539: RPI Zero W not recognized by Windows / Linux

darox, 2020-09-28:
I have successfully set up a RPI Zero W with teslausb, but it's not recognized by Windows or Linux PC. I have tried several different cables. The log of the diagnostics command is at:
https://pastebin.ubuntu.com/p/r5K84K349w/

Thank you

marcone, 2020-09-28:
It appears the teslausb service is not running. Can you run:
`systemctl list-units teslausb.service`
and
`journalctl -u teslausb.service`
on the Pi?

darox, 2020-09-29:
[image]

Thank you

marcone, 2020-09-29:
You apparently forgot to set the ARCHIVE_SYSTEM variable in the teslausb_setup_variables.conf file. I think this should have caused a failure during setup, but maybe you edited the file afterwards and accidentally commented it out or something?

darox, 2020-09-30:
> You apparently forgot to set the ARCHIVE_SYSTEM variable in the teslausb_setup_variables.conf file. I think this should have caused a failure during setup, but maybe you edited the file afterwards and accidentally commented it out or something?

Ok. Is it absolutely required to setup an archive? I plan to add SSH for archiving, but at a first step, I'd like see if it works locally.

marcone, 2020-09-30:
You can set ARCHIVE_SYSTEM to `none` in that case, but it must be set.

closed 2020-09-30 by marcone

darox, 2020-10-03:
I got it set up with rsync but it returns:
`-- Logs begin at Thu 2019-02-14 10:12:02 GMT, end at Sat 2020-10-03 20:20:12 BST. --
Oct 03 15:17:01 teslausb systemd[1]: Started TeslaUSB archiveloop service.
Oct 03 15:17:02 teslausb bash[183]: Powering off HDMI
Oct 03 15:17:02 teslausb bash[183]: /root/bin/setup-teslausb was sourced, setting setup variables
Oct 03 15:17:04 teslausb bash[183]: umount: /backingfiles/snapshots/snap-000000/snap.bin: not mounted.
Oct 03 15:17:08 teslausb bash[183]: fsck from util-linux 2.33.1
Oct 03 15:17:13 teslausb bash[183]: fsck.fat 4.1 (2017-01-24)
Oct 03 15:17:13 teslausb bash[183]: /dev/loop0p1: 463 files, 1645765/2094976 clusters
Oct 03 15:17:13 teslausb bash[183]: find: ‘/tmp/snapshots/snap-000000’: No such file or directory
Oct 03 15:17:23 teslausb bash[183]: sntp 4.2.8p12@1.3728-o (1)
Oct 03 15:17:23 teslausb bash[183]: Can't open KOD db file /var/lib/sntp/kod for writing: Read-only file system
Oct 03 15:17:23 teslausb bash[183]: 2020-10-03 15:17:23.979896 (+0000) +18063.508621 +/- 12042.339187 time.google.com 216.239.35.8 s1 no-leap
Oct 03 20:18:31 teslausb bash[183]: bash: rsync: command not found
Oct 03 20:18:31 teslausb bash[183]: rsync: connection unexpectedly closed (0 bytes received so far) [sender]
Oct 03 20:18:31 teslausb bash[183]: rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]`

marcone, 2020-10-03:
Do you actually have the rsync command installed? This seems to indicate you don't:

> Oct 03 20:18:31 teslausb bash[183]: bash: rsync: command not found

though I'm puzzled as to why it then appears to print error messages from rsync itself:

> Oct 03 20:18:31 teslausb bash[183]: rsync: connection unexpectedly closed (0 bytes received so far) [sender]
> Oct 03 20:18:31 teslausb bash[183]: rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.3]`

In any case, I would suggest trying to do a manual rsync and see what that does, e.g. something like this on the Pi:
```
rsync -av /root yourserver:/tmp/
```

darox, 2020-10-04:
Got it working. Many thanks!