marcone/teslausb #541: Suggestion: Wi-Fi repeater

MarcFinns, 2020-10-06:
A suggestion that I am independently experimenting, but I believe would be useful to many. 
The Tesla Wi-Fi antenna is notoriously poor. A raspberry pi, especially with a simple external antenna mod, has much better sensitivity. 
As there are many projects to turn the raspi into a Wi-Fi repeater, I believe this would be a great integration to teslaUSB. 
Anyone interested?

mjpcomp, 2020-10-06:
The plan can work the way it's done now - probably just need to enable a software bridge (since the TeslaUSB can both connect to an AP, and become an AP)... However, unless you plan to leave your Pi powered up all the time (external battery that can withstand the heat within the cabin), you'll be losing that WiFi connection shortly after you leave the car, unless you're leaving Sentry mode on 24/7...

MarcFinns, 2020-10-06:
I see your point. however, I guess the car will wake up when trying to upgrade. This would boot the PI. Am i too optimistic?

marcone, 2020-10-19:
There are different levels of wakefulness, and there is no guarantee that the USB ports will be powered when the car is checking for or applying updates.
I'm going to close this bug because I won't be working on this, but you're welcome to submit a PR if you end up implementing it.

closed 2020-10-19 by marcone