marcone/teslausb #542: Potential bug: cannot copy songs with certain characters in the filename

MarcFinns, 2020-10-08:
Hi there, i looked at the archiveloop.log and I see that many music files are not copied.
It appears the issue is related to filenames containing certain characters such as > : " é 

Some examples:
18-Tainted Love-Soft Cell-Made In England: Les Années New Wave.m4a
25-The Final Countdown-Europe-Europe: Definitive Collection.m4a
43-Blue Monday (12" Version)-New Order-Power, Corruption & Lies (Collector's Edition).m4a
03-Just Can't Get Enough-Depeche Mode-The Singles 81>85.m4a

These filenames are allowed in MacOS.
Is this a bug? thanks!

marcone, 2020-10-08:
Sounds like you need to tweak your server configuration.

If the server is configured a certain way, the file names may be mangled. E.g. with default options, my Linux Samba server will present the 4 examples you mentioned as:
```
root@teslausb:~# mount /mnt/musicarchive
root@teslausb:~# ls -lN /mnt/musicarchive/testfiles/
total 4096
-rwxrwxrwx 1 root root 4 Oct  8 09:42 0VMIPP~D.M4A
-rwxrwxrwx 1 root root 4 Oct  8 09:40 1AVA4C~1.M4A
-rwxrwxrwx 1 root root 4 Oct  8 09:40 2SP83U~Q.M4A
-rwxrwxrwx 1 root root 4 Oct  8 09:41 4G8JPD~U.M4A
```
so they'll be copied to the Pi with those names. If the files have the correct embedded metadata, they will still show up in the car with their actual title, album and artist info.

However if I turn off name mangling on the server, they show up with their proper file names:
```
root@teslausb:~# ls -lN /mnt/musicarchive/testfiles/
total 4096
-rwxrwxrwx 1 root root 4 Oct  8 09:42 03-Just Can't Get Enough-Depeche Mode-The Singles 81>85.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:40 18-Tainted Love-Soft Cell-Made In England: Les Années New Wave.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:40 25-The Final Countdown-Europe-Europe: Definitive Collection.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:41 43-Blue Monday (12" Version)-New Order-Power, Corruption & Lies (Collector's Edition).m4a
```
There are other server-side options that would cause these files to be hidden from the client entirely.
Note that the above is a for Linux Samba server. I don't know how you'd go about configuring the same on MacOS.

MarcFinns, 2020-10-08:
in fact i am using a share on a synology NAS.
I use TeslaTunes to esport from my iTunes library into the NAS, and from the NAS to TeslaUSB.
I doubt that mangling can be enabled...

marcone, 2020-10-08:
Some things you might want to check:
- access the NAS from another computer, and see if the files show up there. If not, the problem is with the export. Be sure that you're accessing the NAS via SMB, not AFP.
- ssh in to the pi, and run `sudo mount /mnt/musicarchive; find /mnt/musicarchive` and check if you see the files. If you see mangled file names, that's a server option. If you don't see the files at all, that's a different server option. If you see the files with their full names and they're not being copied, that's a different issue, possibly permissions-related.

In any case, as far as I can tell this is purely a server issue, not a client-side teslausb issue. Your Synology NAS is likely running linux and samba, and you should be able to fix the issue as described here: https://www.uhlme.ch/mangled-names-on-synology-diskstation/

MarcFinns, 2020-10-08:
Hi Marcone, things are as follows:
1. From MacOS connected via SMB i see the full filename including special characters, both on finder and cli
2. mounting the share from the PI and ls -l on terminal, I see the full filename BUT between quotes (sometimes single, sometimes double) and some characters appear escaped

Weird.

marcone, 2020-10-08:
Thanks for checking on that.
The quoting is because newer versions of `ls` escape the names when printed to a terminal.
You can show the unescaped names by using `ls -N`, or `ls | cat`

I just noticed that these filenames actually cause the same issue for me when I disable name mangling: the files show up, but they're not accessible:

```
root@teslausb:/mnt/musicarchive/testfiles# ls -lN
total 4096
-rwxrwxrwx 1 root root 4 Oct  8 09:42 03-Just Can't Get Enough-Depeche Mode-The Singles 81>85.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:40 18-Tainted Love-Soft Cell-Made In England: Les Années New Wave.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:40 25-The Final Countdown-Europe-Europe: Definitive Collection.m4a
-rwxrwxrwx 1 root root 4 Oct  8 09:41 43-Blue Monday (12" Version)-New Order-Power, Corruption & Lies (Collector's Edition).m4a
root@teslausb:/mnt/musicarchive/testfiles# cat *
cat: '03-Just Can'\''t Get Enough-Depeche Mode-The Singles 81>85.m4a': No such file or directory
cat: '18-Tainted Love-Soft Cell-Made In England: Les Années New Wave.m4a': No such file or directory
cat: '25-The Final Countdown-Europe-Europe: Definitive Collection.m4a': No such file or directory
cat: '43-Blue Monday (12" Version)-New Order-Power, Corruption & Lies (Collector'\''s Edition).m4a': No such file or directory
```
The problem is that not all file systems support all characters, and SMB seems particularly weird about handling that (showing the actual file name, but then not allowing access via that same file name).

In the examples you gave, the é is not the problem, but " and : and > are.

In any case, you can fix this by doing the opposite of what I suggested above before I knew what the actual problems was. If you turn ON name mangling in the SMB server (add `mangle names = yes` to the server config), then the names will show up as e.g. 0VMIPP~D.M4A, but they should be accessible by the Pi.

There's also something called [catia](https://www.samba.org/samba/docs/current/man-html/vfs_catia.8.html) that may help, but I haven't tried it.

marcone, 2020-10-08:
I just tried "catia", and it seems to work well. Files shows up with some of the characters translated:
```
root@teslausb:/mnt/musicarchive/testfiles# ls -l
total 4096
-rwxrwxrwx 1 root root 4 Oct  8 09:42 "03-Just Can't Get Enough-Depeche Mode-The Singles 81»85.m4a"
-rwxrwxrwx 1 root root 4 Oct  8 09:40 '18-Tainted Love-Soft Cell-Made In England÷ Les Années New Wave.m4a'
-rwxrwxrwx 1 root root 4 Oct  8 09:40 '25-The Final Countdown-Europe-Europe÷ Definitive Collection.m4a'
-rwxrwxrwx 1 root root 4 Oct  8 09:41 '43-Blue Monday (12¨ Version)-New Order-Power, Corruption & Lies (Collector'\''s Edition).m4a'
```
but that's still much more readable than 8.3 DOS names, and the files are accessible with these names.

MarcFinns, 2020-10-08:
I gave it a try. 
- mangling should be the default
- in smb.conf is not disabled
- i made YES explicit
alas, no luck / no visible difference.

marcone, 2020-10-08:
Did you restart the NAS (or the SMB service on the NAS) ?

MarcFinns, 2020-10-09:
i did stop and start the service

MarcFinns, 2020-10-09:
FYI - i did a mass rename on the NAS in the end.  Easier.
This is my legacy library, before spotify & co. Hence it is not going to change.
Once copied over to teslaUSB, can I block the synchronisation?
thanks again, keep up the great work!

marcone, 2020-10-09:
> Once copied over to teslaUSB, can I block the synchronisation?

The whole point of the music synchronization is so that if you add music to the server, it'll end up on your Pi. If you don't want that, you could have just copied the files to the Pi's music drive directly. Probably would have been a lot faster, too.

In any case, you can disable automatic music sync by deleting the /mnt/musicarchive folder, or by deleting the /root/bin/copy-music.sh script.

MarcFinns, 2020-10-09:
You are right... but I don’t want to setup a samba server just for a music library which is mostly static...
Thanks again

marcone, 2020-11-18:
Reopening as a reminder to myself to look into "catia" some more

marcone, 2020-12-20:
Added a note to the README about name mangling and catia

closed 2020-12-20 by marcone