marcone/teslausb #545: help with data recovery

dmartinelli99, 2020-10-22:
I am not sure if this is the best place to ask this question but figured I would try.

I have been using teslausb for a couple months now and love it. It's been working great.

However, I recently got rear ended and the footage did not upload. The electrical system was damaged so it might not have been written to disk.

I believe there is a way to recover the last hour which might have the crash, or at least show the car tailgating us, but have not been able to figure out how to do this.

This is what I tried:
copied snap.bin from the pi to my Mac
used dd to image a ssd drive (dd if=snap.bin of=/dev/disk5s1)
I tried to use dd to image another flash drive with the snap.bin file since it shows up as a FAT32 filesystem. This did not work.  

Any help or pointers would be greatly appreciated.

Dominic

marcone, 2020-10-22:
First off, if you configured Samba on your Pi you can just access the Pi via Samba and see all the available recordings.

If you didn't configure Samba, then the next step would be to look at the cam_disk.bin file itself. That's the actual "CAM" disk that the car sees, and should have the most recent clips. Just plug the Pi in to a PC (or Mac) and the "CAM" drive should come up once the Pi is fully booted. You can then look at the TeslaCam/RecentClips folder in there to find the latest recordings. You could even run data recovery software on that drive to recover files that might have already been deleted.

If that fails, you could look at the snapshot images, but those are only taken every hour, so the data in those might not contain what you want. Depending on your setup, there may be multiple "snap.bin" files, so be sure to look at the most recent one. The snapshots are in the /backingfiles/snapshots folder on the Pi, in sequentially numbered folders.

marcone, 2020-10-22:
Oh also:

> used dd to image a ssd drive (dd if=snap.bin of=/dev/disk5s1)

I think you need to use `dd if=snap.bin of=/dev/disk5` since the snap.bin file is itself a fully formed disk image with a partition table, and so it should be written to the raw drive, not to the first partition of the drive.

dmartinelli99, 2020-10-22:
Thank you for the quick response. It is much appreciated. 

I was able to mount the CAM directory on my Mac but the recent clips directory is empty, so I will attempt to do a data recovery.

closed 2020-10-23 by marcone