marcone/teslausb #547: cannot get anything to work

crackers8199, 2020-10-31:
i follow the instructions to set this thing up, and after a few minutes to a few hours when i give up and pull the card and put it back in the computer, i end up with the log file filled with the following error...

get_script failed, retrying

i've tried everything. i've tried making sure the timezone is correct, i've tried adding the export certificates line...i'm about ready to throw this thing out the window.

everyone on all the forums says this is so easy...what am i doing wrong?

marcone, 2020-10-31:
Can you post the full log file?

crackers8199, 2020-10-31:
> Can you post the full log file?

absolutely. here's the full file from my last attempt before i gave up last night and posted this issue...every other one has been the same thing, but with a different number of the repeating log lines (depending how long i gave it before giving up):

```
Wed 22 Jan 23:46:44 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:46:44 GMT 2020 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Wed 22 Jan 23:46:44 GMT 2020 : Rebooting...
Wed 22 Jan 23:46:45 GMT 2020 : Grabbing main setup file.
Wed 22 Jan 23:47:37 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:47:38 GMT 2020 : Grabbing main setup file.
Wed 22 Jan 23:47:38 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:41 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:44 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:48 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:51 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:54 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:47:57 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:00 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:04 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:07 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:10 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:13 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:17 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:20 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:23 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:26 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:29 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:33 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:36 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:39 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:42 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:45 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:49 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:52 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:55 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:48:58 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:01 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:05 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:08 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:11 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:14 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:17 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:21 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:24 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:27 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:30 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:33 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:37 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:40 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:43 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:46 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:50 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:53 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:56 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:49:59 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:02 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:06 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:09 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:12 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:15 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:18 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:22 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:25 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:28 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:31 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:34 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:38 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:41 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:50:44 GMT 2020 : get_script failed, retrying
[... 147 lines cut ...]
Wed 22 Jan 23:58:39 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:42 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:46 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:49 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:52 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:55 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:58:58 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:02 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:05 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:08 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:11 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:14 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:18 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:21 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:24 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:27 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:30 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:34 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:37 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:40 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:43 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:47 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:50 GMT 2020 : get_script failed, retrying
Wed 22 Jan 23:59:53 GMT 2020 : get_script failed, retrying
```

marcone, 2020-11-01:
Can you ssh in to the Pi?

crackers8199, 2020-11-01:
nope. it doesn't show up on the network.

marcone, 2020-11-01:
You'll need to resolve that first. Double check that you specified the correct SSID and password for your wifi. If your SSID or password contain special characters, you will need to escape them as described in https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md

crackers8199, 2020-11-01:
i've tried. i have done all of that, which is why i'm getting so frustrated...

my wifi password does have special characters in it, but no single quotes...so i should be able to just wrap the whole thing in single quotes and not have to escape any of the characters. right?

i've tried this at least five times. the chances that i've typed my SSID and wifi password wrong all five of those times is almost zero...

marcone, 2020-11-01:
> my wifi password does have special characters in it, but no single quotes...so i should be able to just wrap the whole thing in single quotes and not have to escape any of the characters. right?

Yes, with the exception of backslash, which should still be escaped.

Also note that the Raspberry Pi Zero W only supports 2.4 GHz wifi, so if you have a separate SSID for your 2.4GHz network be sure to use that.

crackers8199, 2020-11-01:
bingo, that's probably the issue. i was attempting to use my 5 ghz network. thank you!

i'll change that to the 2.4 SSID and hopefully that'll resolve my issue.

crackers8199, 2020-11-01:
yep, that was the problem. thanks for the assist!

closed 2020-11-01 by crackers8199

jwaresolutions, 2020-11-02:
I have only numbers and letters in mine and I can't get it to work either.

crackers8199, 2020-11-02:
in my case it was that i was trying to use a 5 ghz network. once i changed to my 2.4 ghz ssid, it started working.