marcone/teslausb #548: usb drive not working when connected to computer

crackers8199, 2020-11-02:
when i connect the pi to my macbook pro, i get a quick double flash on the LED every second or so and the usb drive never shows up on the computer. log stops here every time:

```
Sun  1 Nov 18:35:33 PST 2020: Archiving...
Sun  1 Nov 18:35:34 PST 2020: Ensuring cam archive is mounted...
Sun  1 Nov 18:35:34 PST 2020: Mounting /mnt/archive...
Sun  1 Nov 18:35:36 PST 2020: Mounted /mnt/archive.
Sun  1 Nov 18:35:36 PST 2020: Ensured cam archive is mounted.
Sun  1 Nov 18:35:36 PST 2020: Ensuring music archive is mounted...
Sun  1 Nov 18:35:36 PST 2020: Mounting /mnt/musicarchive...
Sun  1 Nov 18:35:37 PST 2020: Mounted /mnt/musicarchive.
Sun  1 Nov 18:35:37 PST 2020: Ensured music archive is mounted.
Sun  1 Nov 18:35:37 PST 2020: Checking saved folder count...
Sun  1 Nov 18:35:37 PST 2020: Ensuring cam file is mounted...
Sun  1 Nov 18:35:37 PST 2020: Mounting /mnt/cam...
Sun  1 Nov 18:35:37 PST 2020: Mounted /mnt/cam.
Sun  1 Nov 18:35:37 PST 2020: Ensured cam file is mounted.
Sun  1 Nov 18:35:42 PST 2020: There are 0 event folder(s) with 0 file(s) and 0 track mode file(s) to move. 0 short recording(s) will be skipped.
Sun  1 Nov 18:35:42 PST 2020: Trimming free space in /mnt/cam, which has 4 extents
Sun  1 Nov 18:35:43 PST 2020: Trim complete, image now has 4 extents
Sun  1 Nov 18:35:43 PST 2020: Unmounting /mnt/cam...
Sun  1 Nov 18:35:43 PST 2020: Unmounted /mnt/cam.
Sun  1 Nov 18:35:43 PST 2020: Finished archiving.
Sun  1 Nov 18:35:43 PST 2020: Copying music...
Sun  1 Nov 18:35:43 PST 2020: Starting music sync...
Sun  1 Nov 18:35:43 PST 2020: Ensuring music backing file is mounted...
Sun  1 Nov 18:35:43 PST 2020: Mounting /mnt/music...
Sun  1 Nov 18:35:43 PST 2020: Mounted /mnt/music.
Sun  1 Nov 18:35:43 PST 2020: Ensured music drive is mounted.
Sun  1 Nov 18:35:44 PST 2020: Copying music from archive...
Sun  1 Nov 18:35:46 PST 2020: Copied 0 music file(s), deleted 0, skipped 4 previously-copied files, encountered 0 copy errors and 0 delete errors.
Sun  1 Nov 18:35:46 PST 2020: Trimming free space in /mnt/music, which has 1 extents
Sun  1 Nov 18:35:46 PST 2020: Trim complete, image now has 1 extents
Sun  1 Nov 18:35:46 PST 2020: Unmounting /mnt/music...
Sun  1 Nov 18:35:46 PST 2020: Unmounted /mnt/music.
Sun  1 Nov 18:35:46 PST 2020: Finished copying music.
Sun  1 Nov 18:35:46 PST 2020: unmounting /mnt/archive
Sun  1 Nov 18:35:47 PST 2020: Connecting usb to host...
Sun  1 Nov 18:35:47 PST 2020: Connected usb to host.
Sun  1 Nov 18:35:47 PST 2020: Waiting for archive to be unreachable...
```

also, interesting that it said from the beginning that 4 music files were copied...because i only have two files in my music share as a test before putting a larger number of files there. that's part of why i was trying to connect this to the computer in the first place, to see what files exactly it copied over...

marcone, 2020-11-02:
Check your USB cable. Most of the time the problem is using a charge-only cable

closed 2020-11-05 by marcone

crackers8199, 2020-11-09:
@marcone that fixed it, but there are only two files on the music drive (as expected). why is the script telling me there were four files transferred?