marcone/teslausb #549: Enable SMB Share Folders

1activegeek, 2020-11-02:
Perhaps I'm just missing something somewhere, but based on my numerous attempts at different formatting I don't believe it's possible to specify a folder within a share to store the content. For example, I have a share named "backup" on "server1". My config would include:
```
export archiveserver=server1
export sharename=backup
```
This will work as expected. But if I'd like to store the files in a sub-folder on that share "teslausb" for example, it will not take this as part of the specification:
```
export archiveserver=server1
export sharename=backup/teslausb
```
I've attempted 
```
backup/teslausb
backup\/teslausb
'backup/teslausb'
'backup\/teslausb'
"backup/teslausb"
"backup\/teslausb"
```
all with the same result.

The kicker is that the mount test, shows the correct mount:
```
verify-and-configure-archive: mount -t cifs '//server1/backup/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,,'
```

Yet the resulting `/etc/fstab` output shows:
```
//server1/backup /mnt/archive cifs noauto,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,, 0
```

I saw chatter in some issues that look like others were setting this up in this sort of fashion, but I'm not able to find anything that seems to indicate how to configure it contrary to what I've seen so far. Any guidance is greatly appreciated.

MarcFinns, 2020-11-02:
I have the following and works flawlessly. What's your server?
export SHARE_NAME=TeslaShare/CamFootage

1activegeek, 2020-11-02:
I use the sample and the export was actually `sharename`, so I used that. But seeing yours I also tried changing the export to match as `SHARE_NAME`, but that didn't work either. 

This is what my config file has now. 
```
#export sharename='backup/teslausb'
export SHARE_NAME=backup/teslausb
```

@MarcFinns are you using `ARCHIVE_SYSTEM=cifs` or are you using rsync?

MarcFinns, 2020-11-02:
export ARCHIVE_SYSTEM=cifs

marcone, 2020-11-02:
specifying the path after the share name is the correct way to do this, e.g. as in your example:
```
export archiveserver=server1
export sharename=backup/teslausb
```

what happens when you ssh in to the Pi and run `sudo mount /mnt/archive` ?

1activegeek, 2020-11-02:
> teslausb won't update fstab

so if I change the .conf file to a different share later, or a different server, it won't update fstab? That seems a bit of a miss. Happy to make a manual change though if it means it'll work. Thanks for the nudge in the right direction, definitely wasted a bunch of time trying to figure out what I was doing wrong - at least I know I wasn't doing something wrong! :)

1activegeek, 2020-11-02:
Yup, that did it. I'll close this out now. Thanks for getting this fixed. Appreciate the app!!

closed 2020-11-02 by 1activegeek