marcone/teslausb #551: Going to use a pi3 and buy an Endurance microSD card 128Gb. The card will hold the OS and the video  files, correct?

Klagio, 2020-11-06:
Or do you recommend to use a regular, small, microSD for the OS and then connect a SSD as storage in the USB port of the pi3?

marcone, 2020-11-06:
[You can't use a Pi3.](https://github.com/marcone/teslausb#hardware)

closed 2020-11-06 by marcone

Klagio, 2020-11-06:
Ok I saw in the instruction too late. Ok I will use a PI ZERo or PI4,  he card will hold the OS and the video files?
do you recommend to use a regular, small, microSD for the OS and then connect a SSD as storage in the USB port of the pi4 (I guess pi0 has no USB?

marcone, 2020-11-06:
If you only use an sd card, it will hold both the OS and the recordings.
On a Pi0 that's the only option.
On a Pi4 you also have the option of using the sd card only for the OS with a USB drive for the recordings, or to use a USB for both the OS and the recordings (so in the latter case you wouldn't need an SD card at all)