marcone/teslausb #552: Samba share no longer matches the Cam contents

Jughead135, 2020-11-06:
Not sure what to make of this.  I have used Samba to map a network drive on my PC as a quick & dirty way to monitor the Pi's connectivity.

Until recently, the `\\smb_share\pi` "drive" matched (exactly) what I would see on the USB drive, `CAM:\TeslaCam`.  Now, however, I see the same folders (`RecentClips`, `SavedClips`, `SentryClips`); but, they mostly have different dates on them.  Few if any of the subfolders match what the USB drive is presenting; the meta files (`thumb.png` etc.) show up as folders; and all folders are empty.

I assume I broke something in a recent archive server change--but I can't figure out what.  Any ideas?

marcone, 2020-11-06:
Sorry, I don't understand what you're saying. How do you "monitor the Pi's connectivity" using a network drive on your PC?

Jughead135, 2020-11-06:
Probably the wrong term.  If the Pi is running normally & talking to my network, it shows up as a shared drive in Windows.  A single click tells me if it's there.  Can also view the contents of the folders and have a feel for how well (or not) the archiving is going at a glance.  Saves the trouble of ssh'ing in to the Pi or physically going to the car.

As I said, "quick and dirty" monitor--NOT at all any sort of technical activity.

In any case:  Shouldn't the contents of the Samba shared TeslaCam be identical to the TeslaCam folder the car sees on the "USB drive"?  I'd thought it was literally the same files/folders--and, until recently, it appeared to be so....

marcone, 2020-11-06:
Still not sure what exactly you did. Teslausb supports exporting the recordings via Samba. Are you using that, and simply mounting that share semi-permanently on your PC, or did you do something else?

Re:
> Shouldn't the contents of the Samba shared TeslaCam be identical to the TeslaCam folder the car sees on the "USB drive"?
that all depends on what exactly you did. Teslausb only mounts /backingfiles/cam_disk.bin (the disk image that holds the "usb drive" that the car sees) when it is archiving. When it it is not archiving, you'll just see the empty mountpoint at /mnt/cam. Teslausb also deletes the SavedClips and SentryClips recordings after archiving them to your selected archiving destination, and the car deletes any RecentClips recordings that are older than one hour. So it may very well be be normal that you're only seeing empty folders.

Jughead135, 2020-11-06:
> Still not sure what exactly you did.

Sorry, I'm not explaining well.  I setup TeslaUSB per instructions, all working fine.  I enabled Samba in `/root/teslausb_setup_variables.conf` and `/etc/samba/smb.conf`.  I mapped a network drive in Win10 to `\\<NAS_IP>\TeslaCam`.  At that point, I have a shared drive visible in Windows (read-only).

Until recently, that shared drive seemed to look just like the (non-archived portion) of the TeslaCam--i.e., the directory on the "USB drive" that TeslaUSB is presenting to the car.  That's no longer the case.  The change probably (though I can't be sure) happened right around the time I was struggling with changing the archive server from my hard drive to an NAS (which is now solved).

> Re:
>
>
> that all depends on what exactly you did. Teslausb only mounts /backingfiles/cam_disk.bin (the disk image that holds the "usb drive" that the car sees) when it is archiving. When it it is not [...]

OK, I may well have completely misunderstood what I'm looking at.  Is the `/TeslaCam` directory that's shared via Samba _not_ the same directory (of the same name) that the car sees when the "USB" is plugged in?  Or when I connect the RPi via USB directly to my PC, and (after it boots) I see a removable "drive" called `CAM`, with the same `TeslaCam` directory as seen in the car?

Based on that (flawed?) understanding, and what I (thought) I've seen before, I would expect the last hour or so of `RecentClips` to be there (just like they are in the car or plugged into the PC), even if the `SavedClips` & `SentryClips` files have been deleted following archival.

I feel like I'm missing some key (and probably simple) point here.  Your patience is amazing, thank you!

marcone, 2020-11-06:
> Is the /TeslaCam directory that's shared via Samba not the same directory (of the same name) that the car sees when the "USB" is plugged in?  

It is not. The directory visible as teslausb's samba share is /mutable/TeslaCam (or /backingfiles/TeslaCam if you didn't install/update recently). On the Pi, this is a collection of links into the snapshots that teslausb takes about every hour.
So in most cases, you should see *more* files there than in the "CAM" drive that the car sees, not fewer.

> Or when I connect the RPi via USB directly to my PC, and (after it boots) I see a removable "drive" called CAM, with the same TeslaCam directory as seen in the car?

Yes, when connecting via USB, that *is* the same as what the car sees.

Jughead135, 2020-11-06:
OK, I definitely misunderstood what I was looking at.  [Got it, when the Pi is plugged in and TeslaUSB is up & running, the host device "sees" a pair of USB drives--car or PC, or anything else with a USB port; that part I understand.]

Still puzzled over what I _do_ see, though.  See screen shot.  Note old dates on those folders.  **EVERY** subfolder is empty (at least viewing it via the Samba share on my Win10 PC).  The thumb.png & event.json metadatas are "folders" (also "empty").  The two sub-folders in the `SavedClips` directory correspond to actual folders on the NAS archive (with archived video in them); I don't believe any of those other folders have a twin in the archive.

I certainly do not see "more" files than what's on the "CAM" drive; I see **none**.

PS:  As far as I can tell, everything else is working.  Just this weirdness on the Samba share....

[image]

marcone, 2020-11-06:
ssh in to the Pi and check the content of /mutable/TeslaCam
If that doesn't exist check /backingfiles/TeslaCam

Jughead135, 2020-11-06:
> ssh in to the Pi and check the content of /mutable/TeslaCam
> If that doesn't exist check /backingfiles/TeslaCam

Whoops, typing at the same time.  Doing that now....

marcone, 2020-11-06:
BTW the event.json and thumb.png showing up as folders is a bug I still need to fix

Jughead135, 2020-11-06:
Unsurprisingly, I think you've hit the problem here.  I appear to have _both_ `/mutable/TeslaCam` AND `/backingfiles/TeslaCam` on the Pi.

The `/mutable/TeslaCam` appears to be the current/correct stuff (based on dates):
```
root@MimiPi:/mutable/TeslaCam# ls -lah
total 5.0K
drwxr-xr-x  5 root root 1.0K Nov  3 15:47 .
drwxr-xr-x  7 root root 1.0K Nov  6 14:10 ..
drwxr-xr-x 26 root root 1.0K Nov  6 12:26 RecentClips
drwxr-xr-x  4 root root 1.0K Nov  3 15:47 SavedClips
drwxr-xr-x  3 root root 1.0K Nov  3 15:47 SentryClips
```

```
root@MimiPi:/mutable/TeslaCam# cd RecentClips
root@MimiPi:/mutable/TeslaCam/RecentClips# ls -lah
total 453K
drwxr-xr-x 26 root root 1.0K Nov  6 12:26 .
drwxr-xr-x  5 root root 1.0K Nov  3 15:47 ..
drwxr-xr-x  2 root root 1.0K Nov  4 11:42 201416~2.M
drwxr-xr-x  2 root root 1.0K Nov  4 09:17 201DE7~4.M
drwxr-xr-x  2 root root 1.0K Nov  6 12:26 201FE6~8.M
drwxr-xr-x  2 root root 4.0K Nov  4 18:17 2020-10-16
drwxr-xr-x  2 root root 4.0K Nov  4 18:35 2020-10-30
drwxr-xr-x  2 root root 123K Nov  6 12:08 2020-11-03
drwxr-xr-x  2 root root 196K Nov  6 15:31 2020-11-04
drwxr-xr-x  2 root root  26K Nov  5 20:18 2020-11-05
drwxr-xr-x  2 root root  73K Nov  6 16:33 2020-11-06
drwxr-xr-x  2 root root 1.0K Nov  4 10:58 203389~3.M
drwxr-xr-x  2 root root 1.0K Nov  5 10:21 2039D9~6.M
drwxr-xr-x  2 root root 1.0K Nov  5 10:21 2050B0~3.M
drwxr-xr-x  2 root root 1.0K Nov  4 09:18 20643A~5.M
drwxr-xr-x  2 root root 1.0K Nov  5 18:49 208981~1.M
drwxrwxrwx  2 root root 1.0K Nov  4 17:30 209923~7.M
drwxr-xr-x  2 root root 1.0K Nov  5 18:29 20AC03~5.M
drwxr-xr-x  2 root root 1.0K Nov  5 20:18 20AE38~8.M
drwxr-xr-x  2 root root 1.0K Nov  6 12:26 20D258~4.M
drwxr-xr-x  2 root root 1.0K Nov  5 18:37 20DDC7~2.M
drwxr-xr-x  2 root root 1.0K Nov  6 12:27 20E00A~7.M
drwxrwxrwx  2 root root 1.0K Nov  6 12:23 20E4D4~8.M
drwxr-xr-x  2 root root 1.0K Nov  4 14:37 20F086~1.M
drwxr-xr-x  2 root root 1.0K Nov  4 18:35 event.json
drwxr-xr-x  2 root root 1.0K Nov  6 16:33 thumb.png

```

The `/backingfiles/TeslaCam` matches (more or less) what I'm seeing in the Samba share (the dates don't match):
```
root@MimiPi:/mutable/TeslaCam/RecentClips# cd /backingfiles/TeslaCam
root@MimiPi:/backingfiles/TeslaCam# ls -lah
total 0
drwxr-xr-x  5 root root  62 Apr 25  2020 .
drwxr-xr-x  4 root root  81 Apr 25  2020 ..
drwxr-xr-x 11 root root 167 Nov  3 15:23 RecentClips
drwxr-xr-x  4 root root  60 Oct 30 16:24 SavedClips
drwxr-xr-x  4 root root  60 Nov  3 12:25 SentryClips
```

```
root@MimiPi:/backingfiles/TeslaCam# cd RecentClips
root@MimiPi:/backingfiles/TeslaCam/RecentClips# ls -lah
total 180K
drwxr-xr-x 11 root root  167 Nov  3 15:23 .
drwxr-xr-x  5 root root   62 Apr 25  2020 ..
drwxr-xr-x  2 root root 4.0K Nov  3 15:23 2020-10-16
drwxr-xr-x  2 root root 4.0K Nov  3 15:23 2020-10-30
drwxr-xr-x  2 root root  96K Nov  2 18:16 2020-11-02
drwxr-xr-x  2 root root  32K Nov  3 15:23 2020-11-03
drwxr-xr-x  2 root root   26 Nov  3 15:23 20643A~5.M
drwxr-xr-x  2 root root   26 Nov  2 18:16 207446~3.M
drwxr-xr-x  2 root root   26 Nov  2 18:17 20D491~7.M
drwxr-xr-x  2 root root   24 Nov  3 15:23 event.json
drwxr-xr-x  2 root root   23 Nov  3 15:23 thumb.png
```

=======================================================
So, I've apparently got the Samba share pointing at the wrong directory.  I found this in `/etc/samba/smb.conf`:
```
[TeslaCam]
   read only = yes
   locking = no
   path = /backingfiles/TeslaCam
   guest ok = no
   create mask = 0775
   veto files = /._*/.DS_Store/
   delete veto files = yes
   root preexec = /root/bin/make_snapshot.sh
```

Is this as simple as changing `path` to `path = /mutable/TeslaCam`?  I did that, and all seems to be working now.  Thank you!!, though I've no idea how the two got separated/duplicated....

I'm curious how _some_ of the data (all empty folders) in the `/backingfiles/TeslaCam` version have recent dates associated, but that's a minor curiosity if all is well with the new version....

marcone, 2020-11-06:
I [changed](https://github.com/marcone/teslausb/commit/f388bf4458f4207502d9ba94db3a13fd09c9c416) teslausb recently to use /mutable/TeslaCam instead of /backingfiles/TeslaCam.
That change should have moved the folder and updated smb.conf. It looks like you did an upgrade that then got interrupted partway through.

marcone, 2020-11-06:
(you should probably run `setup-teslausb upgrade` again just to make sure everything is up to date)

closed 2020-11-06 by marcone

Jughead135, 2020-11-06:
> (you should probably run `setup-teslausb upgrade` again just to make sure everything is up to date)

Done.

It did _not_ delete the `/backingfiles/TeslaCam` instance--no changes that I can see there.  As I say, doesn't appear to be "hurting" anything, though.

The `setup-teslausb upgrade` ended slightly curiously.  Do I have a problem with r/w access?  (I did run `/root/bin/remountfs_rw` immediately beforehand.)  These are the last two lines:

```
Fixing the modules-load parameter in /boot/cmdline.txt...
/tmp/setup-teslausb: line 248: /boot/cmdline.txt: Read-only file system
```

Anyway, as I say, at this point I'm just feeding my curiosity.  As always, thanks for your help & for providing such a stellar project.  Thankfully you're using your talents for good instead of evil....

marcone, 2020-11-07:
The "Read-only file system" thing is weird. It could be there was an error on /boot that caused it to be remounted read-only again.