marcone/teslausb #555: Unable to connect to my home wifi

aaronac8, 2020-11-16:
I am using a 256 GB endurance SSD in a raspberry pi zero w and have configured my Wifi with my home network password.  My home wifi SSID is 'Farfel2G' and my password consists of only letters (capital and lower case) and numbers with no spaces.  Yet my raspberry pi does not obtain an IP on my home network.  Please help!

marcone, 2020-11-16:
Is your "Farfel2G" network a 2.4GHz network? Because the Pi0 does not support 5GHz wifi.

aaronac8, 2020-11-16:
Yes

Best regards,
Aaron

marcone, 2020-11-16:
Sorry, I don't know then. Could be a defective Pi, or a configuration problem on the Pi, a configuration problem on your wifi access point, etc. You could put the sd card back in a PC and see if the "teslausb-headless-setup.log" file shows anything of interest.
Alternatively, if you connect the Pi to a PC, you may be able to get a USB network connection to it, which would then allow you to log in to the Pi and hopefully diagnose the wifi problem that way.

aaronac8, 2020-11-17:
Please see my attached teslausb-headless-setup log below.

marcone, 2020-11-17:
> Please see my attached teslausb-headless-setup log below

It looks like you forgot to attach it?

aaronac8, 2020-11-17:
Wed 22 Jan 23:46:43 GMT 2020 : Detecting whether to update
wpa_supplicant.conf
Wed 22 Jan 23:46:15 GMT 2020 : Detecting whether to update
wpa_supplicant.conf

marcone, 2020-11-17:
There should be more than that. A first-install log should start with:
```
Sun 22 Sep 22:55:44 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:44 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 22 Sep 22:55:45 BST 2019 : Rebooting...
```
My guess is there's a typo in the config file, and setup errors out while parsing it.

marcone, 2020-11-17:
> My guess is there's a typo in the config file

Actually no, it would have parsed the config file already by the time it logs "Detecting whether to update wpa_supplicant.conf"
The problem appears to be that either SSID or WIFIPASS aren't set.

aaronac8, 2020-11-17:
My SSID is 'Farfel2G' and my WIFIPASS is a set of capital and lower case
letters along with numbers (no spaces).  Do I need to delete the sample
files from the /boot folder?

I also just get a solid green light on the raspberry pi.

Best regards,
Aaron

marcone, 2020-11-17:
> My SSID is 'Farfel2G' and my WIFIPASS is a set of capital and lower case letters along with numbers (no spaces). Do I need to delete the sample files from the /boot folder?

No, but you do need to rename the sample file to "teslausb_setup_variables.conf"

aaronac8, 2020-11-17:
Yes, I did do that.

marcone, 2020-11-17:
Then it appears there's something wrong with the config file. The log you're seeing can only be explained by one or both of the wifi variables not being set. Make sure your config file contains something like:
```
export SSID=Farfel2G
export WIFIPASS=YoUrPaSsWoRd1
```
Make sure there are no spaces around `=` sign.

aaronac8, 2020-11-17:
So I am not supposed to include ' sign in my wifi settings?  For example, I
am literally writing:

export SSID='Farfel2G'
export WIFIPASS='YoUrPaSsWoRd1'

Thank you for all your help.

Best regards,
Aaron

aaronac8, 2020-11-17:
I have attached my config file with generic passwords entered.  Please let
me know what I am doing wrong.  Thank you.

#####################################################################
# SAMPLE CONFIGURATION FILE FOR TESLAUSB Pi Setup
#
# Example config file for teslausb_setup. Lines with "#" are comments/ignored.
# Remove the "#" before "export" to activate a line. Be sure to rename this file
# to "teslausb_setup_variables.conf" and place it in the "boot" folder of your
# SD card.
# Please check the latest version of this file at:
# https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample
#
######################################################################

# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=192.168.1.103
export SHARE_NAME=TeslaCam
export SHARE_USER=Tesla
export SHARE_PASSWORD=PaSSword1
# export cifs_version="3.0"

# Uncomment the following line and select drive if you want to use an external drive instead of built-in MicroSD card.
# External USB drive is supported only for RPi4.
# export USB_DRIVE=/dev/sda

# Uncomment to disable snapshots. This also disables samba and RecentClips-extension
#export SNAPSHOTS_ENABLED=false

# Recommendations on image sizes:
# Notes:
#   * You can use relative sizes to the available space (e.g., 10%) or explicit sizes (e.g., 16G)
#   * The /backingfiles partition holds both the camera and music disk images.
#   * Please note that the setup script will reserve a fixed 6GB of free space regardless of sd card size.
# When using snapshots:
#    Using images that fill the entire available space will make snapshots inoperable.
#    The system removes snapshots when the available space on the /backingfiles partition falls
#    below the size of the camera image.
#    * CAM_SIZE should be a fairly small explicit value to leave room for the snapshots.
#      The recommended size is between 8GB and 16GB.
#    * MUSIC_SIZE should also be an explicit size  but should be large enough to hold your music
#      library (obviously).
# When not using snapshots:
#    It is usually easiest to specify an explicit size for your music drive based on your actual
#    library plus an expected growth factory. You can then set the camera size to a portion of the
#    remaining space.
#    * CAM_SIZE should still be a fairly small explicit value. The size is determined by the number of
#      sentry mode and saved clips you expect to have before the RPi can offload them.
#      * A single SentryClips/SavedClips event is about 1.2GB (MCU2.5) to 1.5GB (FSD) of data.
#      * The hour of RecentClips is about 6.9GB (MCU2.5) to 8.8GB (FSD) of data.
#        The car will delete clips from the RecentClips that are older than an hour.
#    * MUSIC_SIZE can be an explicit or relative size but should be large enough to hold your music
#      library (obviously).
export CAM_SIZE=16G
export MUSIC_SIZE=100G

# If you want to automatically copy music from a CIFS share every time
# the Pi connects to wifi, set the following variable. The share is
# assumed to exist on the same server as the archive share. It can
# be the same share as the share used for backing up recordings, but
# the folder needs to be different.
export musicsharename=TeslaMusic

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID=Farfel2G
export WIFIPASS=PaSSWord1

# If doing a headless (i.e. automatic) setup
export HEADLESS_SETUP=true
[... 83 lines cut ...]
# You can provider either email+password, or accesstoken+refreshtoken:
# export TESLA_EMAIL=joeshmo@gmail.com
# export TESLA_PASSWORD=teslapass
#
# export TESLA_ACCESS_TOKEN=123456789abcdef0123456789abcdef0123456789abcdef0123456789abcdef0
# export TESLA_REFRESH_TOKEN=abcdef0123456789abcdef0123456789abcdef0123456789abcdef0123456789
#
# If you have more than one Tesla vehicle on the account, you must also provide
# the VIN for the vehicle for which this TeslaUSB install will be used, so that
# TeslaUSB can keep the correct vehicle awake.
# export TESLA_VIN=5YJ3E1EA4JF000001

# Uncomment if you want to increase the size of the root
# filesystem so there's extra space for installing additional
# software. This only works if the backing store partition hasn't
# been created yet, i.e. during early setup, and requires an extra
# reboot. Sizes can be specified as for example "500M" or "2G"
# export INCREASE_ROOT_SIZE=500M

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev

aaronac8, 2020-11-17:
Can I use "teslausb-20200709" or should I use an older version?

mobiletru, 2020-11-17:
Don't know if anyone has mentioned this it should be included in the instructions with bold letters  ..if you have hide extensions of known file types with windows selected when you go to save the .conf file on the boot drive it will actually save it as a .sample file  ..i included a pic of the problem 

[image]

aaronac8, 2020-11-17:
So do I just delete the sample file?  Because I have done that and still
have the same problem with not connecting to wifi.  Also, since I have the
conf file there, should having the sample file matter?

I can use the latest version teslausb-20200709?

Thanks.

Best regards,
Aaron

marcone, 2020-11-17:
You don't have to delete the sample file, it will be ignored.
The latest release image is fine.
The configuration file you included above seems OK to me.

Since you're using Windows, you may have a working USB network connection to the Pi if you let it boot while connected to the Windows PC (make sure you're using a proper USB cable, not a charge-only cable), and that it's connected to the USB port on the Pi, not the power port). That should let you ssh in to the Pi (`ssh pi@teslausb.local`) and see what's going on.

aaronac8, 2020-11-17:
Do you think it could be a defective wifi module on the raspberry pi zero W?

Best regards,
Aaron

marcone, 2020-11-17:
I don't think it's gotten to the point where it actually tries to configure wifi yet. At least the content of the setup log doesn't show it doing that.
For a normal setup, the log should show something like:
```
Sun 22 Sep 22:55:44 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sun 22 Sep 22:55:44 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sun 22 Sep 22:55:45 BST 2019 : Rebooting...
```
Since your setup log apparently only contained that first line (per your comment above), it never actually tried to configure wifi.
It's also possible you have a flaky sd card, and setup encountered an I/O error. I'd try flashing it again.

aaronac8, 2020-11-18:
I got it to work by changing the file to .conf (by exposing the file
types).  But now I am stuck at this step with a rapidly flashing green
light.  What do I do now?

pi@teslausb:/boot $ tail -f teslausb-headless-setup.log
Tue 17 Nov 20:28:03 EST 2020 : make-root-fs-readonly: Disabling unnecessary
service...
Tue 17 Nov 20:28:10 EST 2020 : make-root-fs-readonly: Removing unwanted
packages...
Tue 17 Nov 20:31:14 EST 2020 : make-root-fs-readonly: Installing ntp and
busybox-syslogd...
Tue 17 Nov 20:32:41 EST 2020 : make-root-fs-readonly: Configuring system...
Tue 17 Nov 20:32:41 EST 2020 : make-root-fs-readonly: Mounting the mutable
partition...
Tue 17 Nov 20:32:41 EST 2020 : make-root-fs-readonly: Mounted.
Tue 17 Nov 20:32:41 EST 2020 : make-root-fs-readonly: Moving fake-hwclock
data
Tue 17 Nov 20:32:41 EST 2020 : make-root-fs-readonly: done
Tue 17 Nov 20:32:41 EST 2020 : Skipping package upgrade.
Tue 17 Nov 20:32:48 EST 2020 : All done.

marcone, 2020-11-18:
> I got it to work by changing the file to .conf (by exposing the file types).

Oh, I thought you said you already did that in your comment above (the one with the screenshot), but that it still didn't work afterwards.

> But now I am stuck at this step with a rapidly flashing green light. What do I do now?

The rapid flashing light should be temporary, and indicates it is archiving or copying music. You can ssh in to the Pi and see what it's doing by looking at /mutable/archiveloop.log

aaronac8, 2020-11-18:
When can I unplug it and put it in the car?

pi@teslausb:/boot $ /boot/teslausb-headless-setup.log
/boot/teslausb-headless-setup.log: line 1: Wed: command not found
/boot/teslausb-headless-setup.log: line 2: Wed: command not found
/boot/teslausb-headless-setup.log: line 3: Wed: command not found
/boot/teslausb-headless-setup.log: line 4: Wed: command not found
/boot/teslausb-headless-setup.log: line 5: Wed: command not found
/boot/teslausb-headless-setup.log: line 6: Wed: command not found
/boot/teslausb-headless-setup.log: line 7: Wed: command not found
/boot/teslausb-headless-setup.log: line 8: Wed: command not found
/boot/teslausb-headless-setup.log: line 9: Wed: command not found
/boot/teslausb-headless-setup.log: line 10: Wed: command not found
/boot/teslausb-headless-setup.log: line 11: Wed: command not found
/boot/teslausb-headless-setup.log: line 12: Wed: command not found
/boot/teslausb-headless-setup.log: line 13: Wed: command not found
/boot/teslausb-headless-setup.log: line 14: Wed: command not found
/boot/teslausb-headless-setup.log: line 15: Wed: command not found
/boot/teslausb-headless-setup.log: line 16: Wed: command not found
/boot/teslausb-headless-setup.log: line 17: Wed: command not found
/boot/teslausb-headless-setup.log: line 18: Wed: command not found
/boot/teslausb-headless-setup.log: line 19: Wed: command not found
/boot/teslausb-headless-setup.log: line 20: Wed: command not found
/boot/teslausb-headless-setup.log: line 21: Wed: command not found
/boot/teslausb-headless-setup.log: line 22: Tue: command not found
/boot/teslausb-headless-setup.log: line 23: Tue: command not found
/boot/teslausb-headless-setup.log: line 24: Tue: command not found
/boot/teslausb-headless-setup.log: line 25: Tue: command not found
/boot/teslausb-headless-setup.log: line 26: Tue: command not found
/boot/teslausb-headless-setup.log: line 27: Tue: command not found
/boot/teslausb-headless-setup.log: line 28: Tue: command not found
/boot/teslausb-headless-setup.log: line 29: Tue: command not found
/boot/teslausb-headless-setup.log: line 30: Tue: command not found
/boot/teslausb-headless-setup.log: line 31: Tue: command not found
/boot/teslausb-headless-setup.log: line 32: Tue: command not found
/boot/teslausb-headless-setup.log: line 33: Tue: command not found
/boot/teslausb-headless-setup.log: line 34: Tue: command not found
/boot/teslausb-headless-setup.log: line 35: Tue: command not found
/boot/teslausb-headless-setup.log: line 36: Tue: command not found
/boot/teslausb-headless-setup.log: line 37: Tue: command not found
/boot/teslausb-headless-setup.log: line 38: Tue: command not found
/boot/teslausb-headless-setup.log: line 39: Tue: command not found
/boot/teslausb-headless-setup.log: line 40: Tue: command not found
/boot/teslausb-headless-setup.log: line 41: Tue: command not found
/boot/teslausb-headless-setup.log: line 42: Tue: command not found
/boot/teslausb-headless-setup.log: line 43: Tue: command not found
/boot/teslausb-headless-setup.log: line 44: Tue: command not found
/boot/teslausb-headless-setup.log: line 45: Tue: command not found
/boot/teslausb-headless-setup.log: line 46: Tue: command not found
/boot/teslausb-headless-setup.log: line 47: Tue: command not found
/boot/teslausb-headless-setup.log: line 48: Tue: command not found
/boot/teslausb-headless-setup.log: line 49: Tue: command not found
/boot/teslausb-headless-setup.log: line 50: Tue: command not found
/boot/teslausb-headless-setup.log: line 51: syntax error near unexpected
token `
        ('
/boot/teslausb-headless-setup.log: line 51: `Tue 17 Nov 20:25:51 EST 2020 :
crea
      te-backingfiles-partition: Modifying partition table for mutable
(writable) part
                ition for script usage...'

marcone, 2020-11-18:
> pi@teslausb:/boot $ /boot/teslausb-headless-setup.log

That's a log file, not an executable script. Use `tail` like you did in your previous comment.
Setup is done (the setup log ended with "All done"), so you can plug it in to the car whenever you want. I'd plug it in to the PC first to make sure the "CAM" drive shows up though.

aaronac8, 2020-11-18:
I see the following:
pi@teslausb:/boot $ tail -f /mutable/archiveloop.log
Tue 17 Nov 20:33:43 EST 2020: Unmounting /mnt/cam...
Tue 17 Nov 20:33:43 EST 2020: Unmounted /mnt/cam.
Tue 17 Nov 20:33:43 EST 2020: Finished archiving.
Tue 17 Nov 20:33:43 EST 2020: Copying music...
Tue 17 Nov 20:33:43 EST 2020: Starting music sync...
Tue 17 Nov 20:33:43 EST 2020: Ensuring music backing file is mounted...
Tue 17 Nov 20:33:44 EST 2020: Mounting /mnt/music...
Tue 17 Nov 20:33:44 EST 2020: Mounted /mnt/music.
Tue 17 Nov 20:33:44 EST 2020: Ensured music drive is mounted.
Tue 17 Nov 20:33:44 EST 2020: Copying music from archive...

How do I change the pi password?  Can it have letters and numbers on it? Is
it case sensitive?

My error was in not knowing that the file was saving as a .conf.txt
file....the .txt was hidden by windows.  When I unclicked the hide file
type box I saw the error........THANK YOU SO MUCH FOR ALL YOUR HELP!!!

aaronac8, 2020-11-18:
trying to change my password and getting:
pi@teslausb:~ $ passwd
Changing password for pi.
Current password:
New password:
Retype new password:
You must choose a longer password
New password:
Retype new password:
passwd: Authentication token manipulation error
passwd: password unchanged

marcone, 2020-11-18:
The error is because the root file system is read-only, so it can't write the updated password. You need to:
```
sudo -i
bin/remountfs_rw
passwd pi
```

aaronac8, 2020-11-18:
Also, I plugged in the pi into my PC and it does not recognize the drive.
What should I do?

aaronac8, 2020-11-18:
It is working now......have Music drive and Cam drive!!!!!

mobiletru, 2020-11-18:
The hide extensions of known file types is a default windows 10 setting ..can this be added to the instructions ?  windows will not let you save that .conf file when that is checked. but the file will show up as .conf ..   it would solve 50% of the problems on here

aaronac8, 2020-11-18:
I have 20 GB of music to sync, but not all of my music is synching.  I set
aside 100 GB for music in the conf file (out of 256 GB SD card) and 32 GB
for Cam folder.   How do I rectify this?

marcone, 2020-11-18:
How did you determine that not all the music is syncing? Are there specific files missing? Specific folders? Does it just stop syncing after a while? Are there any errors in /mutable/archiveloop.log?

aaronac8, 2020-11-18:
I am getting:
pi@teslausb:/boot $ tail -f /mutable/archiveloop.log
Wed 18 Nov 06:15:29 EST 2020: Running fsck on /backingfiles/cam_disk.bin...
Wed 18 Nov 06:15:29 EST 2020: | fsck from util-linux 2.33.1
Wed 18 Nov 06:15:31 EST 2020: | fsck.fat 4.1 (2017-01-24)
Wed 18 Nov 06:15:31 EST 2020: | /dev/loop1p1: 7 files, 6/2094976 clusters
Wed 18 Nov 06:15:31 EST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Wed 18 Nov 06:15:31 EST 2020: Running fsck on
/backingfiles/music_disk.bin...
Wed 18 Nov 06:15:31 EST 2020: | fsck from util-linux 2.33.1
Wed 18 Nov 06:15:31 EST 2020: | fsck.ext2: No such file or directory while
trying to open /dev/loop1
Wed 18 Nov 06:15:31 EST 2020: | /dev/loop0p1
Wed 18 Nov 06:15:32 EST 2020: | Possibly non-existent device?

aaronac8, 2020-11-18:
Also, I can only ssh to the pi with pi@192.168.1.200
I can not use pi@teslausb.local.....how do I fix this?

marcone, 2020-11-18:
Looks like your music drive image doesn't contain a partition table.
What does `fdisk -l /backingfiles/music_disk.bin` say?

As for "teslausb.local" not working, I think that's a function of the computer that you're running `ssh` on, not of the Pi itself. It needs to support mDNS/zeroconf/bonjour.

aaronac8, 2020-11-18:
"What does *fdisk -l /backingfiles/music_disk.bin* say?"

How do I find the answer to that question?

marcone, 2020-11-18:
ssh in to the Pi and type `fdisk -l /backingfiles/music_disk.bin` :)

aaronac8, 2020-11-18:
pi@teslausb:~ $ fdisk -l /backingfiles/music_disk.bin
Disk /backingfiles/music_disk.bin: 100 GiB, 107374182400 bytes, 209715200
sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5acfca29

Device                        Boot Start       End   Sectors  Size Id Type
/backingfiles/music_disk.bin1       2048 209715199 209713152  100G  c W95
FAT32 (LBA)

marcone, 2020-11-18:
The fdisk output looks fine.
The rapid flashing means the Pi is syncing music or archiving videos. The drives will not be exposed to the car or the PC while it's doing that, and given that you said you have 20GB of music to sync, it's going to take quite a while for it to sync that.

aaronac8, 2020-11-19:
My music did not all download....It stopped downloading when it reached
folders starting with the letter "F"
Here is the log:
pi@teslausb:/boot $ tail -f /mutable/archiveloop.log
Wed 18 Nov 19:57:34 EST 2020: Unmounting /mnt/cam...
Wed 18 Nov 19:57:34 EST 2020: Unmounted /mnt/cam.
Wed 18 Nov 19:57:35 EST 2020: Finished archiving.
Wed 18 Nov 19:57:35 EST 2020: Copying music...
Wed 18 Nov 19:57:35 EST 2020: Starting music sync...
Wed 18 Nov 19:57:35 EST 2020: Ensuring music backing file is mounted...
Wed 18 Nov 19:57:35 EST 2020: Mounting /mnt/music...
Wed 18 Nov 19:57:35 EST 2020: Mounted /mnt/music.
Wed 18 Nov 19:57:35 EST 2020: Ensured music drive is mounted.
Wed 18 Nov 19:57:35 EST 2020: Copying music from archive...
Wed 18 Nov 20:52:27 EST 2020: connection dead, killing copy-music
Wed 18 Nov 20:52:27 EST 2020: Trimming free space in /mnt/music, which has
36 extents
Wed 18 Nov 20:52:28 EST 2020: Trim complete, image now has 36 extents
Wed 18 Nov 20:52:28 EST 2020: Unmounting /mnt/music...
umount: /mnt/music: target is busy.
Wed 18 Nov 20:52:28 EST 2020: Failed to unmount /mnt/music, trying lazy
unmount.
Wed 18 Nov 20:52:29 EST 2020: lazily unmounted /mnt/music
Wed 18 Nov 20:52:29 EST 2020: Finished copying music.
Wed 18 Nov 20:52:29 EST 2020: unmounting /mnt/archive
Wed 18 Nov 20:52:29 EST 2020: Connecting usb to host...
Wed 18 Nov 20:52:29 EST 2020: Connected usb to host.
Wed 18 Nov 20:52:29 EST 2020: Waiting for archive to be unreachable...
Wed 18 Nov 20:52:29 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:30 EST 2020: Retrying...
Wed 18 Nov 20:52:30 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:31 EST 2020: Retrying...
Wed 18 Nov 20:52:31 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:32 EST 2020: Retrying...
Wed 18 Nov 20:52:33 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:34 EST 2020: Retrying...
Wed 18 Nov 20:52:34 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:35 EST 2020: Retrying...
Wed 18 Nov 20:52:35 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:36 EST 2020: Retrying...
Wed 18 Nov 20:52:36 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:37 EST 2020: Retrying...
Wed 18 Nov 20:52:37 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:38 EST 2020: Retrying...
Wed 18 Nov 20:53:44 EST 2020: unmounting /mnt/musicarchive
Wed 18 Nov 20:55:15 EST 2020: waiting up to 90 seconds for idle interval
Wed 18 Nov 20:56:57 EST 2020: couldn't determine idle interval
Wed 18 Nov 20:56:58 EST 2020: taking snapshot of cam disk in
/backingfiles/snapshots/snap-000002
Wed 18 Nov 20:57:01 EST 2020: took snapshot
Wed 18 Nov 20:57:05 EST 2020: comparing new snapshot with
/backingfiles/snapshots/snap-000001/snap.bin
Wed 18 Nov 20:57:05 EST 2020: new snapshot is identical to previous one,
discarding

aaronac8, 2020-11-19:
Can I manually copy my music to the raspberry pi's MUSIC folder?

marcone, 2020-11-19:
Yes, you can copy the music manually.

closed 2020-11-19 by marcone