marcone/teslausb #556: RSYNC configuation: cannot see USB drive when connected to model y (pi zero)

suryadevarauday, 2020-11-17:
i am trying to use 16GB for pi zero on model y. here is my configuration and its not even detecting the Dashcam logo 

`cat /root/teslausb_setup_variables.conf 
export ARCHIVE_SYSTEM=rsync
export RSYNC_USER=sampletsla
export RSYNC_SERVER=192.168.0.110
export RSYNC_PATH=/home/sampletsla/tesla

export CAM_SIZE=100%

export SSID='wifi-link'
export WIFIPASS='mywifipass'
export HEADLESS_SETUP=true
export SAMBA_ENABLED=true

export AP_SSID='evwifi'
export AP_PASS='Tesla@2020'
export AP_IP='192.168.66.1'
export TIME_ZONE="auto"
`
logs:
`==============================================
Tue 17 Nov 18:29:40 EST 2020: Starting archiveloop at 18.58 seconds uptime...
Tue 17 Nov 18:29:41 EST 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000001
Tue 17 Nov 18:29:47 EST 2020: took snapshot
Tue 17 Nov 18:29:47 EST 2020: comparing new snapshot with /backingfiles/snapshots/snap-000000/snap.bin
Tue 17 Nov 18:29:47 EST 2020: new snapshot is identical to previous one, discarding
Tue 17 Nov 18:29:47 EST 2020: Running fsck on /backingfiles/cam_disk.bin...
Tue 17 Nov 18:29:48 EST 2020: | fsck from util-linux 2.33.1
Tue 17 Nov 18:29:54 EST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 17 Nov 18:29:54 EST 2020: | /dev/loop0p1: 4 files, 3/1594696 clusters
Tue 17 Nov 18:29:54 EST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Tue 17 Nov 18:29:55 EST 2020: Connecting usb to host...
Tue 17 Nov 18:29:55 EST 2020: Connected usb to host.
Tue 17 Nov 18:29:55 EST 2020: Waiting for archive to be reachable...
Tue 17 Nov 18:29:56 EST 2020: Archive is reachable.
Tue 17 Nov 18:29:56 EST 2020: Trying to set time...
Tue 17 Nov 18:29:57 EST 2020: sntp failed, retrying...
Tue 17 Nov 18:30:30 EST 2020: Time adjusted by 33.819641 seconds after 2.800000 seconds
Tue 17 Nov 18:30:51 EST 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000001
Tue 17 Nov 18:30:53 EST 2020: took snapshot
Tue 17 Nov 18:30:56 EST 2020: comparing new snapshot with /backingfiles/snapshots/snap-000000/snap.bin
Tue 17 Nov 18:30:56 EST 2020: new snapshot is identical to previous one, discarding
Tue 17 Nov 18:30:56 EST 2020: Disconnecting usb from host...
Tue 17 Nov 18:30:56 EST 2020: Disconnected usb from host.
Tue 17 Nov 18:30:56 EST 2020: Running fsck on /backingfiles/cam_disk.bin...
Tue 17 Nov 18:30:57 EST 2020: | fsck from util-linux 2.33.1
Tue 17 Nov 18:30:58 EST 2020: | fsck.fat 4.1 (2017-01-24)
Tue 17 Nov 18:30:58 EST 2020: | /dev/loop0p1: 4 files, 3/1594696 clusters
Tue 17 Nov 18:30:58 EST 2020: Finished fsck on /backingfiles/cam_disk.bin.
Tue 17 Nov 18:30:58 EST 2020: Archiving...
Tue 17 Nov 18:30:58 EST 2020: Checking saved folder count...
Tue 17 Nov 18:30:58 EST 2020: Ensuring cam file is mounted...
Tue 17 Nov 18:30:58 EST 2020: Mounting /mnt/cam...
Tue 17 Nov 18:30:58 EST 2020: Mounted /mnt/cam.
Tue 17 Nov 18:30:58 EST 2020: Ensured cam file is mounted.
Tue 17 Nov 18:31:01 EST 2020: There are 0 event folder(s) with 0 file(s) and 0 track mode file(s) to move. 0 short recording(s) will be skipped.
Tue 17 Nov 18:31:01 EST 2020: Trimming free space in /mnt/cam, which has 4 extents
Tue 17 Nov 18:31:02 EST 2020: Trim complete, image now has 4 extents
Tue 17 Nov 18:31:02 EST 2020: Unmounting /mnt/cam...
Tue 17 Nov 18:31:02 EST 2020: Unmounted /mnt/cam.
Tue 17 Nov 18:31:02 EST 2020: Finished archiving.
Tue 17 Nov 18:31:02 EST 2020: Music archive not configured or unreachable
Tue 17 Nov 18:31:02 EST 2020: Connecting usb to host...
Tue 17 Nov 18:31:02 EST 2020: Connected usb to host.
Tue 17 Nov 18:31:02 EST 2020: Waiting for archive to be unreachable...
Tue 17 Nov 18:31:43 EST 2020: Sleeping before retry...
Tue 17 Nov 18:31:44 EST 2020: Retrying...
Tue 17 Nov 18:37:52 EST 2020: Sleeping before retry...
Tue 17 Nov 18:37:53 EST 2020: Retrying...
Tue 17 Nov 18:38:06 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:07 EST 2020: Retrying...
Tue 17 Nov 18:38:08 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:09 EST 2020: Retrying...
Tue 17 Nov 18:38:13 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:14 EST 2020: Retrying...
Tue 17 Nov 18:38:19 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:20 EST 2020: Retrying...
Tue 17 Nov 18:38:23 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:24 EST 2020: Retrying...
Tue 17 Nov 18:38:25 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:26 EST 2020: Retrying...
Tue 17 Nov 18:38:33 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:34 EST 2020: Retrying...
Tue 17 Nov 18:38:42 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:43 EST 2020: Retrying...
Tue 17 Nov 18:38:46 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:47 EST 2020: Retrying...
Tue 17 Nov 18:38:48 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:49 EST 2020: Retrying...
Tue 17 Nov 18:38:57 EST 2020: Sleeping before retry...
Tue 17 Nov 18:38:58 EST 2020: Retrying...`

suryadevarauday, 2020-11-17:
Installation looks of i can ssh into that and also the i can login to ssh@sampletsla@192.168.0.110 without password and upload and delete files with key, permissions ok

marcone, 2020-11-17:
Please start here for troubleshooting: https://github.com/marcone/teslausb/issues/35

suryadevarauday, 2020-11-18:
Attaching the logs here Marcone https://gist.github.com/suryadevarauday/b1d9fd81a97d20ee43b8e0173d13dd5a#file-gistfile1-txt

marcone, 2020-11-18:
Your log has the classic signs of using a charge-only cable instead of a full USB cable.
(or alternatively you plugged it in to the PWR port on the Pi instead of the USB port)

suryadevarauday, 2020-11-18:
tried moving the cable from far most port and also connected to the model y, right side USB, still no luck.

suryadevarauday, 2020-11-18:
And when does this rsync runs, can we set to run with always_sync? Does it run automatically when the new file is there? 
ALWAYS_SYNC=TRUE

marcone, 2020-11-18:
There's no such option. https://github.com/marcone/teslausb/wiki/FAQ#q-i-manually-copied-some-files-to-the-cam-drive-but-theyre-not-being-archived

suryadevarauday, 2020-11-19:
1) I was able to fix the problem by changing the cable, i have a similar micro USB cable connected to raspberry Pi Zero middle port and connected to car and it can detect. Not sure how to identify which cable to choose. (they both looks same). 
2) Any way you are planning to enable rsync feature for copying files from RecentClips? Or any recommendations?

marcone, 2020-11-19:
Glad you got it working.
Yes, I plan to add RecentClips at some point. Right now it's not super useful IMO because it only contains recordings from the last hour.

closed 2020-11-19 by marcone