marcone/teslausb #557: Music files stop updating

aaronac8, 2020-11-19:
Here is my log:
My music did not all download....It stopped downloading when it reached folders starting with the letter "F"
Here is the log:

pi@teslausb:/boot $ tail -f /mutable/archiveloop.log
Wed 18 Nov 19:57:34 EST 2020: Unmounting /mnt/cam...
Wed 18 Nov 19:57:34 EST 2020: Unmounted /mnt/cam.
Wed 18 Nov 19:57:35 EST 2020: Finished archiving.
Wed 18 Nov 19:57:35 EST 2020: Copying music...
Wed 18 Nov 19:57:35 EST 2020: Starting music sync...
Wed 18 Nov 19:57:35 EST 2020: Ensuring music backing file is mounted...
Wed 18 Nov 19:57:35 EST 2020: Mounting /mnt/music...
Wed 18 Nov 19:57:35 EST 2020: Mounted /mnt/music.
Wed 18 Nov 19:57:35 EST 2020: Ensured music drive is mounted.
Wed 18 Nov 19:57:35 EST 2020: Copying music from archive...
Wed 18 Nov 20:52:27 EST 2020: connection dead, killing copy-music
Wed 18 Nov 20:52:27 EST 2020: Trimming free space in /mnt/music, which has 36 extents
Wed 18 Nov 20:52:28 EST 2020: Trim complete, image now has 36 extents
Wed 18 Nov 20:52:28 EST 2020: Unmounting /mnt/music...
umount: /mnt/music: target is busy.
Wed 18 Nov 20:52:28 EST 2020: Failed to unmount /mnt/music, trying lazy unmount.
Wed 18 Nov 20:52:29 EST 2020: lazily unmounted /mnt/music
Wed 18 Nov 20:52:29 EST 2020: Finished copying music.
Wed 18 Nov 20:52:29 EST 2020: unmounting /mnt/archive
Wed 18 Nov 20:52:29 EST 2020: Connecting usb to host...
Wed 18 Nov 20:52:29 EST 2020: Connected usb to host.
Wed 18 Nov 20:52:29 EST 2020: Waiting for archive to be unreachable...
Wed 18 Nov 20:52:29 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:30 EST 2020: Retrying...
Wed 18 Nov 20:52:30 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:31 EST 2020: Retrying...
Wed 18 Nov 20:52:31 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:32 EST 2020: Retrying...
Wed 18 Nov 20:52:33 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:34 EST 2020: Retrying...
Wed 18 Nov 20:52:34 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:35 EST 2020: Retrying...
Wed 18 Nov 20:52:35 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:36 EST 2020: Retrying...
Wed 18 Nov 20:52:36 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:37 EST 2020: Retrying...
Wed 18 Nov 20:52:37 EST 2020: Sleeping before retry...
Wed 18 Nov 20:52:38 EST 2020: Retrying...
Wed 18 Nov 20:53:44 EST 2020: unmounting /mnt/musicarchive
Wed 18 Nov 20:55:15 EST 2020: waiting up to 90 seconds for idle interval
Wed 18 Nov 20:56:57 EST 2020: couldn't determine idle interval
Wed 18 Nov 20:56:58 EST 2020: taking snapshot of cam disk in /backingfiles/snapshots/snap-000002
Wed 18 Nov 20:57:01 EST 2020: took snapshot
Wed 18 Nov 20:57:05 EST 2020: comparing new snapshot with /backingfiles/snapshots/snap-000001/snap.bin
Wed 18 Nov 20:57:05 EST 2020: new snapshot is identical to previous one, discarding

Why is my pi disconnecting without copying ALL the files?  

Thank you for the help

marcone, 2020-11-19:
>Why is my pi disconnecting without copying ALL the files?

it's because after 55 minutes of copying, teslausb's connection monitor wasn't able to reach the server anymore, concluded you went out of wifi range, and stopped the transfer:

>Wed 18 Nov 19:57:35 EST 2020: Copying music from archive...
>Wed 18 Nov 20:52:27 EST 2020: connection dead, killing copy-music

I'm not sure exactly why this happens. It seems to be related to poor network flow control in the network stack (the music copying saturates the connection leaving the connection monitor unable to reach the server), but it's pretty rare that it happens after successfully copying for so long.

marcone, 2020-11-19:
(it would have resumed and copied the remaining files the next time it booted, BTW)

closed 2020-11-19 by marcone