marcone/teslausb #559: Change CAM drive or MUSUC drive size after setup

aaronac8, 2020-11-19:
Is there a way to change the CAM drive or MUSIC drive size after setup is complete?  Also, if I want to up date my install with an updated version of teslausb, can I do this without going through a complete redo of my setup?

marcone, 2020-11-19:
https://github.com/marcone/teslausb/wiki/FAQ#q-how-do-i-update-to-the-latest-version-of-teslausb

To change the size of the cam or music drives, you can use [this script](https://github.com/marcone/teslausb/blob/main-dev/tools/resize-image.sh)

closed 2020-11-19 by marcone

aaronac8, 2020-11-19:
I am very new to using a raspberry pi.  How do I use a script?  For
example, if I wanted to change my CAM drive to 50 GB and my MUSIC drive to
150 GB.

Thank you