marcone/teslausb #561: Using Google drive and cellphone as hotspot

aaronac8, 2020-11-25:
Can I setup my tesla cam folder on my Google drive and my tesla music on my PC?  

How do I setup synching my tesla cam folder when connected to my mobile hotspot and my music folder on my home wifi?

If these are doable, then please write the directions in simple to follow instructions since I am new to raspberry pi programing.  Thanks

aaronac8, 2020-11-25:
I found this:
https://www.teslaev.co.uk/using-a-pi-zero-to-store-and-upload-sentry-dash-cam-content-to-google-drive/

However, when I get to this part:
mkdir -p /root/bin
cd /root/bin
wget https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/pi/setup-teslausb
chmod +x setup-teslausb
./setup-teslausb

I get an error...where the variables conf file is not found.  In the SSD card I did nothing to the image except for change the wifi car sample file...as instructed in the article.  I did nothing to the variable conf.sample file....was I supposed to?

marcone, 2020-11-25:
Please follow the instructions [here](https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md) instead of the ones from the article you found.

Music syncing only works for cifs. If you're using rclone or rsync as the archiving method, you should copy music manually (i.e. plug the Pi in to a PC, wait for the music drive to show up, and fill it with the music of your choice).

closed 2020-11-25 by marcone

aaronac8, 2020-11-25:
So, you can only use either cifs or rclone?  You can not use cifs for the
music drive and rclone for the cam drive?

marcone, 2020-11-25:
> You can not use cifs for the music drive and rclone for the cam drive?

Correct