marcone/teslausb #562: Pi4 Setup dies-no SSH

rudiger999, 2020-11-26:
Any help would be appreciated muchly.  The same procedures are successful with a PI Zero W, but always fail with the Pi4 B. 
 Multiple attempts with varying setup conf variables, and multiple SD card and sizes, and wired eth0 connection, all end in the same failure.....
The setup log is:
Wed 22 Jan 23:46:20 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Wed 22 Jan 23:46:20 GMT 2020 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Wed 22 Jan 23:46:20 GMT 2020 : Rebooting...

Not much to go on here.  In the final attempt, an SSD is attached, and this results in a bit more activity light action....but still no joy...

When the SD card contents are investigated, the following files are present (1/22/20 files are newly created by the installation , but the correct date is 11/26/20):
Volume in drive I is boot
 Volume Serial Number is E843-00F0

 Directory of I:\

01/22/2020  11:24 PM    <DIR>          overlays
01/17/2020  10:01 AM            24,201 bcm2708-rpi-b-plus.dtb
06/24/2019  03:21 PM            18,693 COPYING.linux
01/17/2020  10:01 AM             1,594 LICENCE.broadcom
01/17/2020  10:01 AM            24,379 bcm2708-rpi-zero-w.dtb
01/17/2020  10:01 AM            23,938 bcm2708-rpi-b.dtb
01/17/2020  10:01 AM            23,719 bcm2708-rpi-cm.dtb
07/10/2020  05:03 AM         2,884,708 start.elf
01/17/2020  10:01 AM            23,643 bcm2708-rpi-zero.dtb
01/17/2020  10:01 AM            25,265 bcm2709-rpi-2-b.dtb
01/17/2020  10:01 AM            25,394 bcm2710-rpi-2-b.dtb
01/17/2020  10:01 AM            27,054 bcm2710-rpi-3-b-plus.dtb
01/17/2020  10:01 AM            26,435 bcm2710-rpi-3-b.dtb
01/17/2020  10:01 AM            25,249 bcm2710-rpi-cm3.dtb
01/20/2020  01:44 PM            40,697 bcm2711-rpi-4-b.dtb
01/17/2020  10:01 AM            52,304 bootcode.bin
01/22/2020  11:45 PM               174 cmdline.txt
01/22/2020  11:42 PM             1,799 config.txt
07/10/2020  05:03 AM             6,746 fixup.dat
07/10/2020  05:03 AM             5,405 fixup4.dat
07/10/2020  05:03 AM             3,146 fixup4cd.dat
07/10/2020  05:03 AM             8,417 fixup4db.dat
07/10/2020  05:03 AM             8,419 fixup4x.dat
07/10/2020  05:03 AM             2,663 fixup_cd.dat
07/10/2020  05:03 AM             9,820 fixup_db.dat
07/10/2020  05:03 AM             9,818 fixup_x.dat
01/20/2020  01:44 PM         5,143,096 kernel.img
01/20/2020  01:44 PM         5,420,824 kernel7.img
01/20/2020  01:44 PM         5,754,904 kernel7l.img
01/20/2020  01:44 PM        13,521,408 kernel8.img
01/22/2020  11:42 PM               150 wpa_supplicant.conf.sample
07/10/2020  05:03 AM         2,272,992 start4.elf
07/10/2020  05:03 AM           816,124 start4cd.elf
07/10/2020  05:03 AM         3,774,532 start4db.elf
07/10/2020  05:03 AM         3,031,652 start4x.elf
07/10/2020  05:03 AM           694,052 start_cd.elf
07/10/2020  05:03 AM         4,861,512 start_db.elf
07/10/2020  05:03 AM         3,799,144 start_x.elf
01/22/2020  11:45 PM               153 issue.txt
01/22/2020  11:45 PM                65 vl805.sig
11/23/2020  01:45 AM             8,944 teslausb_setup_variables.conf.sample
01/22/2020  11:45 PM            98,904 vl805.bin
01/22/2020  11:45 PM            60,820 recovery.bin
01/22/2020  11:46 PM               237 teslausb-headless-setup.log
01/22/2020  11:46 PM               161 wpa_supplicant.conf
01/22/2020  11:46 PM                 0 WIFI_ENABLED
11/26/2020  06:13 AM                28 listing.txt
              46 File(s)     52,563,382 bytes
               1 Dir(s)     211,173,888 bytes free
Thanks for reading this far, and thank you for any suggestions.

marcone, 2020-11-26:
Normally the next lines in the setup log (after the reboot) should be:
```
<date> : Detecting whether to update wpa_supplicant.conf
<date> : Grabbing main setup file.
```
and then it continues from there.
So not sure what's going on. How long did you wait on the Pi before putting the sd card back in to the PC?

marcone, 2020-11-26:
Some Googling shows those new files (e.g. recovery.bin, vl805.bin, etc) are created by `rpi-eeprom-update`. I'm not sure if that runs as part of normal boot, or if you ran it manually at some point. More importantly though, "recovery.bin" is normally renamed to "recovery.000" after the EEPROM update completed successfully. So it looks it attempted to do an EEPROM update and didn't succeed/finish. I'd suggest letting it boot again and wait a bit longer, to see if it finishes and resumes installation.

rudiger999, 2020-11-27:
Thanks for that quick reply.  I waited what seemed like a reasonable amount of time..20 minutes maybe.  I have re-started the process with a fresh flash to the SD.  I will wait....and report back on this issue after a 2 hour wait.

rudiger999, 2020-11-27:
Still no joy after a number of different wait times, including the longest at 4 hours.  The failure is the same for all wait times.  I power cycled at the Tesla, but still no joy.  The meager activity light action concurs with the log file duration.  The Pi0W activity light is the expected activity, and is much more robust.  For now, the Pi0W will have to do.  But I am still interested in the setup failure for the Pi4.

marcone, 2020-11-28:
I suggest connecting the Pi4 to a display so you can see what's going on.

rudiger999, 2020-12-06:
That sounds like a simple suggestion, but I think it is not so.  How does one monitor a headless setup?  There is no monitor support for the pre-built image.  Any helpful suggestion on how to "manually" install would be welcome.  It appears that some instructions for this would be in the readme.md, but it is not clear which instruction is for a manual install and which is for the pre-built image.

marcone, 2020-12-06:
> That sounds like a simple suggestion, but I think it is not so. How does one monitor a headless setup?

By connecting one of the Pi4's hdmi ports to a computer monitor? When I connect my Pi4 to a display and use the prebuilt image, I see all the setup output scroll by.

rudiger999, 2020-12-07:
I am such a NOOb.....
I do see some lines, which flash by rather quickly and then the screen goes dark....
So I will try and capture those lines before they disappear.

rudiger999, 2020-12-07:
It does reach the reboot at the end....but still no joy.  There is an initial error reported that persists from the very beginning of the process, but that is all I can figure.  I captured two frames from a video of the screen, which are attached.....
[image]
[image]

marcone, 2020-12-07:
Have you tried booting e.g. Ubuntu for Raspberry Pi on it?
(https://ubuntu.com/download/raspberry-pi)

rudiger999, 2020-12-07:
Yes.  Ubuntu 20.10 (RPi 4/400) installs cleanly and runs well.  Also true for Raspberry Pi OS Full, Os Lite, and Pi OS 32 bit Debian.  I have encountered no config or hardware issues with this new unit.  Each OS updates and upgrades to current versions without complaint.

marcone, 2020-12-08:
Are you using an 8GB Pi4 by any chance?

rudiger999, 2020-12-08:
Yes

marcone, 2020-12-08:
There is at least one known problem with the 8GB Pi4 and teslausb (it won't boot from USB), and possibly others. I will be getting one today, so I can investigate and fix.

rudiger999, 2020-12-08:
Super cool.  I was going to send you mine.  Good luck, and let us know.

rudiger999, 2020-12-08:
Also, I would like to learn how to manually build this on top of a PiLite or Ubuntu platform.  I am primarily interested in harvesting recent clips to an attached SSD.  Because my interest is in video recorded when the vehicle is not in motion (Sentry and Recent video), I think I may need this manual build to customize.  Or perhaps I could use your suggestion of the R/W ssh access to your pre-built image.  Thoughts?

marcone, 2020-12-08:
I would suggest using that attached SSD for teslausb's storage. You can then access the recordings from another computer using Samba, or from the Pi in /mutable/TeslaCam.

It also sound like you don't really need teslausb's functionality. If you're not using teslausb functionality, there's no need to use the image or scripts. You could just install whatever linux you want and enable the mass storage emulation on it (it's standard linux functionality), and make your own custom system.

It should be possible to make teslausb work on Ubuntu or other linux, but I have little interest in pursuing that, because it would increase the support burden.

rudiger999, 2020-12-09:
Thank you for those thoughts.  I will read and read some more about your suggested approach without teslausb.  I think this is a great project, and I really like the auto archive functionality.  More importantly, it has sparked new ideas on how I can utilize Tesla cameras for my work (investigator).  I thank you for your project, and for your effort and patience in answering my questions.  It is all truly appreciated.

marcone, 2020-12-12:
The prebuilt image has been updated to support the 8GB Pi4

closed 2020-12-12 by marcone