marcone/teslausb #564: 2021 Refresh: USB not detected in glovebox

chrispazz, 2020-12-06:
Hello, in new 2021 M3 with USB in glovebox and firmware 2020.45.25, there is always an alert that USB is not detected and that Sentry mode will use more power.

Looking in the teslausb storage it seems to store regularly every sentry detected despite the error message present in notifications.

Anyone with the same behaviour as mine?

marcone, 2020-12-06:
This is just a guess, but could it be that the USB port is normally unpowered, and turning on Sentry mode is what causes it to be powered?
That would mean that when you turn on Sentry mode the drive is not yet available because the Pi still needs to start up, and so the car would complain about that. Then when the Pi is booted and makes the drive available to the car, the car starts recording to it.

chrispazz, 2020-12-07:
No it is always powered and also the PI Zero show led activity...

taro777, 2020-12-10:
I also could not my zero to get recognized in my 2021 m3 with usb in glovebox. 

Try to use another usb cable. I have 2 cables and both works when connected to my computer but for some reason only one works in the Tesla.

Both shows the same led activity when connected to the Tesla.

chrispazz, 2020-12-10:
Yes but I noticed that the Raspberry is working correctly and recording everything.....

chrispazz, 2020-12-21:
If it helps, 2021 glovebox USB is a 3.0.

marcone, 2020-12-21:
@taro777 Re: your comment that "only one [cable] works in the Tesla", are you saying it's working for you after switching to another cable?
The thing is, there's really nothing to be done on the Pi/teslausb side as far as I know. It uses standard linux functionality to emulate a USB drive, and if the car doesn't see that drive, then it's probably a hardware issue and there's nothing I can do.

The only time I've seen similar issues was when I used a Pi4 with an attached NVME drive that was drawing too much power for the USB port. The Pi would boot, but the car refused to see it as a drive and complained about "USB malfunction". I got it to work by using a [power/data splitter](https://www.amazon.com/gp/product/B00NIGO4NM) and providing power separately.

taro777, 2020-12-21:
@marcone  yes, np. I was just trying to help crispass since I had a similar issue and I found a way around it

taro777, 2020-12-21:
With the latest 2020.48.12.1 the Tesla no longer recognizes the rasperry. Saw you closed a similar as duplicate. My Tesla is in service now so I can’t give more than that atm, just a heads up something happend in the last update.

marcone, 2020-12-21:
> With the latest 2020.48.12.1 the Tesla no longer recognizes the rasperry

I have 2020.48.12.1 and teslausb still works for me.

chrispazz, 2020-12-21:
I tried again with 2020.48.12.1. Tesla give the not recognized error and the small size (32gb) error but it seems to record. 
I have ordered a new microsd 128gb and v30 speed. I will try to be sure that the problem is not related to the sd size or speed and will let you know.

kolasin2, 2020-12-28:
Can confirm this is happening on my 2021 M3.  I am currently on 2020.48.26 firmware, and using a Pi Zero W, with a 128GB Sandisk Ultra Plus SDXC card.  The files record onto the Pi fine, and it uploads to my fileserver via SMB when WiFi is in range just fine.  I am wondering if it has to do with the USB device class / parameters that are being presented to the MCU when plugged in.  I haven't tried tweaking any of this, but I haven't looked much in the way of how the Pi presents itself as a USB mass storage device to the car.

marcone, 2020-12-28:
@kolasin2 can you clarify the actual, exact, symptoms and messages? It seems odd for the car to say that "USB is not detected", but then record to USB anyway. So is the only symptom that the car shows that message, but everything else works normally?

kolasin2, 2020-12-28:
@marcone take a look at these screenshots.  You can see that the car UI is giving the "UI_a111" error.  However, if I go to look at the footage, all is there on the in-car clip viewer. I also see the clips synchronizing to my fileserver over SMB.
[image]
[image]

chrispazz, 2020-12-28:
Same behavior on my M3 also using new faster SD.

THX723, 2020-12-29:
I suspect the new glovebox USB port isn't actively powered (guessing it comes directly out of the MCU without going through an active hub like the two in the center console). The standard USB 5V/500mA supply is adequate for a passive USB stick, but not enough for a Pi. Until someone can do a teardown/wire trace on their 2021 Model 3, we won't know for sure.

In the meantime, try adding an active powered USB line to the power-only port of the Pi; keep the data-line plugged into he glovebox USB port.

kolasin2, 2020-12-29:
Ok I've done some testing, and I believe I have determined what the issue is.  The Pi Zero W is a USB2 device.

The error message is misleading.  The MCU *thinks* the device is plugged into one of the other USB-C ports in the car when the device plugged into the glovebox USB port is a USB2 device.  This doesn't stop the car from writing to the device, as it operates normally, which is why I am seeing the car write it's dashcam files to the Pi.

Here is what I did: 

- Started with a baseline of using the Tesla branded 128GB USB3 (Samsung brand) flash drive in the glove box.  No errors.
- Plugged in a 3rd party USB3 flash drive into the glove box port.  No errors.
- Plugged in the Pi Zero W into the glove box port.  UI_a111 error appears in warnings. (expected)
- Plugged in the Pi Zero W with an external power source in the power port.  UI_a111 error appears in warnings.
- Plugged in a USB2 flash drive into the glove box port.  UI_a111 error appears in warnings. (ah ha! not expected!)
- Used an Apple USB-C to USB-A adapter to plug the Tesla branded flash drive into a USB-C port in center console.  UI_a111 error appears in warnings.

I do have a Pi4 sitting around I could attempt this with, since it has both USB2 and USB3 ports, and would be an easy confirmation of this behavior.  So it *appears* that this error is specific to USB2 devices plugged into the glove box USB port on the Tesla Model 3 2021 refresh, and not a power issue with port and the Pi.

kolasin2, 2020-12-29:
As an update to my last comment, I put a USB3 hub in between the glovebox port and the Pi Zero W.  No change in behavior, with the UI_a111 error still showing.

THX723, 2020-12-29:
Thanks for the test and updates. I missed the part where TeslaUSB still worked as intended, despite the error message. Knowing that now, it starting to look more like the issue is on Tesla's sw side.

marcone, 2020-12-29:
> I do have a Pi4 sitting around I could attempt this with, since it has both USB2 and USB3 ports, and would be an easy confirmation of this behavior

The Pi4 can only do USB gadget emulation on the USB-C port, which is USB2.

chrispazz, 2020-12-29:
When connecting PI zero to my mac with USB I noticed that it connect, disconnect and then reconnect again.
Could it be that this first "reconnection" tricks Tesla software?

kolasin2, 2020-12-29:
@chrispazz I noticed that too, but it is unrelated. I put in a pull request (#593) to fix that. That behavior is due to some stuff in the archiveloop happening before networking is settled, because systemd does a parallel boot and brings up the teslausb service prior to WiFi being being fully up.

rubin110, 2021-02-05:
Hi there. I posted this in the Discord server a few nights ago but realized it should really get posted here too...

> Heyo. So I think that the 2021 Model 3s changed something with where the car thinks USB storage should be plugged into for dashcam.
> 
> After trying to run the pi for the past couple of weeks and ended up resorting to just using a usb drive again.
> Whenever I get in on the morning the dash will either throw up a ⚠️ about how USB storage isn't connected to the glovebox USB port and that might draw extra power (and the warning cannot be swiped away), or that just no storage is plugged in.
> 
> I've tried a bunch of different configurations of plugging the pi into the port directly, using a hub in the glove box, and plugging it into the center storage console. I couldn't figure out a configuration that worked better. I then used the OEM usb drive in similar configurations. Plugging in a hub into the glove box usb port, then plugging the drive into that also yields warnings that the drive not being plugged into the glove box port will drain battery faster. I grabbed a USB 3.0 90 degree angled adapter and used that between the OEM thumb drive and the glove box USB port, and that too caused the same warning, which is super interesting as there should be no circuitry inside that adapter.
> 
> Somethings like this...
> https://www.amazon.com/dp/B073GTBQ8V?ref=ppx_pop_mob_ap_share
> 
> Leaving the USB drive in that configuration over a few days again put the car into a state where it stopped recognizing that there's anything plugged in at all. So all this leads to me to believe that at least on my version of the car, if it doesn't see a very specific path on the USB tree, it'll get itself into a crap state. I've run out of time to investigate this more and have gone back to just a USB drive sitting in the glove box port, but wanted to circle around and share my findings here.

radhoo2k10, 2021-02-16:
Got the same problem. Mine don't save the sentry events. It only saves the 10 last min after i enter the car. From the outside I see the zero that is blinking 2 blinks pr sec. It also saves the entire trip home (50 min drive).

radhoo2k10, 2021-02-23:
It'a a bug in the Tesla software. It's the same problem with recording sentry clips as when I use the official usb from Tesla.  Have to wait for a bug fix from Tesla.

kodeypatterson, 2021-03-20:
A lot of great information here. Thanks for all the testing

radhoo2k10, 2021-03-20:
Mine has still the warning that the usb is not plugged in, but it is now recording in sentry mode and while driving.

kodeypatterson, 2021-03-20:
Mine is as well

boba51, 2021-03-30:
Agreed, although getting the error on my M3LR with the TeslaUSB it appears to record OK, I've swapped back to the Tesla supplied USB for the moment as I have a previously logged issue with having Sentry mode alerts but no recordings, which I'm booked in for a Service Appointment so don't want to confuse the situation. If I get the opportunity when speaking with one of their engineers I see if they can shed any light on this issue.

cavit99, 2021-03-31:
> Got the same problem. Mine don't save the sentry events. It only saves the 10 last min after i enter the car. From the outside I see the zero that is blinking 2 blinks pr sec. It also saves the entire trip home (50 min drive).

Are you by any chance running teslamate ?

radhoo2k10, 2021-03-31:
Nope. Not running teslamate.

boba51, 2021-03-31:
> It'a a bug in the Tesla software. It's the same problem with recording sentry clips as when I use the official usb from Tesla. Have to wait for a bug fix from Tesla.

Yep, Tesla Service Center engineer confirmed today, bug he's had 4/5 reports of it this week already.

Mastiha101, 2021-04-15:
Also getting the same error with a 2021 M3LR MIC i.e. UI_a111 "dashcam usb drive not plugged into glovebox".

I'm on software version 2021.4.15

LeiChat, 2021-05-19:
Same UI_a111 error on 2021 Model 3 running 2021.4.15 with Sandisk Extreme portable SSD, partitioned for Teslacam and music.
Seems to work fine recording clips and playing music despite the error.
Have not tried testing how much more energy is used when Sentry mode is enabled.

nowjon, 2021-06-11:
I have the same message, on version 2021.4.18.3

taney, 2021-06-21:
Subscribed! Just finished reading this. Can't wait for a fix!

zikronix, 2021-07-01:
this is a problem on my 2021 MY running 2021.4.18.10

missflash, 2021-07-04:
I have the same issues. 🥲

missflash, 2021-07-04:
> I suspect the new glovebox USB port isn't actively powered (guessing it comes directly out of the MCU without going through an active hub like the two in the center console). The standard USB 5V/500mA supply is adequate for a passive USB stick, but not enough for a Pi (especially when loaded). Until someone can do a teardown/wire trace on their 2021 Model 3, we won't know for sure.
> 
> 
> 
> In the meantime, try adding an active powered USB line to the power-only port of the Pi; keep the data-line plugged into he glovebox USB port.

Did this method solve the problem?

Kenny-54, 2021-07-06:
> Did this method solve the problem?

Same error for me with my 2021 TM3 (delivered in march).
I tried to plug both USB ports of my Pi zero : "PWR IN" port to a USB-C port of my TM3 and "USB" port to the USB port in glovebox. Same thing with the "PWR IN" port of the Pi plugged to my MacBook Pro (= as an external power source)

I also tried to plug the OEM USB stick trough a USB-A to USB-C adapter then to a USB-C to USB-A adapter. Yeah it's kind of stupid but the UI_a111 error appears in that case too.

CodeChief, 2021-07-07:
> I suspect the new glovebox USB port isn't actively powered (guessing it comes directly out of the MCU without going through an active hub like the two in the center console). The standard USB [...]

I tested this and initial results suggest it's NOT caused by the power limit. I tested with a USB power meter and the Pi 4 was delivered 5.2v @ 550-600ma which is the same as when it is connected to an active USB3 hub on my PC, just the voltage is a bit higher at 5.2v in the Tesla M3 (Refresh/2021) compared to dead-on 5v on the PC/hub. At idle it only consumes just over 500ma including my Pi4 case with fan running. The Pi0W only used 130-170ma with no case/fan. **So if it was caused by power/brown-out then the Pi0W should still work, but it doesn't.**

With reference to the [power supply guidelines from the Raspberry Pi Foundation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md) the Pi4 should only be drawing 600ma at idle and Pi0W 160ma at idle so these results have official documentation to validate against.

[image]

The Tesla branded USB stick drew almost nothing, 5.1v @ 20ma.

[image]

**So the problem remains, and appears to be something related to the USB "connection" because even my simple power meter (which should do nothing to the USB line except measure power) triggered the warning also with the original Tesla flash drive.**

[image]

My USB Safety Tester is only USB2 specification so I suppose the Tesla stick failure could be caused if the device and Tesla are really both USB3 (I doubt it). I will purchase a USB3 tester also with direct USB-C connectivitiy so I can rule that out and also cross-check the centre console ports. But for now this is already enough proof I think to baseline a Pi0W 100% compatible IF it can be connected in a way which the Tesla will accept as "direct".

**These results suggests the problem is NOT caused by TeslaUSB itself, more that ANY type of modification along the USB connection to the 2021 Model 3 glovebox will not be accepted without error by the current Tesla firmware. HOWEVER, perhaps the Pi USB emulation/driver can be modified in a way to WORKAROUND this Tesla specific (over-)sensitivitiy, so a "fix" is maybe possible on our side???**

s0schen, 2021-07-22:
Confirming this is also a problem in my 2021 Model Y with a Pi0W.  It still seems to work though despite the error.

CodeChief, 2021-08-06:
New error since car update 2021.12.25.7, says "USB tool slow / less than 4MB/s" reported in #239. Is it related to this issue/USB host emulation no longer compatible/acceptable according to Tesla firmware/refresh models???

marcone, 2021-08-06:
People have reported seeing the "USB too slow" message even when using an SSD plugged straight in to the car's USB port. It's unknown what actually prompts the car to display that message.

CodeChief, 2021-08-07:
I forgot to post my USB3 upgraded power tester results. Here they are for the record...

Even the Tesla original USB drive fails with just the new tester plugged between. Confirms the new glovebox USB hub is not happy with any kind of complex device/extension/"hub-like" attachment.

[image]

We can also see the extremely low power draw as expected with such a simple device.

[image]

Here the error (with the stick):

[image]

It is interesting what happens to the power AFTER the error. The car appears to DROP power right down low (or the USB protocol has some kind of goto-sleep/disable mode after negotiation errors?).

[image]

Here are the RasPi0W test results...

[image]
[image]
[image]

Here are the RasPi4 test results...

[image]
[image]
[image]

Tests on a USB3 PC hub confirm that these voltage and amp draws are perfectly normal, so the Tesla M3/MY Refresh glovebox hub IS providing enough power, most likely IS USB 3.x compatible. The Tesla original USB stick has "USB 3.1" etched into it (sure it's Samsung rebrand so we kind of know this already). I doubt Tesla would spec-up a USB 3.1 64/128GB USB stick for the glovebox then only install a USB2 device in it.

Finally, to close off the "not enough power" line of investigation I also bought the load tester and set it to 2A. First on the PC to test the expected behaviour then on the Tesla. Here we go, 2A out the glovebox USB port no problem...

[image]
[image]

Interesting that USB2 specification is 0.5A and USB3 is 1.5A according to Wikipedia. So when it's able to deliver 2A then it's likely USB3 plus quick charge support? Or is this totally wrong, is it breaking the specification, only supposed to have more after some kind of handshake)? I don't want to test with more than 2A on my car.

CodeChief, 2021-08-07:
Does anyone have a USB 3.1capable host tester? I wonder if there is some kind of USB development kit which can probe the behaviour of the RasPI running the USB emulation driver as seen by a PC host, then compare to a regular stick. The goal being to improve the RasPi driver until it matches perfectly, or discover a hard limit like a resistor then we know we are wasting our time unless we physically hack the Pi.

marcone, 2021-08-07:
Thanks for confirming that the port can provide enough power. Regarding the car's port being USB 3.1 compatible, somebody [mentioned earlier](https://github.com/marcone/teslausb/issues/564#issuecomment-751938632) that the problem appears to happen when the device plugged in to the glovebox port is NOT a USB3 device. The Pi's gadget-capable port is USB2 (for both Pi0 and Pi4).

LeiChat, 2021-09-05:
I am getting the alert when using a SanDisk Extreme Portable SSD which I believe is USB 3.1 compatible.

miles267, 2021-09-05:
Am upgrading from a 2019 to 2021 Model 3 and watching these comments closely. Have powered Pi Zero from center console USB hub via USB-A rather successfully thus far. Occasionally the Teslausb corrupts and requires a reimagine but it’s minor. Those with 2021, do you suggest also connecting via center console USB or glovebox?

aaronspruit, 2021-09-05:
> Am upgrading from a 2019 to 2021 Model 3 and watching these comments closely. Have powered Pi Zero from center console USB hub via USB-A rather successfully thus far. Occasionally the Teslausb [...]

I am using the glovebox with a Pi0, and after the last wipe it's been fine.  FYI, you'll need a cable, sadly the pi0 cannot just plug directly in with an adapter :(

CodeChief, 2021-09-05:
I wouldn't suggest using the centre console due to two drawbacks:
1. The whole point of the glovebox is it has PIN protection and no handle. So a thief is less likely to find and remove the USB stick and you may have evidence when the car is recovered or if they just broke in to grab stuff.
2. The center console ports are now USB-C connections, quite high-up and at the front in the way, quite easy to knock whilst digging down to grab hold of something. I think are positioned there to be easier for gamepads. So you would need some kind of angle adapter, then I guess we come back to the "any kind of hub or adapter generates an error even with USB storage" problem.

CodeChief, 2021-09-05:
> I am using the glovebox with a Pi0, and after the last wipe it's been fine. FYI, you'll need a cable, sadly the pi0 cannot just plug directly in with an adapter :(

@aaronspruit are you saying the errors are gone away? I have tried a split data/power USB cable and normal cables which made no difference. I've gone back to USB sticks now for stability, just test TeslaUSB occasionally hoping it's been fixed as we seem to be at a dead end until Tesla remove hurdles with firmware (if at all possible, it seems to be daisy chained to the max off the centre console) or some new update to the USB emulation of the Pi is released.

Did you get the latest firmware yet? I'm stuck on 2021.12.25.7 as I guess others are in Europe, waiting for global release of 2021.24.*.

aaronspruit, 2021-09-05:
> @aaronspruit are you saying the errors are gone away? I have tried a split data/power USB cable and normal cables which made no difference. I've gone back to USB sticks now for stability, just test [...]

Yes, as I wrote in the other post, I had some issues where it didn't mount anymore.  I ended up reflashing the pi0 (only change was using default CAM size instead of upping it to 60GB).  It stopped mounting after my update to 2021.12.25.7 - it seemed to correlate (happened right afterwards), but I don't know if it stopped right away, I noticed it about 3 days after the update.

I reflashed the pi0 and it was good on that version for about 1-2 weeks.  Two days ago I updated to 2021.24.5 and it has continued to work.  Ha, as I sat down to write this, I'm now being prompted for 2021.24.10, so will update to that and see.

aaronspruit, 2021-09-05:
> Yes, as I wrote in the other post, I had some issues where it didn't mount anymore. I ended up reflashing the pi0 (only change was using default CAM size instead of upping it to 60GB). It stopped [...]

Ugh, so I confused this thread/issue with other issue where teslausb wouldn't even mount anymore.

So, clarification, I still get the error notification, but it has been working for dashcam and sentry.  After the update to 2021.24.10 just now, dashcam is still working and I still get the error notification. 

I did have an issue where dashcam stopped working with 12.25.7.  I wiped the pi0 and have been good ever since (dashcam and sentry work, still get error notification). 

I use the glovebox USB port.

CodeChief, 2021-09-08:
So I tested the different connection options in my 2021 M3:

Connect to centre console as workaround - Pi0 does not even power-up with my USB-C to Micro adapter. The Pi4 does of course with a USB-C to C (normal) cable. But then we get the error "Dashcam USB drive not plugged into glovebox".

[image]

So it's pretty clear Tesla are telling us NOT to use the centre console for Dashcam.

The current image with current update ("setup-testlausb upgrade") still generates the annoying errors when correctly installed in the glovebox but "work" (function as dashcam and WiFi sync) with both Pi4 and Pi0. However the beeps and warning messages whilst entering and sometimes driving the car make that not feasible for me, so it doesn't *work* it just "works" (crappy). Swapping a couple of USB sticks occasionally as I enter/leave the car is my workaround until this is fixed. The GUI in the car is much more efficient to manage/delete the mostly junk sentry captures anyway.

There is one thing I did not test which I think is promising and will try to do this lunchtime. The new ExFAT format support allows us to remove one more difference to the standard Tesla formatted USB drive. It's a long shot as it appears the cause of this error is most likely firmware of the Tesla or Linux USB emulation driver, but I'll give it a go then report back later...

sambitshankar, 2021-09-16:
Nothing different to report from me either. I am on 2021.24.10 and I might be just used to the warning/error at this point but what's more concerning to me is the Dashcam icon disappearing altogether during my drives. I have to open the glovebox and unplug and replug the pi0 and it would show up again. Happens during parking sometimes too, sentry icon is on but there is no dashcam icon and I can't view the sentry event - maybe something to do with "Summon standby"? All to say that, if godforbid some even occurs and I want to save a clip, I can't because the dashcam icon isn't on the UI - I am out of luck. I love the convenience of it automatically backing up my clips as it reaches home but can't continue using it just for the previously mentioned reason. Anyone else face this?

zikronix, 2021-09-16:
I had this problem on a previous version and was the reason I stopped using it

Tim

mtehonica, 2021-09-17:
Same issue for me on 2021 Model Y using Pi4
Software version 2021.11.102 (vision only)

miles267, 2021-10-01:
Has this issue been fixed in any of the recent Teslausb builds yet? Or does it report this error but still continue to record sentry and dashcam anyway?

sambitshankar, 2021-10-01:
> Has this issue been fixed in any of the recent Teslausb builds yet? Or does it report this error but still continue to record sentry and dashcam anyway?

Nope. There is a separate issue open for this as well  #654 - no fix yet as far I know - right @marcone ? 

Unfortunately I had to stop using teslausb and revert back to using the old thumb drive because I almost got into an accident last week but there was no dashcam icon on the screen to record, honk didn’t record anything either. I really miss the convenience of teslausb.

CodeChief, 2021-10-01:
> Nope. There is a separate issue open for this as well #654 - no fix yet as far I know - right @marcone ?
>
> Unfortunately I had to stop using teslausb and revert back to using the old thumb drive [...]

Wow that was lucky. I stopped using it too because of similar behaviour, for me it prevented recording in the case of the alarm going off. Luckily, the alarm events are also saved in internal storage and uploaded to Tesla for a while afterwards since a recent firmware update. This issue should be raised in priority then.

The very latest Tesla firmware has the new option to also save internally in case of a crash. You need to set DashCam to "auto" and best also accept the data sharing agreements in various places of the menus to allow the footage to be uploaded as backup or if the car is stolen. However, I wonder if that still works if you have a low speed or parked "crash" and the airbags don't deploy, like somebody backing into your car at a carpark.

Yeah so It's a very thin argument to suggest it still "works" if you ignore the errors. I miss TeslaUSB too and could help fix this but sadly there is no way I will have time in the short erm. It appears to need a load of development on a device driver and perhaps even a hardware hack to get it working and reliable. Tesla don't seem to have any magic fix to make it more flexible like the old models (little to no USB2 compatibility in the glovebox seems to be the core problem on new models).

[119 more replies]