marcone/teslausb #565: Sentry Mode won't turn off

Jughead135, 2020-12-06:
I've been using the API feature to turn Sentry Mode on as a means of keeping the Pi powered when I pull into my garage.  It appears to be working, insofar as all my clips get archived.

What does NOT appear to be working is turning Sentry Mode back off when finished.  As a result, my car stays "awake," burning through roughly 4% / day while parked in the garage.  I also frequently find that the dashcam feature continues running, resulting in nothing but the last 20 (?) hours or so "video" of my garage.  That last part is not evident in the attached log (and I have no idea why the dashcam sometimes stays on & other times does not), but you can see the Pi has been running since I got home Fri afternoon (so, for roughly 48 hours & counting), long after it got done archiving.

I see that it apparently rebooted a few times (power interruptions??); and, sometimes it states `Failed to Enable Sentry Mode`.  However, the usual state of affairs appears to be Sentry Mode stays on indefinitely.

Happy to provide more data if needed.  I have done the setup & upgrade, with no effect.  This is the latest ~2,000 lines of archiveloop.log, I can dig back further if needed.

Any ideas?  Cheers!

[archiveloop_log 06dec20 sentry won't turn off.txt](https://github.com/marcone/teslausb/files/5649029/archiveloop_log.06dec20.sentry.won.t.turn.off.txt)

marcone, 2020-12-06:
From the log it looks like the USB port lost power even though Sentry mode was on, which caused the Pi to turn off.
It then regained power (can't tell if on its own, or if you caused it to be turned on by e.g. opening a door), at which point archiving resumed. See the log around these timestamps: "Fri  4 Dec 12:18:07", "Fri  4 Dec 12:56:22 EST 2020", "Fri  4 Dec 13:03:34", "Fri  4 Dec 14:33:26".
This has been reported before (https://github.com/marcone/teslausb/issues/384). Unfortunately I don't have a solution.

The fact that Sentry won't turn off again after archiving is complete is because when the Pi restarts and starts archiving again, it sees Sentry was already on, and so it leaves it on. There is no "force off" option because then it might turn off Sentry when you manually turned it on.

Jughead135, 2020-12-06:
Yeah, I was afraid it was something of that nature.  Is that consistent with the `Failed to Enable Sentry Mode` message on the reboot...?

marcone, 2020-12-06:
That's just a generic failure message, and could indicate anything from failure to contact Tesla's servers to using the wrong credentials. It seems the tesla api service has gotten a bit flaky lately. I'm seeing it hang a lot.

Jughead135, 2020-12-06:
Grrrr.  OK--as always, thanks for all you're doing with this.  I'm still looking for a satisfactory way to keep the Pi powered all the time, to avoid this kind of issue altogether.

hoopsta1423, 2020-12-07:
I just noticed this on mine as well, was hoping to find a solution :(

mjpcomp, 2020-12-31:
Maybe have an option where:
1 - IF connected to certain WiFi APs
2 - AND No files to archive
3 - THEN Turn off Sentry (if on)

Or:
1 - IF connected to certain WiFi APs
2 - AND No files to archive
3 - AND specific file exists (e.g. "NoSentry.txt" or something)
4 - THEN Turn off Sentry (if on)

Maybe have an option where it's similar to above, but, it will only disable Sentry if the vehicle is at or below a certain battery level?

I've been having this problem recently - rclone completes the transfer to OneDrive, but, Sentry never turns off after, and a few hours later, I realize it's still on, after it's depleted my battery...

Jughead135, 2021-01-01:
> I just noticed this on mine as well, was hoping to find a solution :(

Well, I've found a solution I can live with.  It is NOT a software solution, so it doesn't "solve" this problem so much as side step it.

$10 gizmo from Amazon.  Plugs into OBD-II port, provides constant +5V.  Plenty of cord to wind its way (completely out of sight) into the center console.  https://www.amazon.com/gp/product/B01H38IIY6/?tag=tmc064-20

My Pi ran for two weeks with TeslaUSB running the whole time.  Both were flawless.  I'm still messing around with final placement of the Pi within the console, but this was the solution I was seeking--frankly, I like it better than keeping the whole car awake just to run the Pi.

```
root@MimiPi:~# uptime
 13:18:36 up 14 days, 19:34,  1 user,  load average: 1.20, 1.36, 0.74
```

marcone, 2021-01-01:
@Jughead135 That's what I'm doing too. Unfortunately it's not a solution for people with a Model 3 or Y, since those don't have a standard OBD-II port.

Jughead135, 2021-01-01:
> @Jughead135 That's what I'm doing too. Unfortunately it's not a solution for people with a Model 3 or Y, since those don't have a standard OBD-II port.

Ah.  I didn't know that, thanks.

MarcFinns, 2021-01-03:
Since the last couple of Tesla updates  I am having a related issue. I disabled sentry at home, but after teslausb finished uploading the files to the nas, sentry stays on. It used to work correctly!

miles267, 2021-01-03:
> Since the last couple of Tesla updates I am having a related issue. I disabled sentry at home, but after teslausb finished uploading the files to the nas, sentry stays on. It used to work correctly!

I had encountered same issue. How I resolved it was to delete my home location completely out of favorites and disable Sentry Mode altogether. Then leave home and drive to a location. From there, re-enter your home location, turn on Sentry Mode and set except home in settings. It should then work as you’re expecting.

marcone, 2021-06-26:
I just re-enabled the "keep the car awake" feature, and by default it's using a different method than turning on Sentry mode (though that is still available too in case the new method doesn't work for you for some reason).
Since the new method doesn't affect Sentry, it won't cause Sentry to stay on or turn off when it shouldn't.

closed 2021-06-26 by marcone