marcone/teslausb #566: Cannot mount /mnt/cam

klausholstjacobsen, 2020-12-06:
Hello!

downloaded and installed newest teslausb recently on "pi zero w".

No clips are being backed up.
/mutable/archiveloop.log has this content many times:

Sun  6 Dec 15:27:33 GMT 2020: Retrying...
Sun  6 Dec 15:27:33 GMT 2020: Mounting /mnt/cam...
mount: /mnt/cam: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
Sun  6 Dec 15:27:33 GMT 2020: Failed to mount /mnt/cam.
Sun  6 Dec 15:27:33 GMT 2020: Sleeping before retry...
Sun  6 Dec 15:27:34 GMT 2020: Retrying...

For the record, I've configured the system to use rsync although I doubt it is related to this local mount issue.
I see the backing file:

root@teslausb:/root# ll /backingfiles/
total 36G
drwxr-xr-x  3 root root   43 Dec  4 14:19 .
drwxr-xr-x 23 root root 4.0K Dec  4 12:35 ..
-rw-r--r--  1 root root  40G Dec  6 15:58 cam_disk.bin
drwxr-xr-x  3 root root   25 Dec  6 15:52 snapshots
root@teslausb:/root# 

Setup was successful and the car does not complain when saving clips!
What could be wrong?

Regards
Klaus

marcone, 2020-12-06:
Can you attach the output of `/root/bin/setup-teslausb diagnose` ?
(if that prints an error, run `/root/bin/setup-teslausb selfupdate` first)

klausholstjacobsen, 2020-12-08:
Marcone, thanks for the ultra quick response.

I think I figured it out.
After installing and plugging the pi zero in the "usb drive" came up nicely in the car. 
To make sure everything was matching what the car expected i pushed the button "format usb drive" in the car.
This surely caused the car to format with whatever filesystem it prefers, thus rendering the drive unmountable by the pi itself.

After re-initializing the cam partition/file on the pi there is no more mount errors to be seen.

I will test if clips are actually recorded and stored later.

Thanks

closed 2020-12-08 by klausholstjacobsen