marcone/teslausb #567: TeslaUSB keeps disconnecting from Model X when it's been asleep for a while

lytesout2376, 2020-12-08:
Similar to [#74](https://github.com/marcone/teslausb/pull/74)

Every time the Model X has been asleep for a while, the camera icon does not appear on the touch screen as if the car is not detecting the pi as a USB device to save footage to. I have to disconnect the pi and reconnect for the car to recognize it as a USB drive. My pi is connected to portable battery for power and a data cable from the pi to the Model X. 

Diagnostics.txt (sorry I don't know how to copy the entire text file. I know how to output, but since it is so long I lose the first part of it. Hopefully it's enough info)

```
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: error in submission: ep1out --> -11
[Mon Dec  7 13:38:12 2020] dwc2 20980000.usb: new device is high-speed
[Mon Dec  7 13:38:12 2020] dwc2 20980000.usb: new address 13
[Mon Dec  7 13:38:12 2020] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[Mon Dec  7 13:38:14 2020] systemd[1]: Started Journal Service.
[Mon Dec  7 13:42:10 2020] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[Mon Dec  7 13:42:35 2020]  loop0: p1
[Mon Dec  7 13:42:35 2020]  loop0: p1
[Mon Dec  7 13:42:43 2020]  loop1: p1
[Mon Dec  7 13:42:46 2020]  loop1: p1
[Mon Dec  7 13:42:48 2020] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[Mon Dec  7 13:42:48 2020] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[Mon Dec  7 13:44:09 2020] Mass Storage Function, version: 2009/09/11
[Mon Dec  7 13:44:09 2020] LUN: removable file: (no medium)
[Mon Dec  7 13:44:09 2020] LUN: removable file: /backingfiles/music_disk.bin
[Mon Dec  7 13:44:09 2020] LUN: removable file: /backingfiles/cam_disk.bin
[Mon Dec  7 13:44:09 2020] Number of LUNs=2
[Mon Dec  7 13:44:09 2020] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[Mon Dec  7 13:44:09 2020] g_mass_storage gadget: g_mass_storage ready
[Mon Dec  7 13:44:09 2020] dwc2 20980000.usb: bound driver g_mass_storage
[Mon Dec  7 13:44:09 2020] dwc2 20980000.usb: new device is high-speed
[Mon Dec  7 13:44:09 2020] dwc2 20980000.usb: new address 14
[Mon Dec  7 13:44:09 2020] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[Mon Dec  7 13:51:45 2020]  loop0: p1
[Mon Dec  7 14:51:36 2020]  loop0: p1
[Mon Dec  7 14:51:36 2020]  loop0: p1
[Mon Dec  7 15:51:28 2020]  loop0: p1
[Mon Dec  7 16:10:32 2020]  loop0: p1
[Mon Dec  7 16:10:32 2020]  loop0: p1
[Mon Dec  7 16:51:18 2020]  loop0: p1
[Mon Dec  7 16:51:18 2020]  loop0: p1
[Mon Dec  7 17:51:09 2020]  loop0: p1
[Mon Dec  7 17:51:09 2020]  loop0: p1
[Mon Dec  7 18:03:00 2020] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[Mon Dec  7 18:03:00 2020] brcmfmac: power management disabled
[Mon Dec  7 18:18:09 2020] dwc2 20980000.usb: new device is high-speed
[Mon Dec  7 18:18:09 2020] dwc2 20980000.usb: new address 23
[Mon Dec  7 18:18:09 2020] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[... 964 lines cut ...]
root     10872 10858  0 00:44 pts/0    00:00:00 tr \r \n
root     10873 10858  1 00:44 pts/0    00:00:00 sed /^ *$/d
root     10874 10858  1 00:44 pts/0    00:00:00 grep -a -v ^Reading package lists
root     10875 10858  0 00:44 pts/0    00:00:00 grep -a -v ^(Reading database
root     10877 10858  1 00:44 pts/0    00:00:00 grep -a -v ^Adding 'diversion of
root     10878 10858  1 00:44 pts/0    00:00:00 grep -a -v ^Removing 'diversion of
root     10879 10858  1 00:44 pts/0    00:00:00 sed -E s/\o033\[0;32m//
root     10880 10858  1 00:44 pts/0    00:00:00 sed -E s/\o033\[0m//
root     10900    95  4 00:44 ?        00:00:00 /lib/systemd/systemd-udevd
root     10901    95  3 00:44 ?        00:00:00 /lib/systemd/systemd-udevd
root     10902    95  3 00:44 ?        00:00:00 /lib/systemd/systemd-udevd
root     10903    95  5 00:44 ?        00:00:00 /lib/systemd/systemd-udevd
root     10924 10871  0 00:44 pts/0    00:00:00 ps -eaf
root     17911     1  0 Dec07 ?        00:00:16 /lib/systemd/systemd-journald
root     25018     2  0 Dec07 ?        00:00:00 [cifsd]
root     27033     1  0 Dec07 ?        00:00:00 /sbin/dhclient -4 -v -i -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -df /var/lib/dhcp/dhclient6.wlan0.leases wlan0
root     29463     2  0 Dec07 ?        00:00:06 [kworker/u2:2-brcmf_wq/mmc1:0001:1]
root     30899     2  0 Dec07 ?        00:00:02 [kworker/0:2H-xfs-log/mmcblk0p3]
root     30986     2  0 Dec07 ?        00:00:00 [loop1]
teslausb has been up 12 hours, 54 minutes. System time is Tue  8 Dec 00:44:59 PST 2020
====== end of diagnostics ======
root@teslausb:/tmp# pbcopy diagnostics.txt
-bash: pbcopy: command not found
root@teslausb:/tmp#
```

marcone, 2020-12-08:
Your wifi appears to be toggling on and off repeatedly. I see this very rarely on my own Pi0, I think it's a bug in the wifi driver. Rebooting the Pi usually fixes it.
Loss/regain of wifi makes teslausb start an archive cycle, which removes the drive from the car, so this could be what you're seeing.

To copy the entire diagnostics, do something like `setup-teslausb diagnose > /tmp/diagnostics.txt` on the Pi, and then on the computer where you normally run `ssh` to ssh into the Pi, instead run `scp pi@teslausb.local /tmp/diagnostics.txt .`

lytesout2376, 2020-12-08:
Thanks, I've just rebooted the Pi. I have some questions that might troubleshoot:

Do my diagnostics show the Pi losing power? I ask because sometimes I find my power bank off. There's an on and off button (not switch) and sometimes it just powers itself off. Not sure if this is part of the issue.

Does the diagnostics show whether it's WiFi toggling on/off vs losing connection to WiFi? I ask because the WiFi signal in the garage (where the car is parked) is just inside my WiFi coverage boundary. So I'm wondering if improving my WiFi signal in the garage would help this.

I haven't parked for a long time away from my home (where WiFi is located) so I'll test this while I'm out to see TeslaUSB disconnecting from the Model X occurs when away from WiFi. This would rule out the problem being WiFi.

Here's my full diagnostics after doing a reboot (thanks for that tip):

```
====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
lun0 connected, from file /backingfiles/music_disk.bin
lun1 connected, from file /backingfiles/cam_disk.bin
2 snapshots mounted
USB state (491.1 seconds ago): connected to host and host mounted drive(s)
====== disk / images ======
Model: SD SN64G (sd/mmc)
Disk /dev/mmcblk0: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2261MB  1988MB  primary  ext4
 3      2261MB  63.8GB  61.5GB  primary  xfs
 4      63.8GB  63.9GB  105MB   primary  ext4
cam disk image has 131 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 17.2GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  17.2GB  17.2GB  primary  fat32        lba
music disk image has 1 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 4295MB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  4295MB  4294MB  primary  fat32        lba
====== network ======
ap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.66.1  netmask 255.255.255.0  broadcast 192.168.66.255
        inet6 fe80::ba27:ebff:fecb:4cdf  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:cb:4c:df  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 103  bytes 18689 (18.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 79  bytes 6633 (6.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 79  bytes 6633 (6.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.29.248  netmask 255.255.255.0  broadcast 192.168.29.255
        inet6 fe80::ba27:ebff:fecb:4cdf  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:cb:4c:df  txqueuelen 1000  (Ethernet)
        RX packets 11535  bytes 1799811 (1.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 45345  bytes 66098565 (63.0 MiB)
[... 2359 lines cut ...]
[Tue Dec  8 09:35:31 2020] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[Tue Dec  8 09:35:31 2020] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[Tue Dec  8 09:35:31 2020] of_cfs_init
[Tue Dec  8 09:35:31 2020] of_cfs_init: OK
[Tue Dec  8 09:35:31 2020] Waiting for root device PARTUUID=27cf9d8e-02...
[Tue Dec  8 09:35:31 2020] random: fast init done
[Tue Dec  8 09:35:31 2020] mmc0: host does not support reading read-only switch, assuming write-enable
[Tue Dec  8 09:35:31 2020] mmc0: new high speed SDXC card at address aaaa
[Tue Dec  8 09:35:31 2020] mmcblk0: mmc0:aaaa SN64G 59.5 GiB
[Tue Dec  8 09:35:31 2020]  mmcblk0: p1 p2 p3 p4
[Tue Dec  8 09:35:31 2020] mmc1: new high speed SDIO card at address 0001
[Tue Dec  8 09:35:31 2020] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[Tue Dec  8 09:35:31 2020] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[Tue Dec  8 09:35:31 2020] devtmpfs: mounted
[Tue Dec  8 09:35:31 2020] Freeing unused kernel memory: 468K
[Tue Dec  8 09:35:31 2020] This architecture does not have kernel memory protection.
[Tue Dec  8 09:35:31 2020] Run /sbin/init as init process
[Tue Dec  8 09:35:32 2020] systemd[1]: System time before build time, advancing clock.
[Tue Dec  8 09:35:32 2020] NET: Registered protocol family 10
[Tue Dec  8 09:35:32 2020] Segment Routing with IPv6
[Tue Dec  8 09:35:32 2020] systemd[1]: systemd 241 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[Tue Dec  8 09:35:32 2020] systemd[1]: Detected architecture arm.
[Tue Dec  8 09:35:32 2020] systemd[1]: Set hostname to <teslausb>.
[Tue Dec  8 09:35:32 2020] systemd[1]: Failed to bump fs.file-max, ignoring: Invalid argument
[Tue Dec  8

marcone, 2020-12-08:
> Do my diagnostics show the Pi losing power?

Only indirectly, since the archiveloop shows when it starts up on boot (look for "Starting archiveloop" in the diagnostics or /mutable/archiveloop.log)

> Does the diagnostics show whether it's WiFi toggling on/off vs losing connection to WiFi?

I can't say for sure, but your log toggled between "wlan0: link is not ready" and "wlan0: link becomes ready", and looking at my own log I only see the "link becomes ready" when I come back home. So what you're seeing is different from just going out of wifi range and coming back.

lytesout2376, 2020-12-09:
I'm not sure that TelsaUSB disconnecting from the Model X is caused by the WiFi toggling on/off. Here's why:

At 12/8 at 09:35 I rebooted the pi per your instructions. I disconnected the data cable and reconnected and ensured the pi was connected to the Model X and the camera icon was showing on the touch screen.

At 12/8 at 11:55 I got in the car and started a drive. I noted the pi was not connected to the Model X so at 12:00pm I disconnected the cable and reconnected the cable to get the camera icon to show on the touch screen.  During this drive I tapped on the camera icon two times to capture dashcam footage.

At 12/8 12:16 I came back home and the pi reconnected to WiFi and archived the footage. The Model X fell asleep...

At 12/8 15:46 I checked on the Model X and noted the camera icon is not showing on the touch screen again.  I reviewed the logs and `wlan0: link is not ready` last appeared at 12:00.

Here's the relevant part of diagnostics.txt: https://pastebin.com/zRjpS3Sz

marcone, 2020-12-09:
It looks like you have something at your house that accesses teslausb's Samba share every two minutes. Not saying that that's the cause of the problem, but thought I'd mention it since it shows in the log and seems unusual.
As far as the car not seeing the drive anymore, I don't know. The log shows usb mass storage being enabled, and as far as the Pi is concerned that's all there is to it. Could be bad hardware/cabling.

lytesout2376, 2020-12-09:
Thanks, I think it's my computer, it has mounted that samba share as a drive. I have unmounted it. What is the log entry that tells you that?

I am going to try with a brand new Pi 4 that I just got. Would I be able to just take the SD card and put it in the Pi 4?  Or do I need to do a new flash / install?

marcone, 2020-12-09:
The log contained "loop0: p1" every two minutes, which is logged when a snapshot is taken, which is (also) done when a new Samba connection is established.

Transferring the sd card should work to some degree, though the AP functionality relies on a config file storing the network MAC address of the Pi, so that wouldn't work. In other words, for full functionality you should reinstall from scratch.

Any full USB cable should work in theory. I've never had cable problems myself, but I know others have had to try multiple cables because their cables turned out to be charge-only.

closed 2021-03-08 by marcone