marcone/teslausb #568: 2020.48.10 advisory: Minimum USB size is 32GB

THX723, 2020-12-11:
Noticed the latest Tesla SW will now flag the minimum USB size if less than 32GB. I don't believe I've seen this advisory prior to 2020.48.10. I'm currently configured for 16GB per the .conf default.

I would suggest updating `teslausb_setup_variables.conf.sample` to reflect 32GB default and related verbiages.

MarcFinns, 2020-12-11:
+1
Saw it today after the upgrade.
Is there a way to resize the volume without reinstalling the whole thing?

SergioRius, 2020-12-12:
I'm also getting an alert advising that videos will not be saved because the small size.
Looking after a solution on increasing the size without reinstalling.

miles267, 2020-12-12:
Same here.  Oddly I have my EXPORT CAMSIZE=64G and EXPORT MUSICSIZE=4G yet after one fairly short drive on 2020.48.10, I get the warning prompt on the screen that reports that the usb drive is getting full and should be reformatted.  Also seeking an option to increase the volume size without a reimage/reinstall.

SergioRius, 2020-12-12:
Found this:
https://github.com/marcone/teslausb/issues/265

marcone, 2020-12-12:
https://github.com/marcone/teslausb/issues/560#issuecomment-730722972

marcone, 2020-12-12:
@SergioRius as far as I can tell it's just a warning, but everything still works with a smaller size. Are you actually seeing a message telling you that videos will not be saved?
In any case, I've updated the default size in the sample config size.

closed 2020-12-12 by marcone

SergioRius, 2020-12-13:
> Are you actually seeing a message telling you that videos will not be saved?

Yes. I saw the warning message about size, but as it apparently worked, I didn't pay so much attention. Until I received the error message. My setup was new and I suppose it jumped when I reached some usage.
I'll try to find it on notifications and share with you.

I only replied above for sharing more info. Good work!