marcone/teslausb #570: Model 3, 2021 version, Published WiFi and Power Consumption

ThomasHej, 2020-12-12:
Downloaded the image and did everything according to the scripts.  It kind of works - it does pick up the clips and it does sync them to my share.  It also sync the music from my share onto the Pi, so far so good.

A few issues though:

#1 The car constantly comes up with a warning stating: No USB-flash drive is inserted in the glovebox-USB so sentry mode will use extra power.  The Pi has not been removed at any time and the message is constantly "active" in the car.

#2 No WiFi network is published from the Pi (settings was changed to publish local WiFi), so I can't get to the files as announced while on the road.  The "Tesla USB WiFi" does not show up on any of my devices.

#3 The sync seems to be fairly show, and I can't really figure out when exactly the sync is taking place.

#4 It seems to be using a lot of power.

My setup config file (passwords removed) enclosed.  I don't know if this is due to the Model 3 being the new 2021 model?

Thanks,

Thomas
[teslausb_setup_variables.zip](https://github.com/marcone/teslausb/files/5683858/teslausb_setup_variables.zip)

marcone, 2020-12-12:
Please attach diagnostics. See https://github.com/marcone/teslausb/issues/35

ThomasHej, 2020-12-13:
Hi Marcone,

Selfupdate returned "up to date"
Diagnostics here:

root@teslausb:/tmp# cat ./diagnostics.txt
====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
lun0 connected, from file /backingfiles/music_disk.bin
lun1 connected, from file /backingfiles/cam_disk.bin
root@teslausb:/tmp#

Let me know if you want me to run anything else...

Do you know the answer to the following question:  When you have a sentry incident, there is a event.jason file that states the reason for the event and a timestamp.  The timestamp might show 12:08:34 and now you have the last 10 minutes up to about that time saved.  But what about the next few minutes.  What happened then?   Can you get to that somehow?   Is that also saved?

Thomas

marcone, 2020-12-13:
There should be a lot more diagnostic output than that. Were there any error messages when you ran `setup-teslausb diagnose`?

The car is always recording. Anything that wasn't moved into a SentryClips folder will be in RecentClips.

ThomasHej, 2020-12-13:
Thanks for a quick reply.  I ended up re-flashing and after having done that, everything seems to be working.  I believe I had a partial install that didn't complete automatically.  

On the "where are the next 5 minutes", what I mean is that when the trigger event happens, is only the past saved?  The rest potentially could be overwritten...

Thanks again.

marcone, 2020-12-13:
The car is continuously recording 1-minute clips. When it detects an event, it saves the last 10 minutes worth of recordings. The actual event will be somewhere in the last minute worth of recordings, so there's a usually a little bit of the "after" as well.
As for recordings being overwritten, that's why teslausb takes snapshots of the recordings, so they're still available after the car deletes them.

closed 2020-12-13 by marcone