marcone/teslausb #573: Setup successful, CAM drive doesn't show up.

murtuzahabib, 2020-12-13:
I can see the 'Teslausb' wifi and can connect to the pi using my phone.  

Whilst connected to my Mac, Pi LED is lit up and not blinking. CAM folder doesn't show up.  

Could you please look into this issue?  Diagnostics file attached.

Thanks.

[diagnostics.txt](https://github.com/marcone/teslausb/files/5685353/diagnostics.txt)

marcone, 2020-12-13:
The archiveloop script is not running for some reason, and apparently never ran because /mutable/archiveloop.log doesn't exist.
Does `journalctl -u teslausb` show any errors?

murtuzahabib, 2020-12-13:
this is the log:

-- Logs begin at Thu 2019-02-14 10:12:02 GMT, end at Sun 2020-12-13 20:17:02 GMT. --
Dec 13 16:17:01 teslausb systemd[1]: Started TeslaUSB archiveloop service.
Dec 13 16:17:01 teslausb bash[186]: Powering off HDMI
Dec 13 16:17:02 teslausb bash[186]: /root/bin/setup-teslausb was sourced, setting setup variables
Dec 13 16:17:02 teslausb bash[186]: /root/bin/archiveloop: line 44: ARCHIVE_SYSTEM: unbound variable
Dec 13 16:17:02 teslausb systemd[1]: teslausb.service: Main process exited, code=exited, status=1/FAILURE
Dec 13 16:17:02 teslausb systemd[1]: teslausb.service: Failed with result 'exit-code'.

marcone, 2020-12-13:
You need to set ARCHIVE_SYSTEM in the config file.

closed 2020-12-13 by marcone

murtuzahabib, 2020-12-13:
Did the trick setting to none. Thanks.