marcone/teslausb #574: Rclone config file issue

murtuzahabib, 2020-12-14:
After successfully setting up rclone, getting the error when trying to ls into mapped google drive.

"Failed to save config after 10 tries: Failed to create temp file for new config: open /root/.config/rclone/rclone.conf705945223: read-only file system"

24 hours ago, could ls into the mapped drive and see the files in it.

marcone, 2020-12-14:
Did you change the archive method to "rclone" in the config file and run setup again?

murtuzahabib, 2020-12-14:
Yes, did as per instructions. I could access the files in the gdrive from terminal. 

When i try doing that 24 hours later, i get the above error.

marcone, 2020-12-14:
Check that /root/.config/rclone is a link to /mutable/configs/rclone

murtuzahabib, 2020-12-15:
using <ls -l /mutable/configs/rclone/>, it gives me: "ls: cannot access '/mutable/configs/rclone/': No such file or directory"

marcone, 2020-12-15:
And what about `ls -la /root/.config/` ?

murtuzahabib, 2020-12-15:
root@teslausb:~# ls -la /root/.config/
total 12
drwxr-xr-x 3 root root 4096 Dec 13 23:36 .
drwx------ 6 root root 4096 Dec 14 00:03 ..
drwxr-xr-x 2 root root 4096 Dec 13 23:41 rclone

marcone, 2020-12-16:
OK, so that's an actual folder on the root partition, which explains the "read-only file system" error you are seeing.
/root/.config/rclone is supposed to be a symbolic link to /mutable/configs/rclone
The setup script achieves this by moving /root/.config/rclone to /mutable/configs/rclone and then creating the link.
The fact that it is not a link would therefore seem to indicate that that setup script didn't run. You said it worked before, though, so I think that previously the root partition wasn't read-only, e.g. because you had made it writeable in order to configure rclone, and didn't re-run setup and reboot afterwards.

You should be able to get things working again by ensuring that the required variables are specified in the config file, and then running setup again. See the "Easy rclone setup" section in https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md

murtuzahabib, 2020-12-19:
Re-run the rclone setup, variables supplied. Working now. Hopefully stays working.

Thanks again.

murtuzahabib, 2020-12-19:
Hi, would the application archive to google drive automatically whilst connected to the wifi, or, does one need to run a certain rclone command to do that?

If the latter, could you please share the rclone command format? 

Thanks.

marcone, 2020-12-20:
If you configured rclone and teslausb correctly it will archive automatically.

closed 2020-12-20 by marcone