marcone/teslausb #575: "couldn't determine idle interval"

Jughead135, 2020-12-15:
Not necessarily an issue, but a question--this topic is the only one I've found related [closed issue #396 ]:

What does the OP's highlighted text mean? Is there something "wrong" (or, at least not fully working) when the system can't determine the idle interval...?

I just copied from the OP below, but I get the exact same message in my `/mutable/archiveloop.log` file every time it sits idle long enough (> 1 hr).

```
Sun 12 Jan 14:11:01 GMT 2020: waiting up to 90 seconds for idle interval
Sun 12 Jan 14:12:42 GMT 2020: couldn't determine idle interval
```

marcone, 2020-12-15:
Perfectly normal. It's logged when teslausb is waiting for the car to write something, but doesn't see any writes from the car for 90 seconds. This generally means that the Pi is on, but the car is not.

closed 2020-12-15 by marcone

Jughead135, 2020-12-15:
Excellent!  Thank you!