marcone/teslausb #576: Cannot connect to Hidden Network

TonyBrobston, 2020-12-16:
When I have my wireless network set to hidden, my pi will not connect. If I flip this one setting, making the network visible, my pi is able to connect.
[image]

From a bit of googling it sounds like it is possible, involving adding `scan_ssid=1` to `wpa_supplicant`. However this seems to go outside the route of using `teslausb_setup_variables.conf` and I'm not sure if that is a good thing or bad thing?

https://www.raspberrypi.org/forums/viewtopic.php?t=250588

A little more backstory. I recently created a few vLANs on my network in order to separate my Home Automation devices from other devices they shouldn't be able to see or talk to. The result is that I have 6 wireless networks: Main, Guest, IoT, NoT (however IoT and NoT end up with 2 wireless networks each to lock in which AP a device is connect to). In an effort to decrease the number of SSID's I'm showing, I hide my IoT and NoT networks. So far, all of my 40-some devices have worked great.

marcone, 2020-12-17:
You can add this option to wpa_supplicant.conf yourself if you want/need it.
See also https://github.com/marcone/teslausb/issues/254

closed 2020-12-17 by marcone

TonyBrobston, 2020-12-18:
@marcone Sorry to ask a dumb question, but can you help with more detailed steps?

I am assuming I need to `cp wpa_supplicant.conf.sample wpa_supplicant.conf`, then updated `ssid` and `psk` and add `scan_ssid=1` to `network`. Unfortunately when I run that copy command I am faced with an error: `cp: cannot create regular file 'wpa_supplicant.conf': Read-only file system`. What causes this to become read-only? Once I am able to copy/modify the `wpa_supplicant.conf`, do I have to rerun a command or will it pick up these modify settings on the fly?

marcone, 2020-12-18:
If you have a working installation already, run `/root/bin/remountfs_rw` first, then edit /etc/wpa_supplicant/wpa_supplicant.conf

If you're doing a new install, add the scan_ssid line to the wpa_supplicant.conf.sample file. Setup will take care of the rest.

TonyBrobston, 2020-12-23:
Thanks for the details @marcone, this worked perfectly.