marcone/teslausb #577: 8GB Pi4B with 1 TB SSD- No USB mode at Tesla

rudiger999, 2020-12-19:
Everything seems to go fine in the setup, but no Camera Icon appears for Pi4B.  I can SSH to the unit while it is in the Tesla, and can see the drives and partitions.  Any thoughts?

pi@teslausb:~ $ df
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root        1801584  1285520    406496  76% /
devtmpfs         3900380        0   3900380   0% /dev
tmpfs            4032476        0   4032476   0% /dev/shm
tmpfs            4032476     8628   4023848   1% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs            4032476        0   4032476   0% /sys/fs/cgroup
/dev/sda1        1870104     5664   1751392   1% /mutable
/dev/sda2      974291628 38283196 936008432   4% /backingfiles
/dev/mmcblk0p1    258095    55111    202984  22% /boot
tmpfs             806492        0    806492   0% /run/user/1000
pi@teslausb:~ $ fdisk
fdisk: bad usage
Try 'fdisk --help' for more information.
pi@teslausb:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk
├─sda1        8:1    0   1.9G  0 part /mutable
└─sda2        8:2    0 929.6G  0 part /backingfiles
mmcblk0     179:0    0 476.9G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0   1.8G  0 part /

rudiger999, 2020-12-21:
And the setup log (split into two parts) follows:

Sat 12 Dec 05:18:47 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Sat 12 Dec 05:18:47 GMT 2020 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat 12 Dec 05:18:47 GMT 2020 : Rebooting...
Sat 12 Dec 05:19:07 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Sat 12 Dec 05:19:07 GMT 2020 : Grabbing main setup file.
Sat 12 Dec 05:19:17 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:19:30 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:19:43 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:19:57 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:10 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:23 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:36 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:49 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:02 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:16 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:29 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:42 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:55 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:08 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:21 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:35 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:48 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:01 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:14 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:27 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:40 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:54 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:24:07 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:24:20 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:24:33 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:24:46 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:24:59 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:25:13 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:25:26 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:25:39 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:25:52 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:26:05 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:26:18 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:26:32 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:26:45 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:26:58 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:27:11 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:27:24 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:27:37 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:27:50 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:28:04 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:28:17 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:28:30 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:28:43 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:28:56 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:29:09 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:29:23 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:29:36 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:29:49 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:30:02 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:30:15 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:30:28 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:30:42 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:30:55 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:31:08 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:31:21 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:31:34 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:31:47 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:32:00 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:32:14 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:32:27 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:32:40 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:32:53 GMT 2020 : get_script failed, retrying
[... 1746 lines cut ...]
Mon 21 Dec 00:20:50 MST 2020 : Downloaded /tmp/make-root-fs-readonly.sh ...
Mon 21 Dec 00:20:50 MST 2020 : curl -o /tmp/configure.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/setup/pi/configure.sh
Mon 21 Dec 00:20:51 MST 2020 : Downloaded /tmp/configure.sh ...
Mon 21 Dec 00:20:51 MST 2020 : Fixing the modules-load parameter in /boot/cmdline.txt...
Mon 21 Dec 00:20:51 MST 2020 : Fixed cmdline.txt.
Mon 21 Dec 00:20:56 MST 2020 : create-backingfiles-partition: USB_DRIVE is set to /dev/sda
Mon 21 Dec 00:20:56 MST 2020 : create-backingfiles-partition: Looks like backingfiles and mutable partitions already exist. Skipping partition creation.
Mon 21 Dec 00:20:56 MST 2020 : create-backingfiles-partition: backingfiles already defined in /etc/fstab. Not modifying /etc/fstab.
Mon 21 Dec 00:20:56 MST 2020 : create-backingfiles-partition: mutable already defined in /etc/fstab. Not modifying /etc/fstab.
Mon 21 Dec 00:20:56 MST 2020 : create-backingfiles-partition: Done.
Mon 21 Dec 00:20:57 MST 2020 : calling configure.sh
Mon 21 Dec 00:20:57 MST 2020 : configure: Pushover not configured.
Mon 21 Dec 00:20:57 MST 2020 : configure: Gotify not configured.
Mon 21 Dec 00:20:57 MST 2020 : configure: IFTTT not configured.
[... cut ...]
Mon 21 Dec 00:20:57 MST 2020 : curl -o /root/bin/send_sns.py https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/send_sns.py
Mon 21 Dec 00:20:58 MST 2020 : Downloaded /root/bin/send_sns.py ...
Mon 21 Dec 00:20:58 MST 2020 : configure: Checking archive configs: 
Mon 21 Dec 00:20:58 MST 2020 : configure: done
Mon 21 Dec 00:20:58 MST 2020 : configure: Using archive module: run/cifs_archive
Mon 21 Dec 00:20:58 MST 2020 : configure: Installing base archive scripts into /root/bin

rudiger999, 2020-12-21:
Setup log part 2:
Mon 21 Dec 00:20:58 MST 2020 : curl -o /root/bin/archiveloop https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/archiveloop
Mon 21 Dec 00:20:58 MST 2020 : Downloaded /root/bin/archiveloop ...
Mon 21 Dec 00:20:58 MST 2020 : curl -o /root/bin/waitforidle https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/waitforidle
Mon 21 Dec 00:20:59 MST 2020 : Downloaded /root/bin/waitforidle ...
Mon 21 Dec 00:20:59 MST 2020 : curl -o /root/bin/remountfs_rw https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/remountfs_rw
Mon 21 Dec 00:20:59 MST 2020 : Downloaded /root/bin/remountfs_rw ...
Mon 21 Dec 00:20:59 MST 2020 : configure: Skipping tesla_api.py install because no credentials were provided
Mon 21 Dec 00:20:59 MST 2020 : configure: Installing archive module scripts
Mon 21 Dec 00:20:59 MST 2020 : curl -o /tmp/verify-and-configure-archive.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/verify-and-configure-archive.sh
Mon 21 Dec 00:20:59 MST 2020 : Downloaded /tmp/verify-and-configure-archive.sh ...
Mon 21 Dec 00:20:59 MST 2020 : curl -o /tmp/write-archive-configs-to.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/write-archive-configs-to.sh
Mon 21 Dec 00:21:00 MST 2020 : Downloaded /tmp/write-archive-configs-to.sh ...
Mon 21 Dec 00:21:00 MST 2020 : curl -o /root/bin/archive-clips.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/archive-clips.sh
Mon 21 Dec 00:21:01 MST 2020 : Downloaded /root/bin/archive-clips.sh ...
Mon 21 Dec 00:21:01 MST 2020 : curl -o /root/bin/connect-archive.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/connect-archive.sh
Mon 21 Dec 00:21:01 MST 2020 : Downloaded /root/bin/connect-archive.sh ...
Mon 21 Dec 00:21:01 MST 2020 : curl -o /root/bin/disconnect-archive.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/disconnect-archive.sh
Mon 21 Dec 00:21:02 MST 2020 : Downloaded /root/bin/disconnect-archive.sh ...
Mon 21 Dec 00:21:02 MST 2020 : curl -o /root/bin/archive-is-reachable.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/cifs_archive/archive-is-reachable.sh
Mon 21 Dec 00:21:02 MST 2020 : Downloaded /root/bin/archive-is-reachable.sh ...
Mon 21 Dec 00:21:02 MST 2020 : verify-and-configure-archive: Installing/updating required packages if needed
Mon 21 Dec 00:21:04 MST 2020 : verify-and-configure-archive: Done
Mon 21 Dec 00:21:04 MST 2020 : verify-and-configure-archive: Verifying that the archive server your_archive_name_or_ip is reachable...
Mon 21 Dec 00:21:04 MST 2020 : verify-and-configure-archive: STOP: The archive server your_archive_name_or_ip is unreachable. Try specifying its IP address instead.
Mon 21 Dec 00:29:07 MST 2020 : Detecting whether to update wpa_supplicant.conf
Mon 21 Dec 00:29:07 MST 2020 : Starting setup.
Mon 21 Dec 00:29:07 MST 2020 : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Mon 21 Dec 00:29:08 MST 2020 : curl -o /tmp/593/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/setup/pi/setup-teslausb
Mon 21 Dec 00:29:09 MST 2020 : Downloaded /tmp/593/setup-teslausb ...
Mon 21 Dec 00:29:09 MST 2020 : WARNING: main-dev contains a different version of /root/bin/setup-teslausb. It is recommended to run '/root/bin/setup-teslausb selfupdate' to update to that version
Mon 21 Dec 00:29:09 MST 2020 : curl -o /tmp/593/rc.local https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/pi-gen-sources/00-teslausb-tweaks/files/rc.local
Mon 21 Dec 00:29:10 MST 2020 : Downloaded /tmp/593/rc.local ...
Mon 21 Dec 00:29:10 MST 2020 : rc.local is up to date
Mon 21 Dec 00:29:10 MST 2020 : Updating package index files...
Mon 21 Dec 00:29:18 MST 2020 : Verifying that the requested configuration is valid...
Mon 21 Dec 00:29:18 MST 2020 : curl -o /tmp/verify-configuration.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/setup/pi/verify-configuration.sh
Mon 21 Dec 00:29:18 MST 2020 : Downloaded /tmp/verify-configuration.sh ...
Mon 21 Dec 00:29:18 MST 2020 : USB_DRIVE is set to /dev/sda. This will be used for /mutable and backingfiles.
Mon 21 Dec 00:29:18 MST 2020 : Verifying that there is sufficient space available on the USB drive ...
Mon 21 Dec 00:29:18 MST 2020 : There is sufficient space available.
Mon 21 Dec 00:41:56 MST 2020 : Downloading additional setup scripts.
Mon 21 Dec 00:41:56 MST 2020 : curl -o /root/bin/remountfs_rw https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/remountfs_rw
Mon 21 Dec 00:41:57 MST 2020 : Downloaded /root/bin/remountfs_rw ...
Mon 21 Dec 00:41:57 MST 2020 : curl -o /root/bin/make_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/run/make_snapshot.sh
Mon 21 Dec 00:41:57 MST 2020 : Downloaded /root/bin/make_snapshot.sh ...
[... cut ...]
Mon 21 Dec 06:15:33 MST 2020 : WARNING: main-dev contains a different version of /root/bin/setup-teslausb. It is recommended to run '/root/bin/setup-teslausb selfupdate' to update to that version
Mon 21 Dec 06:15:33 MST 2020 : curl -o /tmp/594/rc.local https://raw.githubusercontent.com/marcone/teslausb/33c244924b7d1e050ef840a6031ae391ffe2e21d/pi-gen-sources/00-teslausb-tweaks/files/rc.local
Mon 21 Dec 06:15:34 MST 2020 : Downloaded /tmp/594/rc.local ...
Mon 21 Dec 06:15:34 MST 2020 : rc.local is up to date
Mon 21 Dec 06:15:34 MST 2020 : Updating package index files...

rudiger999, 2020-12-21:
A question:  Tesla is expecting to see TeslaCam and Music folders.  Is this different when using TeslaUSB?

rudiger999, 2020-12-21:
Any thoughts on this?

rudiger999, 2020-12-21:
Also, I just completed the Pi4B setup WITHOUT the external drive.  It worked perfectly and is operating as expected.  so this is definitely a problem with the external drive setup.

marcone, 2020-12-21:
Your setup is incomplete, because you didn't specify the name of your file share server:

> Mon 21 Dec 05:59:05 MST 2020 : verify-and-configure-archive: STOP: The archive server your_archive_name_or_ip is unreachable. Try specifying its IP address instead.

closed 2020-12-21 by marcone