marcone/teslausb #579: Rsync not working with 20201211 (new build)

cleanev, 2020-12-20:
Can’t seem to make rsync work with latest update.

Diagnostics file uploaded at - https://gist.github.com/cleanev/273f60cf4ef8d59ffe5d86f29145501c#file-gistfile1-txt

marcone, 2020-12-20:
Looks like there's some kind of fatal error during archiving, since it logs the start message but not the completion message.
What does it say in /tmp/archive-rsync-cmd.log and /tmp/rsynclog ?

cleanev, 2020-12-20:
Thanks @marcone - i ended up using cifs yet again on two teslausb. I never checked /tmp for the files you mentioned however i am building this yet again to check and see those files.
Will revert soon with response.

marcone, 2020-12-25:
I can't reproduce this, rsync works for me.

closed 2020-12-25 by marcone