marcone/teslausb #581: Domain Credential Capablities

kareml0re, 2020-12-22:
Hi,

To add domain CIFS support, one only has to add a domain=<DOMAIN_NAME> entry to the /tmp/teslaCamArchiveCredentials file which is executed in the following file:

write-archive-configs-to.sh

They are also defined in the following:

/tmp/setup-teslausb
/tmp/configure.sh

I would like to test a change so wondered if I can get a branch to test with of the current main branch.

Thanks,
Anthony

marcone, 2020-12-22:
The beauty of github is that you can just create your own copy of the project to experiment or modify.
Click the green "code" button on the "code" tab.

closed 2020-12-22 by marcone

kareml0re, 2020-12-22:
Hi,

This is correct, and I changed the code locally.  I will fork and point the
conf file to my fork, I think this is the easiest way, I want to be able to
use the existing upgrade utility to test.

Thanks,

Anthony