marcone/teslausb #585: Changing archive target and re-running setup-teslausb doesn't update /etc/fstab

MikeBishop, 2020-12-26:
When I got a NAS, I needed to point my existing teslausb install to the new location.  I updated the config file and re-ran setup; setup said the archive target was reachable and I could mount it manually, but the log kept showing an error that the share couldn't be reached.

I finally found that the problem was that `/etc/fstab` hadn't been updated to point to the new target, and the scripts simply mount `/mnt/archive`.  When I deleted the lines from fstab and ran setup again, it was re-added with the correct target.

marcone, 2020-12-26:
Right, not every configuration change can be achieved just by rerunning setup.
PR welcome.

closed 2020-12-26 by marcone

MikeBishop, 2020-12-28:
Happy to take a look.  The simplest approach seems like it would be to simply remove the fstab entries for `$archive_path` and `$music_archive_path` and regenerate them.  However, that breaks anyone who needs to make manual modifications to the fstab entry after the fact.  Are you aware of people doing that, or is stomp-and-recreate a viable tactic here?

marcone, 2020-12-28:
I'm not aware of anyone doing that, but I don't know everyone who uses this project. There must be [dozens of us](https://www.youtube.com/watch?v=Ax96DCF5ptU)!

marcone, 2020-12-28:
Honestly, I suspect it's sufficiently rare for people to switch their archive share that it's not worth adding code for, especially since editing /etc/fstab will be much faster than rerunning setup. Maybe just update the documentation?

MikeBishop, 2020-12-29:
Speaking as a user, I only touch it when there's a problem or I need to change something; but whenever I touch it (have the f/s mutable), I re-run setup with "selfupdate" to update the scripts, do an `apt update`, etc.  Anytime I change settings, I re-run regular setup because I don't know well enough how those settings get propagated.  It's kind of a shotgun approach and might be overkill, but it (usually) works.

CodeChief, 2021-01-16:
As a new user, finding how to update the configuration was the most difficult part to get this tool running. I expected everything would be updated somehow or at least documented as a manual step. Couldn't a RegEx be used to remove the old fstab entries?

Moving or renaming archive folders, the storage/NAS or the server/device is a common use case even if not done every day. I think a significant number of new users will play around with the locations until they are happy, especially if they had no share/NAS in the home before.

MikeBishop, 2021-01-16:
Since someone else had the same issue, I went ahead and pushed the change I made a few weeks ago.

PR #600 (2021-01-24): Always regenerate fstab entries from setup
This *should* work based on unit tests, but full disclosure:  I haven't done an end-to-end re-setup to be sure.  Fixes #585.