marcone/teslausb #589: rclone archive script no longer calls send-push-message

joemiller, 2020-12-26:
It looks like this commit https://github.com/marcone/teslausb/commit/bbd851665288774abeb2707a776a499bd57f637e removed the call to `send-push-message` from the rclone archive-clips.sh script.

Thoughts on re-adding it? I see that the archive operations are meant to call the script with the number of moved files. It doesn't appear that the current flags used when calling `rclone` log the number of files moved.

For my own needs I'd be ok with a simple call like `Finished moving files` without a count of the files. Reason being is that I'd like to use a webhook to start a post processing job I'm running on GCP w/ cloud run and cloud build.

marcone, 2020-12-26:
Oops, I meant to replace that with a different message, but forgot to add it back in.

marcone, 2021-01-02:
I moved the messaging to archiveloop so now it sends the same messages for every archive method, including rclone: https://github.com/marcone/teslausb/commit/fee840aebaffc467b874643eac7d46633be547a8

closed 2021-01-02 by marcone