marcone/teslausb #595: All of the sudden, Tesla is saying "Dashcam USB drive is too small"

shackrock, 2021-01-03:
See image from the car below:

[image]

This started happening a week or two ago with no changes to the teslausb installation. It had been working for about a year, auto-uploading to my NAS when it saw wifi flawlessly.

Since this happened I decided to SSH in and do an update/upgrade.  I think it was successful, but to be honest I can't figure out how to check the version that I'm on, or where to find the conf file anymore =).  

Anyhow, I'm not sure what else to try here, i.e. if I should reinstall the entire thing with the latest image, or if I can somehow correct the partition sizes through SSH?  And, how can I confirm i'm on the latest version?

Thanks!

cleanev, 2021-01-04:
I have been facing this issue since Tesla pushed 2020.48.10. I currently have 2020.48.26 on Model 3 and face this issue, Model Y with 2020.48.30 also shows up with this message. I have to reboot TeslaUSB a few times to get it going again only to begin with above error. Something has changed on Tesla’s end or is it the file system type mismatch?

BTW, I use 256GB MicroSD cards in both vehicles.

marcone, 2021-01-04:
@shackrock there's no way to tell which version of teslausb you're on. Teslausb doesn't even really have versions as such. But if you ran `setup-teslausb upgrade`, then you're on the latest.

@cleanev Not sure what you mean by "I have to reboot TeslaUSB a few times to get it going again".
The car will show this message if the "cam" drive is smaller than 32GB. It should still work, it'll just show that notification.

You can do a fresh install and specify a larger `CAM_SIZE`, or if you're feeling adventurous you can try https://github.com/marcone/teslausb/issues/560#issuecomment-730722972

closed 2021-01-04 by marcone

cleanev, 2021-01-05:
@marcone - i have to unplug and replug for vehicles to see USB to record. This is daily routine for both cars.

I am using full space from 256 MicroSD as TeslUSB with below settings and have been using this for a while
export musicsize=0%
export camsize=16G

And this is how the final outcome is:
/dev/mmcblk0p1     vfat      253M   29M  224M  12% /boot
/dev/mmcblk0p4     ext4       93M   11M   76M  13% /mutable
/dev/mmcblk0p3     xfs       237G  215G   22G  91% /backingfiles

Is this the correct way to use all of the disk space for USB?

marcone, 2021-01-06:
`export camsize=16G` means that the car will see a 16GB drive.

cleanev, 2021-01-06:
Thanks @marcone 
I switched to following config file entries switching variables to newer ones and rebuilt 
#export MUSIC_SIZE=0%
export CAM_SIZE=100%

I am hoping that this change does it - 

Current df -hT after rebuild with new variables as indicated above shows 

/dev/mmcblk0p4     ext4       93M  1.6M   85M   2% /mutable
/dev/mmcblk0p1     vfat      253M   56M  197M  22% /boot
/dev/mmcblk0p3     xfs       237G  1.8G  235G   1% /backingfiles

What I am wondering is xfs volume is still the same 237G, wonder what changes by changing CAM_SIZE=100%. Will report if I still see drive too small error and whether pulling power requirement goes away

cleanev, 2021-01-06:
@marcone - Thank GOD this is a closed thread.

After creating new TeslaUSB last night I figured to go over all the documentation and issues to get a better handle on things as I was questioning and answering them and with my perfectionism I was going nuts. 
Indeed it takes a short while to read thru readme file however poring over issues is a tall order but am happy I did.

I see that the readme page (home page) mentions clearly about sizing camsize to [16GB](https://github.com/marcone/teslausb#optional-allocate-sd-card-storage). Would be prudent to change that to 32GB or even 33GB now that newer software from Tesla has started to complain?

I did go over this [issue](https://github.com/marcone/teslausb/issues/568) that references #560 & #265. I learnt a new way to change settings

Thanks for your patience and assistance

miles267, 2021-01-06:
Hi @cleanev, I've been consistently using EXPORT CAMSIZE=64G since back in September 2020 without issue.  To date I've not received the "Dashcam USB Drive is Too Small" error at any point either.  Since a recent update, I have received some sort of USB port not functioning alert under same alert menu where low tire pressure is displayed, but I simply ignore it and teslausb seems to work fine and the alert/error eventually just goes away.

shackrock, 2021-01-06:
Should we limit to 64G if we aren't including our own music?  Any reason **not** to leave CAMSIZE at 100%, for example?

cleanev, 2021-01-06:
@shackrock - As @marcone mentions in README.md file by keeping camsize at 16GB gave more to save for sentry and saved. Rather than making inferences i would like @marcone to chime in 

Readme.md mentions 
If you anticipate needing more space for saved recordings, for example if your car generally reports much more Sentry events, you manually save recordings a lot, or if you're going to be away from wifi for multiple days, then increase size as needed. In order for teslausb to preserve recordings older than an hour, there needs to be enough free space on the sd card, at least 'CAM_SIZE' worth, preferably much more.

shackrock, 2021-01-06:
But if I have a 128GB card, why would I not use 100G for cam and 20G for music, and call it a day?  Is there an "other" category besides the reserved 6G for the setup stuff?

marcone, 2021-01-07:
> But if I have a 128GB card, why would I not use 100G for cam and 20G for music, and call it a day?

The car doesn't really make good use of that storage, since it deletes RecentClips that are older than an hour. Allocating less storage to the car and leaving more for teslausb lets you keep more than the last hour of RecentClips.

> Is there an "other" category besides the reserved 6G for the setup stuff?

There's linux/teslausb itself, weighing in at about 2G, plus that 128GB card is about 128 * 10<sup>9</sup> bytes, so as far as teslausb is concerned it's really only 119 GB (119 * 2<sup>30</sup>)

shackrock, 2021-01-07:
@marcone I see, would you call this reasonable then for 128GB, or still goo high on CAM_SIZE?
export CAM_SIZE=60G
export MUSIC_SIZE=20G

marcone, 2021-01-07:
It all depends on your use case. If you expect to accumulate many hours of sentry recordings between wifi connections, you might need 60G. I'd been using 24G until the car started complaining and then I bumped it to 30G (which is now the default in the sample config file)