marcone/teslausb #598: USB Device Malfunction - Center Console Port

Nakatomi2010, 2021-01-08:
After 2020.48.26 I started getting this error in my Model 3:

https://imgur.com/lS9PRoG

And, now that I have my Model X back, I can confirm I get it on my Model X as well, after 2020.48.30.

In both vehicles I'm using a Raspberry Pi 4.  The fix is enough enough, unplug and plug it back in again.  Not sure if it is because there's a point in time where it starts drawing too much power or something, but it is a noticeable issue.

I don't have a PI Zero to determine if there's an issue with this there now too.

CMDRPhaedra, 2021-01-08:
I’m only a recent user of TeslaUSB, i’m using a Pi Zero W and was getting this exact issue.. resolved it by changing to a better quality micro USB cable. Haven’t had it since.

Nakatomi2010, 2021-01-08:
Got a link to the cable(s) you're using?

CMDRPhaedra, 2021-01-08:
Here you go https://www.anker.com/products/variant/powerline-4-in-micro-usb/B8135011

discojon, 2021-01-08:
I get the same error message after about an hour of driving.  I tried removing my hub and everything and nothing helped.

ScottESanDiego, 2021-01-12:
Same.  This same setup (Pi Zero W, center console port, Model 3 from a few years ago) has worked since for ages, and my wife started noticing this error about once a day since the 2020.48.30 update.  The Pi is on a battery backup (I have a battery connected to the microUSB power port, in addition to the cable on the data port) - not sure if that's a common factor or not with the other reports.

GoodyGizmos, 2021-01-23:
I'm using the same cable as @CMDRPhaedra and still getting this issue.

marcone, 2021-01-24:
The only times I've seen "USB device malfunction" was when the Pi was drawing too much power (the car seems to complain when there's a high load on the USB port for more than a few seconds, like for a Pi4+SSD booting up), or when I unplugged a (battery powered) Pi0 from the car immediately after getting in to the car.

ScottESanDiego, 2021-01-24:
It seems to happen when the Pi loses WiFi (e.g., when about a block from the house) - Is there anything that happens in the archiveloop when that event occurs?

discojon, 2021-01-25:
I'm seeing the issue after about an hour of driving.  Pi0, with or without hub / phone chargers / other usb devices plugged in.  It has persisted through at least two fw updates and a complete reinstall of tesla USB

Nakatomi2010, 2021-01-25:
> I'm seeing the issue after about an hour of driving. Pi0, with or without hub / phone chargers / other usb devices plugged in. It has persisted through at least two fw updates and a complete reinstall of tesla USB

Have you tried it on 2020.48.35.5?

discojon, 2021-01-25:
Yeah.  Still saw it a couple times already unfortunately.

Nakatomi2010, 2021-01-25:
> The only times I've seen "USB device malfunction" was when the Pi was drawing too much power (the car seems to complain when there's a high load on the USB port for more than a few seconds, like for [...]

My Pi4 has no USB attached.  Just an SD card.  Admittedly, Pi4 versus Pi0, so more power hungry to start.

I am not 100% sure this is TeslaUSB though.   Someone on the Tesla Discord server mentioned getting the same error with a flash drive.  He said his car was being brought in to have a working session with the Tesla engineers on the phone.

Same error, but different circumstances.

Nakatomi2010, 2021-01-25:
> Yeah. Still saw it a couple times already unfortunately.

Damn.  I haven't driven with 2020.48.35.5 installed yet, was hoping this was one of the bugs.

I did report the issue to Tesla via the support chat.  But they stated TeslaUSB wasn't supported, so they don't appear to be giving the complaint too much weight.

Not sure if others want to report the issue as well, see if we can it sorted through a volume of complaints.

1activegeek, 2021-01-26:
Same issue here. Details below:

- PiZeroW
- Happens intermittently - sometimes continually after parking and getting back in, sometimes halfway through a drive, sometimes at the end of a drive
- Is not related at all to Wifi capability as it often happens or starts some times when im not even in range of Wifi that it can connect to (aka my home)
- Am on the latest release of software my car can take: 2020.48.35.5 - been happening for a few versions now as well
- Not a cable issue as I have also tried a different cable
- Also not related to other devices plugged in or not as I’ve tried variations there as well

JohPePe, 2021-01-26:
I've had the same problem and error messages using a few of the latest car firmware versions including 2020.48.35.5

But earlier i used a power bank connected to the pi to supply backup power. Disconnected the power bank, no more error messages. 👍😃

ScottESanDiego, 2021-01-27:
@JohPePe Isn't that solution amounting to "reboot the Pi when the car comes out of sleep"? (I have my PiZeroW connected to a battery bank as well, so I can SSH into it "whenever", not just when the car is awake)

JohPePe, 2021-01-27:
@ScottESanDiego yes, I guess that’s why it’s working for me... 😀

TokugawaHeavyIndustries, 2021-01-28:
Same here.  Sucks though as this is probably a car issue, not a Pi issue.

RichardP-1, 2021-03-15:
Same issue here, happens randomly and regularly since late December 2020. Pi0, SD card only connected.

brettjenkins, 2021-03-17:
Getting the error but it's still working fine. More annoying than anything.

ScottESanDiego, 2021-04-02:
I removed the battery that was driving my PiZeroW (per the above discussion), and didn't have any change in behavior.  There's still the more-or-less random failure, and associated noise and message on the screen.  Looks like I have to reboot the Rpi (aka, unplug/replug the microusb connector) to get the car the recognize it again.

This is super annoying. :-/

TokugawaHeavyIndustries, 2021-04-03:
For what it's worth, I moved over to a RPi4 and the problem disappeared.

miles267, 2021-04-03:
> For what it's worth, I moved over to a RPi4 and the problem disappeared.

How do you power a RPi4? I’d do it in my Model 3 if I could figure out how without taking console apart.

TokugawaHeavyIndustries, 2021-04-03:
@miles267 USB-A Y-Cable ([link](https://www.amazon.com/gp/product/B00NIGO4NM/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)) and a USB OTG cable ([link](https://www.amazon.com/Cerrxian-Plated-Right-Angle-Charge/dp/B075QZ8M2Y)).  The OTG cable plugs into the power side of the Y cable, and the data side of the Y cable plugs into one of the Pi's USB ports.  Then just plug in to the console.

TokugawaHeavyIndustries, 2021-04-03:
@miles267 I deleted my previous comment because I had my mind all messed up haha.  Basically just get a USB-Y cable and plug the data side into the Pi's front USB ports, and the power side into the USB-C connector (you'll need an adapater).  Then just plug it into the console's port

miles267, 2021-04-03:
> @miles267 I deleted my previous comment because I had my mind all messed up haha. Basically just get a USB-Y cable and plug the data side into the Pi's front USB ports, and the power side into the USB-C connector (you'll need an adapater). Then just plug it into the console's port

I just ordered the two cables you had previously linked. Are there others that are recommended instead?

(Am sure I’ll use those prev 2 cables for another project sometime in future anyway; won’t hurt to have them on hand)

TokugawaHeavyIndustries, 2021-04-03:
@miles267 I was thinking about it wrong and the ones I linked are the wrong gender, if I remember right.

miles267, 2021-04-03:
> @miles267 I was thinking about it wrong and the ones I linked are the wrong gender, if I remember right.

If you wouldn’t mind linking the appropriate cables it use it would be greatly appreciated. Will definitely grab them and try a RPi4. Also like the benefit of 5 ghz WiFi as my RPi Zero is only 2.4 ghz. I have one sitting, unused.

Also, I’ve found that if I turn on Sentry Mode at home it keeps my USB powered in the console and the car doesn’t go to sleep. That caused my errors to go away. It confirms they were happening since teslausb isn’t keeping the car awake.

TokugawaHeavyIndustries, 2021-04-03:
@miles267 Now that I'm thinking about it, use that Y-cable I linked, and plug the data side (all black) into the RPi4's USB port.  Plug the USB OTG cable into the charge-only side (the red/orange side).  Then use this ([link](https://www.amazon.com/Monoprice-Male-24AWG-Cable-Plated/dp/B002KKZRYM)) to plug the Y-cable into the car.

And yeah, that makes sense.  I keep sentry on here at home too.  I did, however, get random disconnects while driving with the PiZero.

marnikvde, 2021-04-05:
I'm getting the same random failures on my 2021 Model 3, using a Pi0 with an SD card, and a single USB cable from the Pi0 to the glovebox USB slot. I'm using sentry mode all the time/everywhere.

I'm also missing some video clips from time to time, and I guess it's related. In a sentry-folder, there is usually about 10 minutes of footage, from 4 different camera's, so about 40 files (all 1 minute long), but I'm often missing a few of those minutes, making the footage unusable. I thought it was a CIFS/rsync issue, but it looks like the files just are not on the SD-card to start with, which makes sense if the car and Pi0 lose connection from time to time.

Any hints on how to start debugging this? Maybe I should write a script with some Pi0 debugging data, such as the power it's drawing, or the read/write bandwidth etc, to find out what could be causing this?

miles267, 2021-04-09:
> @miles267 Now that I'm thinking about it, use that Y-cable I linked, and plug the data side (all black) into the RPi4's USB port. Plug the USB OTG cable into the charge-only side (the red/orange [...]

Thanks.  It looks like I also need to get a "coupler" adapter to connect the male USB-A end of the OTG cable to the Y adapter.  Didn't realize that.  I'll find one.  You're not having any issues supplying enough power to the RPi4 this way?

TokugawaHeavyIndustries, 2021-04-10:
@miles267 I've had no problems so far!

CodeChief, 2021-04-10:
Although it's good that we have ingenious workarounds. How about we get towards identifying the problem and finding a solution? Randomly buying different types of USB cables in the hope the problem goes away, left with a mess of cables in the glove box is no "solution".

1. Is it hardware or RasPi EEPROM firmware?
2. Is it isolated to the RasPi Zero?
3. Does the indication that RasPi 4 could behave better give us a hint to where the problem lies/potential fix for the Zero?
3. Is it actually (more likely) a software/firmware issue of (TeslaUSB OTG driver confusing the USB consumer/Tesla car).

It's less plausible that Tesla broke their USB compatibility with an update. Much more likely that the OTG method used in the TeslaUsb/Linux image isn't sound accordingt to USB power+storage standards or was broken by an update.

**Who knows the most about the method used here to emulate a USB stick in Linux? Which configuration/script/driver files are involved and when did they last change?**

I'd like to know soon as if it continues much longer I have to ditch TeslaUsb, but don't really want to. If there is no fix, then in the end we still need confirmation of exactly which Raspberry Pi models are compatible and how to connect them normally (with standard cables).

drlauridsen, 2021-04-12:
Just upgraded from MCU1 to MCU2 and AP2.5 to AP3.0.
At the same time sw was updated from 2020.48.37.1. to 2021.4.10.1
With MCU1 I had to use two cables to boot teslausb correctly, one data only cable and one power, otherwise I would be stuck in a boot loop.
This is not necessary after the MCU upgrade.

However I did not have the usb device error with MCU1 and sw 2020.x, but now have the error daily, sometimes multiple times in one drive, but quite random.
Pi Zero.

I updated the Pi Zero OS/teslausb a couple of days ago, and have not seen the error on my last drive, but need more test time..

fiveangle, 2021-04-14:
Came here to report the "USB device malfunction" error... 2018 P3 w/HW3 upgrade.  Running 2021.4.12. SSHed into it no problem... seems to be working 🤷.  What can we do to try and determine root cause ?  The Tesla is treating this slightly different than the typical errors because it doesn't flash the "\!/" icon to indicate listing of errors... so I can't get the timestamp of when exactly it occurred.  However, when I went into the pi to investigate, there's no logs in the typical places, so not sure even how to start on if something has observed from the pi side.

Plenty of people here willing to dig in, if you can just point us to the right shovel...

drlauridsen, 2021-04-16:
After updating teslusb pi zero with

`sudo -i /root/bin/remountfs_rw
sudo /root/bin/setup-teslausb selfupdate
sudo -i /root/bin/setup-teslausb upgrade
sudo /root/bin/setup-teslausb`

(No car sw update in between.)

I have not experienced the error.

marcone, 2021-04-16:
@drlauridsen FYI to update to the latest you only need the `setup-teslausb upgrade` part of that. The rest is unnecessary.

miles267, 2021-04-16:
> @miles267 I've had no problems so far!

@TokugawaHeavyIndustries what is the trick to getting the Dashcam icon to appear once you connect your RPi4 to the car?  I've connected all of the cables as you've shared.  All is good.  RPi4 powers up without issue, it connects to wifi, I'm able to SSH into it, but it doesn't seem to be recognized by the car.  I've tried rebooting the car's MCU, rebooting the RPi 4, waiting then connecting the USB data cable to the RPi 4, etc.  I've not yet managed to figure it out.  Thanks.

rmyadsk, 2021-04-16:
> what is the trick to getting the Dashcam icon to appear once you connect your RPi4 to the car?

A clean installation, patience, and luck. It can take upwards of a minute for the icon to appear as the pi completes its startup sequence and makes the mounts available. On my pi Zero this usually happens a dozen seconds or so after entering the "two blinks" runtime.

A useful sanity check is to plug the pi into the USB port of a Windows or Mac. Does the computer show and/or mount the CAM (and MUSIC, if it is defined) slices? If not, then there may be an issue with the installation.

Also keep in mind all cables are not created equal. Some cables are power only and will not transmit data.

fiveangle, 2021-04-20:
to add, already running latest

Jeffersonian, 2021-04-21:
Hitting this constantly on my Model 3.  This is really distracting when driving (and not recording is too risky for me).  I'm replacing my TeslaUSB with a real USB drive for now.

CodeChief, 2021-04-22:
Here are some test results which took a lot of time but might be helpful to isolate it:

0. Original situation was RasPi Zero W installed a while ago but with all APT, Firmware and TeslaUSB updates.
1. Reinstall RasPi 0 = same problem.
2. Update reinstalled RasPi 0 = same problem.
3. Different cables, Y-power-split, direct, several short/direct = same problem.
3. Swap card out into a RasPi 4 = same problem.
4. Format the SD via the Tesla = *** For a short while it worked with NO ERRORS, but then the DashCam went away and after a reboot the errors came back ***
5. Reinstall image on RasPi 4 = same problem.
6. Update firmware and all OS/APT packages on RasPi 4 = same problem.

As a side note I thought I fixed it when using the Tesla option for format the "DashCam USB drive" (TeslaUSB). It would be nice if this was supported, I don't think it is as I couldn't get it to mount afterwards. But the errors went away in this situation so I wonder if the problem with the USB is something to do with the way the partitions are mounted, not the USB emulation itself? Is it possible to test different layouts from the view of the car in the USB emulation/mount commands?

What I wonder is, if we get the "format USB" to work from the car, then maybe we fix the other problem. Because shouldn't this TeslaUSB appear *exactly* like a USB stick. So long as it doesn't support formatting, then it suggests it is not fully compatble as a "USB stick". Perhaps fixing that fixes the other problem.

rmyadsk, 2021-04-22:
@CodeChief I would strongly advise against having the vehicle attempt to format that drive - it is anticipating a greenfield USB device, which your teslausb CAM slice most certainly is not - it is a living, breathing Linux partition masquerading as one.

I would instead advise updating your teslausb installation in-situ, then copying debug logs here so we can help diagnose. A recent commit, live now, [will help](http://github.com/marcone/teslausb/commit/3b42a321d60a1e1854f18ce1bbc0c68ab2e90c97)

rmyadsk, 2021-04-22:
Also @CodeChief be aware some cables are just not meant to carry data - indeed, sometimes they transmit power only, or they transmit power and intermittent and/or unreliable data. You will have to pick and choose to find a production-ready solution.

marcone, 2021-04-22:
> Because shouldn't this TeslaUSB appear _exactly_ like a USB stick. So long as it doesn't support formatting, then it suggests it is not fully compatble as a "USB stick".

It does appear exactly like a USB stick. The problem with formatting it in the car is that the car will format it as ExFAT, which TeslaUSB doesn't support.

CodeChief, 2021-04-22:
> @CodeChief I would strongly advise against having the vehicle attempt to format that drive - it is anticipating a greenfield USB device, which your teslausb CAM slice most certainly is not - it is a [...]

BTW I'm a developer not a PC noob with lots of experience with virtual file formats for numerous technologies (Microsoft, VMware, Hyper-Converged). This is pretty low tech in my opinion and if there is an issue with formatting a virtual drive then the system used for emulation is not good.

CodeChief, 2021-04-22:
> It does appear exactly like a USB stick. The problem with formatting it in the car is that the car will format it as ExFAT, which TeslaUSB doesn't support.

Again so what, and why not use the same default format as Tesla chose for their system? If not possible, again then the problem is the tool selected for drive emulation and this could be the reason it's confusing the car into thinking the USB port/flash-drive has gone bad.

CodeChief, 2021-04-22:
> Also @CodeChief be aware some cables are just not meant to carry data - indeed, sometimes they transmit power only, or they transmit power and intermittent and/or unreliable data. You will have to pick and choose to find a production-ready solution.

I've amassed hundreds of USB cables in the 30 years of IT experience and actively sort and throw-out crappy power-only cables. I keep a few labelled for nostaliga/tests. I've tested TeslaUSB with Pi0, Pi 4 with new and old single and split cables. All of these cables work on a PC even high speed NVMe drive external enclosures. It has to work, it's not the cable.

CodeChief, 2021-04-22:
Based on my own device driver experience, I'd bet this issue has to do with the *behaviour* of the TeslaUSB scripts. You can't claim to be a USB stick but then behave like a network service (popping-up for a few seconds then disappearing before finally returning for a long time, then off again randomly when WiFi is available, like an on-demand service).

The problem I think is rooted in the constant dismounting of the drive. Why is this necessary? Is it the snapshot mechanism (actually violating the whole point of a snapshot, by not being online)? It's also not very flexible or user friendly.

Think about a design where the drive was always mounted, then remote commands could freely check status, even the SMB server could be used to remote browse regardless if/when it syncs. Why have this limitation that the TeslaUSB has to "see" WiFi then dissappear and reappear randomly?

Given this is a Linux system, what other snapshot/locking systems are available and fully featured (online snapshots or robust locking/retry behaviour)? Is the current tool already capable then all is needed are some other options to support online snapshot/difference scans?

TokugawaHeavyIndustries, 2021-04-23:
@CodeChief Your comments are coming off very arrogant. If you're so sure of the issue, I invite you to fix the issue on your own and submit a pull request. 

As you say you're a developer, please review the TeslaUSB source and you can see how the snapshot process and drive emulation works for yourself. Additionally, as you can see at the start of this issue, the problem in question originated after a car update, not a TeslaUSB push. 

And finally, to address your concerns about exFAT - the exFAT file system was added and made default in the car MCU as of version 2020.12. The Raspberry Pi bootloader does not support exFAT. Additionally, as of my last check, USB gadget tools does not support the exFAT FUSE module.

CodeChief, 2021-04-23:
Experience, not arrogance. Like if you tell an electrician more than once how to turn the lights on and off, don’t be surprised if you get an abrupt answer ;-) Anyway, no offence intended. Let's get on with troubleshooting...

1. **Cable** - I suggest this is closed as a potential cause if it works for power and data on a PC and is just a "cable", not a tailored interface (it's easy to confuse with phone specific and OTG/hub cables).
2. **Pi 4** - Close too unless some extra setting is needed which was not documented in quick setup?
3. **Tesla's fault** - Doesn’t sound plausible else we'd have dozens of devices not working anymore and news articles about compatibility problems by now. Seems like only TeslaUSB stopped working after the update, because as you confirm it does not behave like a USB stick. But okay then if it is their fault we should have somebody here with another device which stopped working about the same time and hence a way to re-create/prove it?
4. **Power** - I remember something was written about people with backup power/UPS HATs having some success, was that confirmed? I do have a power HAT around here so will test this at the weekend. I’d be surprised if that was the cause because then I’d also expect the power-hungry Pi 4 not work at all compared to a Pi Zero.
5. Behaviour - That's my favourite line of investigation right now because I also like the potential simplification and benefits which would come with the fix. In the mid-term I want to look at the scripts and tools used but do not have much time now. Linux is not my favourite system, so it would also be better if somebody who knows the toolset could jump in to help or comment. For example, how can we quickly test with the drive kept online the whole time to confirm or deny this path. I can at least try to comment out the script after the drive is mounted in the hope it stays online, then see what happens.
6. Format - I agree this is not a root cause. As said, it is just a test case to the behaviour goal (if at all possible, should the bootloader be an absolute limit). Ignore for now. Actually, what do the people at Raspberry Pi say about their emulation, any news there or beta versions of new firmware with better OTG support? I noticed in the recent EEPROM firmware update that they are testing a new way to drive the GPU/chip which is also responsible for the boot sequence. It might be linked.

What's left to examine as a potential root cause? I've thrown in some possibilities in a hope to trigger some positive line of investigation. Let's not repeat old responses which cannot be recreated/confirmed.

rmyadsk, 2021-04-23:
If the pi mounts the shares to a PC using the same cable, yes, we can eliminate the cable as a potential cause.

There used to be a .conf option to not use snapshots, but that has been deprecated and that functionality recently transferred to a lesser-used branch. I don't know if (strictly speaking ) `"not using snapshots" == "always on";` but would be interested to see if there is a change in the problematic behavior.

Nakatomi2010, 2021-04-27:
Ok, so, I'm not 100% certain, but I think I have something on this.

To preface this a bit, historically, when I get this error it's been on long drives and it occurs about every 45 minutes or so on the long drive.  Alternatively the car has been sitting still somewhere and the Service>Notifications area is littered with errors from Sentry mode being on.

I also have set "Cam storage" to 100% because, why not, it's not like I'm using it for music, I want the most storage available for videos.

This is the default settings in the TeslaUSB config file:

# Notes on sd card and image sizes:
#   * A 128 GB or larger sd card (or USB drive, when using Pi4) is recommended. The
#     minimum supported size is 64 GB.
#   * When specifying sizes for the recordings and music drives, you can use specific
#     sizes (e.g. 24G) or use sizes relative to the available space (e.g. 10%)
#   * The setup script will reserve a fixed 6 GB of space regardless of sd card size.
#   * CAM_SIZE should generally be somewhat small, around 16G to 32G, unless you plan
#     on accumulating a lot of footage between archive operations, in which case you
#     should increase the value (an hour of recordings, or 6 Sentry events, is about
#     7-9 GB of data).
#   * As of Tesla software version 2020.48.10, the car will warn when the recording drive
#     is smaller than 32 GB, but smaller values will still work. Tesla appears to use
#     power-of-ten based sizes, presumably because most storage manufacturers do too.
#     Because teslausb uses power-of-two sizes, you can specify "30G" for the size and
#     not get the warning.
#   * MUSIC_SIZE should be large enough to hold your music library, obviously.
#   * If no music size is specified, no music drive will be created.
export CAM_SIZE=30G
#export MUSIC_SIZE=4G

In speaking with Marcone, on Discord, using this 30G setting is recommended because TeslaUSB works predominantly off of snapshots, so you don't want the setting at 100% because the snapshots won't work.

Taking this recommendation to heart I rebuilt both my TeslaUSB instances to use the above 30G configuration.  My Model X has had no issues since this change was made.  My Model 3 has had the issue with a vastly lower frequency count.  When it does occur the logs seem to show that the snapshots ran out of space and were cleaning up a bit.

However, ultimately, for me at least, the issue appears to tie back to the CAM_SIZE setting.

As opportunity permits, can you all look and see what your CAM_SIZE setting is and perhaps lowering it, then seeing if the issue goes away?

rmyadsk, 2021-04-27:
To add: take note of both CAM_SIZE and also the physical capacity of your media. The first should be significantly smaller than the second. If it's not, that is a good indiction the snapshots are being stomped on and potentially causing issues.

> As opportunity permits, can you all look and see what your CAM_SIZE setting is and perhaps lowering it, then seeing if the issue goes away?

marnikvde, 2021-04-29:
I'm about to go for a complete reinstall on my RPI0, 64GB SD card, and I will use a CAM_SIZE of 30GB as suggested (it was 32GB untill now)? Not using music partition.

JakeShirley, 2021-05-05:
@marnikvde any update on this? How'd it work out?

marnikvde, 2021-05-05:
I'm still getting malfunction errors/warnings from my Tesla Model 3. There's always a triangle on the left of the sentry icon (about the malfunctioning USB device), no way to get rid of the error, so I would conclude that the issue persists.

Nakatomi2010, 2021-05-05:
> I'm still getting malfunction errors/warnings from my Tesla Model 3. There's always a triangle on the left of the sentry icon (about the malfunctioning USB device), no way to get rid of the error, so I would conclude that the issue persists.

Which version is your Tesla on?

Mine are on 2021.4.15.5 and 2021.4.15 now.  Maybe the issue as gone away in a later version and the CAM_SIZE variable is a fluke to when I got those updates?

marnikvde, 2021-05-05:
It has been on 2021.4.15 for a few weeks now.

AXEBro, 2021-08-10:
> It does appear exactly like a USB stick. The problem with formatting it in the car is that the car will format it as ExFAT, which TeslaUSB doesn't support.

How does one recover from this? I have this problem - because my pi0 had the slow USB error. And seemed to stop recording altogether, as I never saw the dashcam icon on the car after that. I formatted it in the car. 

Do I SSH in and delete the backing file, or is there a nicer way to handle it (or re-install from scratch)?

[12 more replies]

closed 2024-11-04 by marcone