marcone/teslausb #603: No drive on PC and car doesn't see it

Poncherelly, 2021-01-22:
has anyone been having problems with not being able to see the TESLAUSB drives show up in Windows and not being see by the car?

Here is where I am so far:
SSH to TESLAUSB = no issues
See folders on PC = No
checked /boot/teslausb-headless-setup.log = All Done
checked /mutable/archiveloop.log = Copying music from archive...
...waiting ...
See folders on PC = No
re-checked /mutable/archiveloop.log = new snapshot is identical to previous one, discarding
...waiting ...
re-checked /mutable/archiveloop.log = Waiting for archive to be unreachable...
See folders on PC = No
...waiting ...
re-checked /mutable/archiveloop.log = new snapshot is identical to previous one, discarding
See folders on PC = No
changed USB cable to a new data cable
...waiting ...
re-checked /mutable/archiveloop.log = new snapshot is identical to previous one, discarding
See folders on PC = No
tried
sudo -i
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb diagnose > /tmp/diagnostics.txt 

I now have a very lengthy diag. file but unfortunately not sure what I'm looking for.   The diag. file can be found here if anyone have time and knowledge to read through.
https://drive.google.com/file/d/1crZZYeNEZBnPwNIAGlDmBpkpUGTmmdP5/view?usp=sharing

re-checked /mutable/archiveloop.log and currently sitting at = new snapshot is identical to previous one, discarding 
with an updated timestamp ... seems to be recycling through the process but never stopping to connect to the PC

Much appreciated

marcone, 2021-01-22:
(tl;dr: the diagnostics show everything working normally. Check your cable)

The diagnostics show the Pi booting up at 11:22. A little later teslausb starts, and begins copying music from your music archive to the music drive. This starts at 11:23 and ends at 12:30, so took 53 minutes.
After that the Pi published its drives, so they should be visible to an attached PC or the car. If they're not, chances are your USB cable is bad, or you plugged it into the Pi's power port instead of the USB port.

Poncherelly, 2021-01-22:
Thanks for responding.
I've replace the cable with several I had then bought 2 new ones and still have the same problem.

From a network share point it seems to be working great, from a PC point it just won't see it. And I have it plugged in to the power/data port (middle) and not the power (end) port.

I'll try in the car again and worst case buy a new pi to try that I guess.

Richard

Purchasing a TESLA?  Please use My Referral Code<https://ts.la/richard48613> for free supercharging

Reply-to: reply@reply.github.com
Cc: richard-carpentier@outlook.com; author@noreply.github.com

marcone, 2021-01-22:
@Poncherelly I deleted your comment because you included your actual email address and Tesla referral code, but here's what you said:

> Thanks for responding.
> I've replace the cable with several I had then bought 2 new ones and still have the same problem.
>
> From a network share point it seems to be working great, from a PC point it [...]

Poncherelly, 2021-01-26:
I have read that if the pins touch, they can short out the USB data portion and that's what seems to have happened.

I performed some tests with a meter and when I touch the data pin (either one) the meter reacts.  I decided to buy a new board and just see ... low and behold, the drives show up on my PC.  I'm going to throw it in the car but I have no reason to think that my issue hasn't been fixed.

Thanks again Marcone for all your help with this project.  Can you point me to a donations link I can send you some coffee money to?

Thanks again!!

closed 2021-01-26 by Poncherelly